cedricp/ddt4all #103: Renault Megane 3

neutrondev03, 2018-10-05:
Hello! Not really a bug.

I am having problems with a Megane 3 from 2015. It had manual AC and it got replaced with an automatic AC.

Can someone please provide me all configuration/calibration data from PTC and all stuff AC related please?

All the best. Thank you very much for this tool I will donate. It is amazing.

neutrondev03, 2018-10-06:
Sorry for the noob question. What are the MTC parameters. I have a CAN CLIP gold chip too. I can't do autoconfig procedure. It says that the files from the reporog database are too old. I can give you the VIN if needed. Where do I look from the MTC params?

shrlnm, 2018-10-06:
CAN CLIP had to make the file with your MTC when you connected to car for a first time.

But if you have a CLIP then you may find MTC by yourself.

Find the folder BVMEXTRACTION and a subfolder with your platform (X95 for M3). 
Here  you can find some files and folders. MTC.dat contains all possible sets of MTC parameters for M3/S3. REF.dat contains all possible sets firmware and corresponding configurations. You need to find your lines in these files

Get 3 first characters of your VIN and go to appropriate folder. Actually only VF1 for  M3. Then get next 6 characters of your VIN and choose appropriate .dat fale. Reneme it to .zip and unpack. In one appropriate text file, find the line started with the last characters of your VIN.

This line should looks like 

51253186;06768777;0000086377;MNEMOCR:SSCEDC/402/TAR22/TAV32;DATE_MADC:05.06.2014;T_BOI:ND4

second field is a refernce to line in MTC.dat
third  field  is a reference to line in REF.dat

neutrondev03, 2018-10-09:
I went to those folders however I cannot find the file for my car.

I extracted the right file but didn't find next 6 VIN digits.

VIN: VF1BZLC05 (cannot find BZLC05).

Anyway even if I find those files. How can I auto configure UCH with the options I want. Can you describe the procedure?

shrlnm, 2018-10-09:
ok. It is looks like your car is absent in BVMEXTRACTION db.
Let's try another way
Which XML were chosen during scan by ddt4all for UCS and USM?

neutrondev03, 2018-10-09:
Sorry, Yeah I understand what you said. Unfortunately it is not my car and I did not save the ECU list.