cedricp/ddt4all #107: Megane 4 DisplayMode change with ddt4all

killuadk, 2018-11-05:
Hey 

if somme one could help plz i did by mistake change the DisplayMode as showing in photo 
i can connect with DDT4ALL but i can't find that parameter 

[image]

zatropin, 2018-11-12:
[image]
[image] bu bölümü denedim işe yaramıyor

zatropin, 2018-11-12:
İletişim kuralım WhatsApp 05345003932

zatropin, 2018-11-14:
https://github.com/cedricp/ddt4all/issues/110 
check the solution URL available

closed 2018-12-21 by cedricp