cedricp/ddt4all #108: Stop lamp control

maxclio, 2018-11-10:
after changing something on light settings that i do not remember on my clio 4 2014, I recieve stop lamp check light on even though stop lamp works perfectly. Can anyone check with theirs and tell me which one has wrong setting ? 

[image]

[image]
[image]
[image]

maxclio, 2018-11-10:
> If you did not delete your log files, it's possible to recover original parameters.

Unfortunately I dont have log files. Is it somewhere stored ?

cedricp, 2018-11-10:
in your installation directory there's a log directory with 2 files.

shrlnm, 2018-11-11:
The default configuration depends on ecus and functions installed in your car. So the values from other clio 4 2014 may not be suitable for you.
If you have deleted log file, then the original configuration can be restored only by your vin

shrlnm, 2018-11-11:
Give me your email

cedricp, 2018-11-11:
@shrlnm can you CC me your answer, I'm interested too

maxclio, 2018-11-11:
> Give me your email

Can you also send me the details about how to reset according to VIN?

shrlnm, 2018-11-11:
Most simplest way to get your default configuration is to make ECU emulator, connect it to CLIP and do reprog autoconfiguration

shrlnm, 2018-11-11:
164

mnigten, 2018-11-11:
I am interested too . Nigten@gmail.com

And about CLIP . Can it be done with a Chinese version?

shrlnm, 2018-11-11:
Actually any clip which may do a reprog. Emulator is a simple program and  ELM (usb is much better).

shrlnm, 2018-11-11:
Configuration database on reprog disk is strongly encrypted and get the data from it is very hard. But with simple emulator built from ELM you may trick the clip and make it to send commands of the default configuration to your emulator which log them.

cedricp, 2018-11-11:
Ok I see, smart way to get back parameters but need a CLIP hardware..

ohayyam, 2018-11-11:
> 
> 
> Give me your email

I am interested too.   omern7100@gmail.com

Thanks &  Regards

pcboys, 2018-12-06:
> Give me your email

I am interested too.  pcboys@163.com
Thanks  & Regards

maxclio, 2019-02-17:
It's Nbx_light_Pack_Active. When I set it to 0, the warning got dissappeared :)

closed 2019-02-17 by maxclio