cedricp/ddt4all #110: DeveloperSide Display Mode Failure

zatropin, 2018-11-13:
[image]
I pressed the marked key in the picture and the screen was black. Music and sound are working but there is no image. how do i fix it, ddt4all is suitable for this process.

Ddt4all> MeganeIV> Audio> RadNav 2.x v2.5> Configuration> RadNav Configuration Part-I> Screen Oriantation> Landscape     **COULDN'T WORK**
[image]

Ddt4all> MeganeIV> Audio> RadNav 2.x v2.5> Diagnosis> Buttons     
**2 sesc press  Button 1-2-3-4 active  Dısplay active**
[image]

Ddt4all> MeganeIV> Audio> RadNav 2.x v2.5> Configuration> Engineering Configuration         
[image]
**the values here are the same as my car**

[image]
**I write the values here, but it doesn't record, it goes back to the original original. results black screen unfortunately**

I'm waiting for your help. NOTE: my OBD device has no working problem

cedricp, 2018-11-13:
At first sight, I can see you're not using the good XML file, your ECU is detected as a close version but not the exact version (colored in red). You should not use such a file for writing.

zatropin, 2018-11-13:
ddt4all-installer.exe. (09.09.2018) I am using these files and parameters. Yes red color. should use older version parameters.

cedricp, 2018-11-13:
I'm talking about database file, not the application.

zatropin, 2018-11-13:
so the ecu list

cedricp, 2018-11-13:
Yes, ECU database. If the application cannot find the exact version, it gives you the closest version.

zatropin, 2018-11-13:
Well, should I use the older version, or should I search for a newer version? for ecu

shrlnm, 2018-11-13:
Did you reset r-link2 after changes by long press on power button?

shrlnm, 2018-11-13:
If yes, then:
- turn on the logs,
- open every screen
- share ecu* logs

KarelSvo, 2018-11-13:
>Yes red color. 

Do not panic, the 2180 - software versions in the ecu database are not constantly updated.

zatropin, 2018-11-13:
yes I have reset the rlink power button for 10-15 seconds and reset it. then shut down the car for 3 minutes, then run again. there was nothing changing.

zatropin, 2018-11-13:
If yes, then:

turn on the logs,
open every screen
share ecu* logs

[image]

are you referring to the values ​​in the picture

zatropin, 2018-11-13:
ok I understand, I will share the text file as soon as possible ...

Ddt4all> MeganeIV> Audio> RadNav 2.x v2.5> Configuration> Engineering Configuration
 I don't have the right parameters for the 7 "screen, I'll try the parameters in the picture.

shrlnm, 2018-11-13:
Just walk thru all screens. I'll try to find right  parameters

KarelSvo, 2018-11-13:
> I need text file "ddt4all/logs/ecu_ddt"

and the file "screens.txt"

shrlnm, 2018-11-13:
Main diff between MID and HIGH is as follows

`< MID  7"
> HIGH 8.7"

------------------------------
RadNav Configuration 2.0.Screen Orientation
< 0x00
> 0x01
------------------------------
CP_Display.PixelClockFrequency
< 0x0ad7
> 0x11c5
------------------------------
CP_Display.PixelClockEdgeSelect
< 0x01
> 0x00
------------------------------
CP_Display.VerticalActiveArea
< 0x01e0
> 0x0300
------------------------------
CP_Display.VerticalPeriod
< 0x01f4
> 0x0328
------------------------------
CP_Display.VerticalBackPorch
< 0x08
> 0x13
------------------------------
CP_Display.VerticalFrontPorch
< 0x08
> 0x13
------------------------------
CP_Display.VerticalSyncLPulseWidth
< 0x04
> 0x02
------------------------------
CP_Display.HorizontalActiveArea
< 0x0320
> 0x0400
------------------------------
CP_Display.HorizontalPeriod
< 0x039d  
> 0x0450
------------------------------
CP_Display.HorizontalBackPorch
< 0x05
> 0x1e
------------------------------
CP_Display.HorizontalFrontPorch
< 0x6e
> 0x1e
------------------------------
CP_Display.HorizontalSyncLPulseWidth
< 0x0a
> 0x14

`
except orientation all these parameters are on screen "Configuration / Engineering configuration"

for MID 
[image]

for HIGH
[image]

After changing any parameters you have to reload the device by long (10 seconds) press on power button

If it doesn't helps then collect and share your log file "ddt4all/logs/ecu_ddt"

shrlnm, 2018-11-13:
> I write the values here, but it doesn't record, it goes back to the original original. results black screen unfortunately

Then I also need "ddt4all/logs/elm_ddt" file. Probably the problem is in sending long commands to the CAN bus.

shrlnm, 2018-11-13:
For switching back to MID we have to send command

2E230301F404080801E03C0000039D000A0005006E03200AD7022C006967

zatropin, 2018-11-14:
THANKS DUDE....
yes it was finally. 
I entered no other value, because it was the same as the sample values.
I've done this before, but it has never been.
now the only difference is that you first did your own ECU scan of the DDT4AL tool.
The scan result from RadNav 2.x v2.5 menu was overwritten.
then I did the reset process.
and the screen arrived.
but slightly distorted, stripes.
Again, I went to the menu in the DEVELOPERSİDE menu and press the MIN key.
The display turns on and off automatically. 
and everything is perfect 🤩

closed 2018-11-14 by zatropin

mitakbas, 2019-03-24:
I solved the problem, I entered the following values in the central panel-display parameters 1 Display.VerticalPeriod 525 line
2 Display.VerticalSyncLPulseWidth 2 line
3 Display.VerticalBackPorch 8 line
4 Display.VerticalFrontPorch 35 line
5 Display.VerticalActiveArea 480 line
6 Display.FrameRate 60 Hz
7 Display.PhaseDifference 0 PCLK
8 Display.HorizontalPeriod 1056 PCLK
9 Display.HorizontalSyncLPulseWidth 128 PCLK
10 Display.HorizontalBackPorch 10 PCLK
11 Display.HorizontalFrontPorch 118 PCLK
12 Display.HorizontalActiveArea 800 PCLK
13 display.pixelclockfrequency 3333mhz
14 display.touchscreenreset Present
15 display.pixelclockedgeselect Present
16 Display.I2CPassThroughAll Not Present
17 display.deserializerı2c 2c
18 display.scgcontrol 00
19 display.touchscreencontrollerı2cboot 69
20 display.touchscreencontrollerı2coperating 67
After resetting the image came side by side, then the developer entered the mode and set the image to mid adjust everything

letrasluis2, 2024-04-03:
> [image] I pressed the marked key in the picture and the screen was black. Music and sound are working but there is no image. how do i fix it, ddt4all is suitable for this process.
>
> Ddt4all> [...]

Hello,

I have 8.7" and I clicked on MID - display is black.

what cable should I purchase to reserve this settings same as you did?

Cocobep, 2024-08-23:
> CẢM ƠN ANH BẠN....vâng cuối cùng thì nó đã.Tôi không nhập giá trị nào khác vì nó giống với giá trị mẫu.Tôi đã làm điều này trước đây, nhưng nó chưa bao gi.bây giờ sự khác biệt duy nhất là lần đầu tiên bạn tự quét ECU của công cụ DDT4AL.Kết quả quét từ menu RadNav 2.x v2.5 đã bị ghi đè.sau đó tôi thực hiện quá trình reset.và màn hình đã đến.nhưng hơi méo mó, sọc.Một lần nữa, tôi đã đi đến menu trong menu DEVELOPERSİDE và nhấn phím MIN.Màn hình tự động bật và tắt.và mọi thứ đều hoàn hảo 🤩

Can you let me know when you get a black screen?
How did you get into DEVELOPERSİDE?

Cocobep, 2024-08-23:
> > I write the values here, but it doesn't record, it goes back to the original original. results black screen unfortunately
> 
> Then I also need "ddt4all/logs/elm_ddt" file. Probably the problem is in sending long commands to the CAN bus.

Hello.

Please help me my espace 5 has a black screen.

I don't know where to fix the error.

I can send you 3 files

ecu.ddt
elm tree.ddt
screen.txt

Can you help me see it?
thank so much