cedricp/ddt4all #114: Bluetooth issue

alikalfaoglu, 2018-12-08:
Having exactly the same issue (HFE Kadjar)

https://github.com/cedricp/ddt4all/issues/105
Since issue is closed, I had to open another one..

But I believe I can fix this with ddt4all since I had a bricked main screen & fixed it also opening hidden features. Could not resolve the bluetooth audio problem. While fixing main I screen, I tried to reset config settings of Audio ECU. I entered back almost all values from another kadjar ( same year, 2018).

I searched for any bluetooth related settings but couldn't find any. 

Media streaming is working fine but audio is not working. I cannot hear calls also seems like my microphone is not working so other party cannot hear me ( completely silent on IOS pairings, and robotic voice on Android pairing)

Any ideas? Or settings to look for?

Thanks for your help..

alikalfaoglu, 2018-12-08:
VIN : VF1RFE00660687317
Com : 5
ELM : bt obdlink lx

No problems on accessing obd. No freezing, nothing, app is working great. Just needed to know where to look at..

Thanks.

shrlnm, 2018-12-09:
https://cloud.mail.ru/public/EJPD/8mBCqA9HU
"Phone" parameters responsible for acoustic of BT paired devices
I made complete config for you, but you may comment unwanted commands

Please let me know when you download the script

For running unpack it to ddt4all folder and double-click on cmdr_radnav_7317_repair.py

shrlnm, 2018-12-09:
Please send me the logs If you will run this script

alikalfaoglu, 2018-12-09:
Downloaded the script and will execute commenting out all except phone acoustic. I did not get what SPVR and whole car acoustic is standing for. If phone acoustic is not enough then I'll execute those also.

But the rest is already working I guess/hope. :) 

I'll send you the logs after executing the script.

Thanks for your help..

shrlnm, 2018-12-09:
I suppose that SPVR is something related to Voice Recognition. 
Car acoustic it is fine tuning for all channels of acoustic system in accordance to your car body geometry and materials (gain, equalisation, reverberation and so on)

shrlnm, 2018-12-09:
Ok. Thank you for logs

didoisp, 2019-11-30:
Hello, 
I had same problem with Megane IV, may you help me ?

shrlnm, 2019-11-30:
@didoisp could you please describe what kind of problem do you have?

didoisp, 2019-11-30:
After  Ecu reset I can't hear voice on car hands-free and other side can't hear what I am saying. Media streaming is working fine

shrlnm, 2019-12-01:
@didoisp Ok. First try to implement one of two acoustic macro
https://gitlab.com/py_ren/pyren/tree/master/pyren/macro/rlink2
rlink2_acoustic_3_and_7.cmd is for LGU v3 and v7
rlink2_acoustic_8.cmd for LGU v8 and probably v9
Instruction is here
https://gitlab.com/py_ren/pyren/wikis/how%20to%20rlink2%20black%20screen
If it doesn't helps then let me know your e-mail

didoisp, 2019-12-01:
@shrlnm Thank You very much, this macro solved the problem

Sv22bari, 2019-12-14:
Hi, I have this provblem After R-Link2 replacement that I have configured. 
Now only this problem with bluetooth Is remained suspended. I can reproduce media with bluetooth but I have no call audio. 
With pyren as mentioned above I do not solve because pyren not work. The app Will close when I push Macro Button. Let me know please

shrlnm, 2019-12-15:
> The app Will close when I push Macro Button.

What OS do you use? Which error message the pyren shows in the command line tool when is closing?

Sv22bari, 2019-12-15:
> > The app Will close when I push Macro Button.
> 
> What OS do you use? Which error message the pyren shows in the command line tool when is closing?

Windows 10 and CMD telo me:
Openning file: ./macro/init.txt
Openning ELM 

and close all

shrlnm, 2019-12-15:
@Sv22bari check that you selected correct ELM port in launcher and the field "Other options" is empty

Sv22bari, 2019-12-15:
Is the correct ELM Port because in DDT and Pyren mode work and Scan all ECUs.

shrlnm, 2019-12-15:
Do you have the latest version of pyren?

Sv22bari, 2019-12-15:
> Do you have the latest version of pyren?

Yes, I think. 
I hv downloaded it from the link (istruction) posted above in this discussion.

Sv22bari, 2019-12-16:
Here a new try with other Windows 10 pc. 
[image]

shrlnm, 2019-12-19:
@Sv22bari try to comment line 802 in mod_term.py

viktk, 2020-08-28:
Hi, thanks for the work you do, it's crazy ! 

So, i have the same problem as "didoisp" and "Sv22bari", and I followed the steps to try the 2 macro files v7 and v8 but it doesn't work.

Pyren is running fine and the macro v7 and v8 show me errors and restart the rlink only.

PS: After changing the Ecu Rlink and reset, I updated from v7 to v8 and reprogrammed myself but my ELM237 cannot read the ecu for acoustic parameters, do you think it comes from my elm? (ELM237 USB V1.5 chip 25K8)

Can you help me please ?

Thanks everyone!

AlMeu, 2020-10-26:
Hy guys 
I have the same problem with Bluetooth after downgrade to v7 a R-Link2 with 8.7" screen.
I upgrade back to v8 but not getting back Bluetooth functionality.
I try load the the macro with acoustic paramaters for v8 but is getting error telling that the parameters are not correct.
There is any chance to fix this Bluetooth?
May a have all acoustic parameters from a similar unit to test?

Thanks

viktk, 2020-10-26:
> Hy les gars,
> j'ai le même problème avec Bluetooth après avoir rétrogradé à la v7 un R-Link2 avec un écran de 8,7 ".
> Je reviens à la v8 mais je ne récupère pas la fonctionnalité Bluetooth.
> J'essaye de charger la macro avec des paramètres acoustiques pour la v8 mais j'obtiens une indication d'erreur que les paramètres ne sont pas corrects.
> Il y a une chance de réparer ce Bluetooth?
> Peut-on avoir tous les paramètres acoustiques d'une unité similaire à tester?
> Je mentionne que j'utilise OBD Link SX.
> 
> Merci

Hi, I had the same problem and it got fixed.
With my Micom 5615 configuration, I followed this procedure :
-downgrade to version 7
-apply macro 3/7
-when restarting, you go to system information and you do a reset

From there everything works for me, then I upgrade v8, the bluetooth is ok, and I upgrade v9

AlMeu, 2020-10-26:
> Hi, I had the same problem and it got fixed.
> With my Micom 5615 configuration, I followed this procedure :
> -downgrade to version 7
> -apply macro 3/7
> -when restarting, you go to system information and [...]

I tried as you said but not succes. When apply the SPVR parameters I have got WRONG RESPONCE at every command.

```
Opening ELM
Checking port speed: 115200
Start COM speed:  115200
7A:7E0:ELM# # Phone_acoustic and SPVR for RadNav 3.3 and 7.0
7A:7E0:ELM# $addr = 13
13:712:ELM#
13:712:ELM# can500  # init can macro
AT WS   ELM327 v1.3a  >
AT E1 OK  >
AT S0 OK  >
AT H0 OK  >
AT L0 OK  >
AT AL OK  >
AT CAF0 OK  >
AT CFC0 OK  >
AT SH 712 OK  >
AT CRA 732 OK  >
AT FC SH 712 OK  >
AT FC SD 30 00 00 OK  >
AT FC SM 1 OK  >
AT ST FF OK  >
AT AT 0 OK  >
AT SP 6 OK  >
AT AT 1 OK  >
13:712:CAN500#
13:712:CAN500# delay 2
13:712:CAN500#
13:712:CAN500# 1003
50 03 00 32 01 F4
# delay: 2
13:712:CAN500#
13:712:CAN500# # SPVR
13:712:CAN500# 2E213105B8000122030000000000000002000000000000000042534420535345204261736963205369676
E616C70726F63657373696E672044617461204352433A3058393931363830373720000000000000000000000000000000000
200000000000A0D0A10000000000000000C0000000B0000000100000001000000010000000000000000000000D00000001E1
00D0A5353455F4D41494E5F436F6E66696700380000000606060606030403030303030303030303030304030303030403030
303030303030303030303030303030303030306060303030603030303FB01000100401F80000000010000000000000002000
0000200000002000000020000000100000000000000010000000000000001000000010000000100000003000000000000000
1000000010000000100000003000000000000000100000001000000000000000200000001000000010000000100000001000
0000200000000000000010000000000000001000000010000000200000002000000000000000100000001000000020000000
10000000100A00F010000000100000000000000000001000000010000000100000002000000000A0D0AC5010000000A0D0A0
F00000000000000040000000B000000010000000100000001000000000000000000000018000000030F0D0A5353455F41474
35F436F6E666967000C000000050505050505050505050505480000A8FD00001CF39001E8032003E80374FF74FFC800C8000
00A0D0AB4090000000A0D0A1300000000000000010000000B00000001000000010000000100000000000000000000000C000
00062100D0A5353455F414E414C5F4D6963436F6E66696700040000000505030314792879280000000000000000000A0D0A4
2010000000A0D0A1300000000000000010000000B00000001000000010000000100000000000000000000000C00000066100
D0A5353455F414E414C5F526566436F6E66696700040000000505030314007D007D00000000FA000000000A0D0AF40100000
00A0D0A0E000000000000000E0000000B000000010000000100000001000000000000000000000008000000D10E0D0A53534
55F4E525F436F6E6669670004000000050505051818FC0000112B73FF000A0D0AC2020000000A0D0A1000000000000000010
000000B00000001000000010000000100000000000000000000001E000000900F0D0A5353455F4E525F4E66436F6E6669670
00F0000000505050505050505050505050505055A18FCF4011405D4FE00000000000001000000000064000000881C16FD30F
8000A0D0A38070000000A0D0A1500000000000000000000000200000001000000010000001C0000000000000000000000700
00000DA110D0A5353455F524543565F46696C746572466C6F617400000000400000D0410000A040D36A783F378EECBFF3C66
13F861BECBF6A175B3FAB78833F7A53E1BF11AA683F7A53E1BF899B6F3F48C37D3F0343C6BF77DC6C3F0343C6BFBE9F6A3FE
4DA843F4A43CDBE1843393F4A43CDBE23F9423FF6EE8B3F4017DD3E50195F3EDECB053F1591613E000A0D0A22310000000A0
D0A1300000000000000040000000B00000001000000010000000100000000000000000000001800000037100D0A5353455F5
24543565F414743436F6E666967000C0000000505050505050505050505054800000000000000005802E8032003E80374FF7
4FFC800C800000A0D0AC9060000000A0D0A1100000000000000020000000B000000010000000100000001000000000000000
000000018000000CE0F0D0A5353455F53594E54485F436F6E6669670007000000030303030305052001000000FF7F0000010
0000064000000F2060000A00FD007000A0D0A62040000000A0D0A12000000000000000000000005000000010000000100000
00F00000000000000000000001E00000014100D0A5353455F53594E54485F45514E6F646573002800FA001AFC5E01C800640
[... 44 lines cut ...]
D3F349DC5BFB01E473F349DC5BF0D35623F779D613F37AA3FBF986B393F37AA3FBFFF081B3FF12EC73FA66429BF444FA23EA
A25223D72F92F3E000A0D0A2B2F0000000A0D0A1300000000000000040000000B00000001000000010000000100000000000
000000000001800000037100D0A5353455F524543565F414743436F6E666967000C000000050505050505050505050505485
802A8FD000000005802E8032003E80374FF74FFC800C800000A0D0AC8080000000A0D0A1100000000000000020000000B000
000010000000100000001000000000000000000000018000000CE0F0D0A5353455F53594E54485F436F6E666967000700000
0030303030305052001000000FF7F00000100000064000000F2060000A00FD007000A0D0A62040000000A0D0A12000000000
00000000000000500000001000000010000000F00000000000000000000001E00000014100D0A5353455F53594E54485F455
14E6F646573002800FA0018FC5E01C8006600F401C80018FC1C02C8002D01C4095802F801000A0D0AC809000099999999999
9999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999
9999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999
9999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999
9999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999
999999999999999999999
WRONG RESPONSE
# delay: 2
13:712:CAN500#
13:712:CAN500# # reload rlink2
13:712:CAN500# 1101
:7F:NR: Service Not Supported In Active Session
# delay: 2
13:712:CAN500#
13:712:CAN500# exit

C:\PyREN>
```

viktk, 2020-10-27:
> > Hi, I had the same problem and it got fixed.
> > With my Micom 5615 configuration, I followed this procedure :
> > -downgrade to version 7
> > -apply macro 3/7
> > -when restarting, you go to system information and you do a reset
> > From there everything works for me, then I upgrade v8, the bluetooth is ok, and I upgrade v9
> 
> I tried as you said but not success. When apply the SPVR parameters I have got WRONG RESPONCE at every command.
> 
> ```
> Opening ELM
> Checking port speed: 115200
> Start COM speed:  115200
> 7A:7E0:ELM# # Phone_acoustic and SPVR for RadNav 3.3 and 7.0
> 7A:7E0:ELM# $addr = 13
> 13:712:ELM#
> 13:712:ELM# can500  # init can macro
> AT WS   ELM327 v1.3a  >
> AT E1 OK  >
> AT S0 OK  >
> AT H0 OK  >
> AT L0 OK  >
> AT AL OK  >
> AT CAF0 OK  >
> AT CFC0 OK  >
> AT SH 712 OK  >
> AT CRA 732 OK  >
> AT FC SH 712 OK  >
> AT FC SD 30 00 00 OK  >
> AT FC SM 1 OK  >
> AT ST FF OK  >
> AT AT 0 OK  >
> AT SP 6 OK  >
> AT AT 1 OK  >
> 13:712:CAN500#
> 13:712:CAN500# delay 2
> 13:712:CAN500#
> 13:712:CAN500# 1003
> 50 03 00 32 01 F4
> # delay: 2
> 13:712:CAN500#
> 13:712:CAN500# # SPVR
> 13:712:CAN500# 2E213105B8000122030000000000000002000000000000000042534420535345204261736963205369676
> E616C70726F63657373696E672044617461204352433A3058393931363830373720000000000000000000000000000000000
> 200000000000A0D0A10000000000000000C0000000B0000000100000001000000010000000000000000000000D00000001E1
> 00D0A5353455F4D41494E5F436F6E66696700380000000606060606030403030303030303030303030304030303030403030
> 303030303030303030303030303030303030306060303030603030303FB01000100401F80000000010000000000000002000
> 0000200000002000000020000000100000000000000010000000000000001000000010000000100000003000000000000000
> 1000000010000000100000003000000000000000100000001000000000000000200000001000000010000000100000001000
> 0000200000000000000010000000000000001000000010000000200000002000000000000000100000001000000020000000
> 10000000100A00F010000000100000000000000000001000000010000000100000002000000000A0D0AC5010000000A0D0A0
> F00000000000000040000000B000000010000000100000001000000000000000000000018000000030F0D0A5353455F41474
> 35F436F6E666967000C000000050505050505050505050505480000A8FD00001CF39001E8032003E80374FF74FFC800C8000
> 00A0D0AB4090000000A0D0A1300000000000000010000000B00000001000000010000000100000000000000000000000C000
> 00062100D0A5353455F414E414C5F4D6963436F6E66696700040000000505030314792879280000000000000000000A0D0A4
> 2010000000A0D0A1300000000000000010000000B00000001000000010000000100000000000000000000000C00000066100
> D0A5353455F414E414C5F526566436F6E66696700040000000505030314007D007D00000000FA000000000A0D0AF40100000
> 00A0D0A0E000000000000000E0000000B000000010000000100000001000000000000000000000008000000D10E0D0A53534
> 55F4E525F436F6E6669670004000000050505051818FC0000112B73FF000A0D0AC2020000000A0D0A1000000000000000010
> 000000B00000001000000010000000100000000000000000000001E000000900F0D0A5353455F4E525F4E66436F6E6669670
> 00F0000000505050505050505050505050505055A18FCF4011405D4FE00000000000001000000000064000000881C16FD30F
> 8000A0D0A38070000000A0D0A1500000000000000000000000200000001000000010000001C0000000000000000000000700
> 00000DA110D0A5353455F524543565F46696C746572466C6F617400000000400000D0410000A040D36A783F378EECBFF3C66
> 13F861BECBF6A175B3FAB78833F7A53E1BF11AA683F7A53E1BF899B6F3F48C37D3F0343C6BF77DC6C3F0343C6BFBE9F6A3FE
> 4DA843F4A43CDBE1843393F4A43CDBE23F9423FF6EE8B3F4017DD3E50195F3EDECB053F1591613E000A0D0A22310000000A0
> D0A1300000000000000040000000B00000001000000010000000100000000000000000000001800000037100D0A5353455F5
> 24543565F414743436F6E666967000C0000000505050505050505050505054800000000000000005802E8032003E80374FF7
> 4FFC800C800000A0D0AC9060000000A0D0A1100000000000000020000000B000000010000000100000001000000000000000
> 000000018000000CE0F0D0A5353455F53594E54485F436F6E6669670007000000030303030305052001000000FF7F0000010
> 0000064000000F2060000A00FD007000A0D0A62040000000A0D0A12000000000000000000000005000000010000000100000
[... 47 lines cut ...]
> 000000000001800000037100D0A5353455F524543565F414743436F6E666967000C000000050505050505050505050505485
> 802A8FD000000005802E8032003E80374FF74FFC800C800000A0D0AC8080000000A0D0A1100000000000000020000000B000
> 000010000000100000001000000000000000000000018000000CE0F0D0A5353455F53594E54485F436F6E666967000700000
> 0030303030305052001000000FF7F00000100000064000000F2060000A00FD007000A0D0A62040000000A0D0A12000000000
> 00000000000000500000001000000010000000F00000000000000000000001E00000014100D0A5353455F53594E54485F455
> 14E6F646573002800FA0018FC5E01C8006600F401C80018FC1C02C8002D01C4095802F801000A0D0AC809000099999999999
> 9999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999
> 9999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999
> 9999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999
> 9999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999999
> 999999999999999999999
> WRONG RESPONSE
> # delay: 2
> 13:712:CAN500#
> 13:712:CAN500# # reload rlink2
> 13:712:CAN500# 1101
> :7F:NR: Service Not Supported In Active Session
> # delay: 2
> 13:712:CAN500#
> 13:712:CAN500# exit
> 
> C:\PyREN>
> ```

I had the same errors with OBD Link SX, try with the ELM327 v1.5 USB PIC18F25K80. For me, it worked. Good luck!

AlMeu, 2020-10-27:
I have also ELM327 v1.4 USB and tested also with this, but I have got same response from unit.
What you are telling me that ELM327 is better then OBD Link SX. As I knew was contrary, OBD Link works with faster speed 115200 and can manage long strings like SPVR and with ELM327 works with lower speed and many times not even display the Acoustic parameters. 
Which is the difference between ELM327 v1.4 and v1.5? Should I search for it?

alexal100, 2021-04-16:
After Ecu reset I can't hear voice on my Kadjar  hands-free. Media streaming is working fine. Can someone help me please!

Lauddt94, 2023-04-29:
> ...link removed... "Phone" parameters responsible for acoustic of BT paired devices I made complete config for you, but you may comment unwanted commands
> 
> Please let me know when you download the script
> 
> For running unpack it to ddt4all folder and double-click on cmdr_radnav_7317_repair.py

@shrlnm Is it possible to share this script? I think I have the same issue, calling through bluetooth is not possible and I have issues with calling with Apple Carplay. Can you please help?

shrlnm, 2023-04-29:
@Lauddt94 Now you may do the script by yourself. 
read https://www.drive2.ru/b/643019689764615344/

Lauddt94, 2023-04-30:
> @Lauddt94 Now you may do the script by yourself. read https://www.drive2.ru/b/643019689764615344/

@shrlnm Thanks! That looks like a solid solution! I gave my information to the pyren bot but there is no respone. Is it normal that a response will take some time?

Lauddt94, 2023-04-30:
> @Lauddt94 Now you may do the script by yourself. read https://www.drive2.ru/b/643019689764615344/

@shrlnm unfortunately it is not working, I did everything mentioned on the page and it worked accordingly. But it did not solve the audio problems and it turned my portrait screen settings into landscape settings. I fixed the landscape back to portrait but I still need some help with the audio. Can you help?

shrlnm, 2023-04-30:
@Lauddt94 what is your current LGU version and version of micom?

Lauddt94, 2023-04-30:
> @Lauddt94 what is your current LGU version and version of micom?
@shrlnm 

LGU = 9.12.178.756131, and where can I find the micom version?

shrlnm, 2023-04-30:
This version was installed on a plant or you upgraded it yourself from version 7?

Lauddt94, 2023-04-30:
The plant updated from V3 to V7 and I did V8 and later V9 myself. I had never any issues with that, the problems started when I changed the 7'' display to the 8.7'' display. Before this change V9 worked correct.

shrlnm, 2023-04-30:
@Lauddt94 For solving this issue you have to downgrade to version 7 first and only then implement a macro. (I wrote about it in my article). I had absolutely the same issue twice. It related to incompatibility your micom version intended for version 7 with LGU 8 and 9. 
After you did an upgrade from 7 to 8/9 you can not change anything thru the CAN it will always lead to problems with hands free.

So, you need to downgrade to v7, then implement the macro, then change the screen mode and then upgrade to v8/9. After that you can't change anything thru the CAN

Lauddt94, 2023-04-30:
@shrlnm Thanks a lot! I will work that through. Do you know how to downgrade the system? I have a v7 file but the system won’t see it.

shrlnm, 2023-04-30:
@Lauddt94 Yes. The image have to be prepared especially. I used the image from https://www.msieurlolo.fr/forum/index.php?topic=57457.0
Also, it may be need to virginize the rl2 after downgrade to v7 and before implementing the macro

Lauddt94, 2023-04-30:
@shrlnm I will try that one. How do you virginize the RL2 and what does that? Sounds quite heavy :)

shrlnm, 2023-04-30:
@Lauddt94 it is a button in a developer menu
https://www.youtube.com/watch?v=L_0RBC3nIPE
and yes, it sounds quite dangerous). After pressing this button your screen will become black until you implement the macro. You have to be sure that your ELM is good enough for implementing all commands from macro. You may check it before virginizing by implementing the macro and checking the log.

Lauddt94, 2023-04-30:
@shrlnm Is there another place where I can find the V7 image? Somehow I cannot register on that site and download the link

shrlnm, 2023-04-30:
https://mega.nz/file/HM9mBIxR#kBhG5zDLx_UudPsVvfRML1p1p76-khHuLBCBTXARjIo

Lauddt94, 2023-04-30:
@shrlnm To exclude some errors, I have in Pyren not have the following option: 15: ACI : Auto Config Info. I have the screens below to get the information. 
[image]
[image]

Can you confim that: /vin VF1RFB00258186469 76:281151240R is the right command for the pyren-bot?

shrlnm, 2023-05-01:
@Lauddt94 Yes, for RL2 it is ID028, but check it after downgrade to v7

Lauddt94, 2023-05-01:
@shrlnm So I did everything as described. 

- Downgraded to V7
- Applied the Macro, 
- No change, still echo and audio problems with calling
- Tried the acoustic marco, still no change
- Virgin mode
- Applied Marco

And now after applying the Macro my screen stays black. Can you please help?

shrlnm, 2023-05-01:
@Lauddt94 Don't wary. First check the logs. I suppose the problem is related to HIGH screen you changed.
First try to use macros pyren3/macro/rlink2/rlink2-HIGH1-screen.cmd  or  pyren3/macro/rlink2/rlink2-HIGH2-screen.cmd
Then send me your current dump

shrlnm, 2023-05-01:
@Lauddt94 did you uncommented #1101 in the end of macro for reloading the box?

Lauddt94, 2023-05-01:
@shrlnm how do you uncommendt that line? Removing the #?

shrlnm, 2023-05-01:
@Lauddt94 yes. Just remove #

shrlnm, 2023-05-01:
But you may also reload the box with long press on power button

Lauddt94, 2023-05-01:
@shrlnm So the screen is working again but the audio quality is still really bad on the other side of the connection. I tried the V7 acoustic macro but this made it even worse. Do you know what to do? Do you need a dump?

shrlnm, 2023-05-01:
@Lauddt94
Please send me your current dump.

Lauddt94, 2023-05-01:
[1682927259_11547.txt](https://github.com/cedricp/ddt4all/files/11365094/1682927259_11547.txt)

@shrlnm Is it this one? It is the latest file in the pyren/dumps map

shrlnm, 2023-05-01:
@Lauddt94 Did you checked the log? Does all commands was implemented correctly?

Lauddt94, 2023-05-01:
[ecu_log.txt](https://github.com/cedricp/ddt4all/files/11365119/ecu_log.txt)
[elm_log.txt](https://github.com/cedricp/ddt4all/files/11365120/elm_log.txt)

@shrlnm You can check yourself but I saw no errors or so??

shrlnm, 2023-05-01:
@Lauddt94 
https://cloud.mail.ru/public/4xqy/ynj3TAG35

shrlnm, 2023-05-01:
It is already for a big screen

Lauddt94, 2023-05-01:
@shrlnm So I should run that one?

Lauddt94, 2023-05-01:
@shrlnm i tried your macro but still no succes. At the end of the macro there was an error.
[image]

:22:NR: Conditions Not Correct Or Request Sequence Error

maybe that has something to do with it?

Lauddt94, 2023-05-01:
@shrlnm do you know if there is anything left to try?

[22 more replies]