cedricp/ddt4all #115: Connection timeout

solidv, 2018-12-10:
Hi guys, trying to install android auto in a clio iv phase 2, i get this in the elm test: 

v1.0 ATZ [OK]
v1.0 ATE1 [OK]
v1.0 ATL0 [OK]
v1.0 ATI [OK]
v1.0 AT@1 [OK]
v1.0 ATAL [OK]
v1.0 ATBD [OK]
v1.0 ATBI [OK]
v1.0 ATCAF0 [OK]
v1.0 ATCFC1 [OK]
v1.0 ATCP 01 [OK]
v1.0 ATCS [OK]
v1.0 ATCV 1250 [OK]
v1.0 ATD [OK]
v1.0 ATDP [OK]
v1.0 ATDPN [OK]
v1.0 ATH0 [OK]
v1.0 ATI [OK]
v1.0 ATIB 10 [OK]
v1.0 ATIB 96 [OK]
v1.0 ATL1 [OK]
v1.0 ATM0 [OK]
v1.0 ATNL [OK]
v1.0 ATPC [OK]
v1.0 ATR1 [OK]
v1.0 ATRV [OK]
v1.0 ATSH 001122 [OK]
v1.0 ATSH 012 [OK]
v1.0 ATSP A6 [OK]
v1.0 ATSP 6 [OK]
v1.0 ATCM 123 [OK]
v1.0 ATCM 12345678 [OK]
v1.0 ATCF 123 [OK]
v1.0 ATCF 12345678 [OK]
v1.0 ATST FF [OK]
v1.0 ATSW 96 [OK]
v1.0 ATSW 34 [OK]
v1.0 ATTP A6 [OK]
v1.0 ATTP 6 [OK]
v1.0 ATWM 817AF13E [OK]
v1.0 ATWS [OK]
v1.1 ATFC SD 300000 [OK]
v1.1 ATFC SH 012 [OK]
v1.1 ATFC SH 00112233 [OK]
v1.1 ATFC SM 1 [OK]
v1.1 ATPP FF OFF [OK]
v1.1 ATPP FF ON [OK]
v1.1 ATPPS [OK]
v1.2 ATAR [OK]
v1.2 ATAT1 [OK]
v1.2 ATIFR H [OK]
v1.2 ATIFR0 [OK]
v1.2 ATIIA 01 [OK]
v1.2 ATKW0 [OK]
v1.2 ATSR 01 [OK]
v1.3 ATCRA 012 [OK]
v1.3 ATCRA 01234567 [OK]
v1.3 ATD0 [OK]
v1.3 ATFE [OK]
v1.3 ATJE [OK]
v1.3 ATJS [OK]
v1.3 ATKW [OK]
v1.3 ATRA 01 [OK]
v1.3 ATS1 [OK]
v1.3 ATSP 00 [OK]
v1.3 ATV0 [OK]
v1.4 ATCV 0000 [OK]
v1.4 ATIB 48 [OK]
v1.4 ATIGN [OK]
v1.4 ATPB 01 23 [OK]
v1.4 ATRD [OK]
v1.4 ATSD 01 [OK]
v1.4 ATSI [OK]
v1.4 ATFI [OK]
v1.4 ATSS [OK]
v1.4 ATTA 12 [OK]
v1.4 ATCSM1 [FALHA]
v1.4 ATJHF1 [FALHA]
v1.4 ATJTM1 [FALHA]
v1.4 ATJTM5 [FALHA]
v1.4b ATCRA [FALHA]
STN TEST 1 STI [FALHA]
STN TEST 2 STDI [FALHA]
STN TEST 3 STPR [FALHA]
STN TEST 4 STPBRR [FALHA]
STN TEST 5 STPRS [FALHA]
Resultado: 76 bem sucedidos sobre 86
Versão ELM máxima:v1.4 

I can connect with ddt4all but after a i choose mfd 4.6, i get connection timeout with port closed in the console then disconnects from the software. tried to change connection timeout to 1000ms, same problem.
i leave the elm log file here: 
[elm_ddt.zip](https://github.com/cedricp/ddt4all/files/2662993/elm_ddt.zip)

Thank you

shrlnm, 2018-12-10:
Is it BT or WiFi ?
Try to open elm.py file in any text editor and change all substrings "AT WS" on "AT Z"
Also try to open any screen immediately after starting the program.
Your adapter shows strange behaviour

solidv, 2018-12-10:
> 
> 
> Is it BT or WiFi ?
> Try to open elm.py file in any text editor and change all substrings "AT WS" on "AT Z"
> Also try to open any screen immediately after starting the program.
> Your adapter shows strange behaviour

It´s wifi, going to try those steps.

shrlnm, 2018-12-10:
I came across such an adapter which has very short tcp session timeout. 
@cedricp  I suppose that reopening tcp session in init_can and in init_iso may solve such problem

solidv, 2018-12-10:
Already tried, didn´t work :(

shrlnm, 2018-12-10:
Could you please share new logs

shrlnm, 2018-12-10:
Now the log is other. 
I think that the MFD module is on the other CAN bus. 
Do you have a switch on elm?

solidv, 2018-12-10:
> 
> 
> Now the log is other.
> I think that the MFD module is on the other CAN bus.
> Do you have a switch on elm?

Yes i have, i tried with and witouth the switch

solidv, 2018-12-10:
Also tried to search can for ecus, without switch i found some, with switch i cant find anything. Should i find mfd with the switch on?

shrlnm, 2018-12-10:
MFD should be on canH="13" canL="12"

solidv, 2018-12-10:
I think the switch is fine

mtlsi, 2018-12-19:
I have a similar problem. ELM check give the same results as in the first post.
76 from 86 ok.
After that I get Port disconnected.
[elm_ddt.zip](https://github.com/cedricp/ddt4all/files/2696606/elm_ddt.zip)

cedricp, 2018-12-21:
Should be fixed in the last version, could you check so I can close this issue ?

mtlsi, 2018-12-21:
Which is the last version?
Because Windows installer is from 10th of November.

cedricp, 2018-12-21:
There's no release for that now, you must download the source code and replace it in the installation directory
https://github.com/cedricp/ddt4all/archive/master.zip

mtlsi, 2018-12-21:
Quick test on a NON Renault car showed it is now ok. 
No more port connection dropped.
Thank you.

mtlsi, 2018-12-26:
Tested on a Renault Kadjar, no port connection dropped.

Furtif, 2023-06-02:
this bug is fixed at https://github.com/cedricp/ddt4all/pull/779 if not please use discutions module https://github.com/cedricp/ddt4all/discussions

closed 2023-06-02 by Furtif