cedricp/ddt4all #119: Are there any experiences with klavkarr dongles from France?

pyromixer, 2018-12-16:
From France come the klavkarr OBD dongles. This company also knows the problem of fake China clones of the ELM327.
The dongles from France are based on version 2.1 and should be absolutely real.
The software and the app of which works very well and is visually great. Everything runs on Windows, Mac, iOS and Android.

The question is: Does the klavklarr OBD dongle work with ddt4all?

Does anyone have experiences?

[https://www.boutiqueobdfacile.com](https://www.boutiqueobdfacile.com)

cedricp, 2018-12-21:
As it's based on an ELM327, I'd say it should be OK. If you test it, please report your experience so I update the adapter compatibility list.

pyromixer, 2018-12-21:
The adapter is on the way to me. I will report :-)

pyromixer, 2018-12-21:
The best adapter I've ever tested!
I have the Klavkarr 210 (Expert Adapter). It is a Bluetooth AND USB adapter. The first Bluetooth adapter that works with iOS also works in my collection.

The version is ELM 2.1
[image]

It works smoothly with ddt4all. With Bluetooth or USB. Both work with ddt4all. With USB cable, the adapter is much faster than with the Bluetooth connection.

The original Klavkarr software EOBD Facile runs directly on macOS! There is also a Windows version and for Android and iOS an EOBD Facile App. The software is included in the price of the adapter. Likewise, a long and smooth USB cable is included.

Very good adapter! 5 stars from me :-)
And it comes from good old Europe. It is not China Ware!

[image]

In Attachment the logfiles in USB and Bluetooh Version.

[klavkarr210_usb_logs.zip](https://github.com/cedricp/ddt4all/files/2704425/klavkarr210_usb_logs.zip)
[klavkarr210_bluetooth_logs.zip](https://github.com/cedricp/ddt4all/files/2704426/klavkarr210_bluetooth_logs.zip)

pyromixer, 2018-12-21:
Here are additional logfiles with the ELM TEST on a USB cable only (without connection with a car)

[ELM Test Logs.zip](https://github.com/cedricp/ddt4all/files/2704589/ELM.Test.Logs.zip)

shrlnm, 2018-12-22:
Could you please try usb adapter on higher port rate? 115200, 230400 or even 500000? What response time it will show?
On 38400 it shows  not too good results. About 55 ms

<[0.063]03220109
<[0.062]03220118
<[0.047]03220204
<[0.063]03220203
<[0.062]03220117
<[0.062]03220122
<[0.047]03220106
<[0.064]03220121
<[0.047]03220109

Good adapters show about 15ms on 38400 and about 5-6ms on 230400 boud rate

For measuring you may modify a bit a __del__ function in elm.py. Add only one line

`    def __del__(self):
        try:
            if not self.sim_mode:
                print self.response_time
                self.port.write("ATZ\r")
        except:
            pass
`
After exiting application it will shows average response time among 10 last responses

HZx_QR25_China5&6_20180712 ECM
**0.034**0319187182
ddt4all-master $

There is 34 ms response time of my wifi adapter

pyromixer, 2018-12-22:
With best adapter I mean in this case: For **me**, the best adapter
Because: I want an adapter that works on all my devices (macOS, Windows, iOS etc.) And that works with ddt4all and other OBD software.

I have inserted the line but something is probably wrong.
[image]

shrlnm, 2018-12-22:
You have probably inserted code of function from the mail as one string. According to python syntax this function have to be written on 7 strings as in my message above.

I do not want to say that your adapter is bad. It is probably slow ECU. Try to read from different ecus. Usually Engine ecu is much faster then others.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

f86sabre, 2024-12-10:
Hi I have just bought the Klavkarr 210 (PLUS EDITION) and no user manual was included. Unfortunately their instruction videos are all in French with Englsh subtitles in black font which makes it harder to follow with reading and viewing. I know from their instruction videos that they include a user manual. I am at a loss as when I hooked the OBD to my car it kept telling me that "Device is not connected" although I went through the whole procedure to hook the OBD to the car. I need to obtain an English user manual in order to ascertain what is really going on. The green light remains flickering. I am using a mobile phone and tablet all of which the Bluetooth was activated and connected. Any help is appreciated. Thanks