cedricp/ddt4all #121: Cruise Control of Megane 3

buetgi, 2018-12-19:
Hey, did anyone successfully activated cruise control on a Megane 3? I couldn't find the relevate ECUs and unfortunately my frensh isn't the best so is there an option to switch the language of the option pages to another language?

KarelSvo, 2018-12-19:
TDB  - config generale RV/LV
ECM - CRUISE

cedricp, 2018-12-21:
Yes RV/LV is _regulateur de vitesse / limiteur de vitesse_ in french -> cruise control
There is no translation for these XML files, unfortunately.

KarelSvo, 2018-12-21:
Alternatively use PyREN with ddt database.
There are  configurations translated, with the CLIP language file.

aliguler9254, 2019-03-21:
In my Megane 3, the injection brain [ ECM ] DCM34-P4C8x_X95_L38_B32_V22 appears. How do you activate Cruise Control in this brain?

aliguler9254, 2019-03-21:
@cedricp In my Megane 3, the injection brain [ ECM ] DCM34-P4C8x_X95_L38_B32_V22 appears. How do you activate Cruise Control in this brain?

cedricp, 2019-03-21:
On my Megane 2, I did it in the Meter Cluster ECU,not in ECM try to find **cruise control** or **RV/LV** (régulateur/limiteur de vitesse)

sv3pjl, 2019-11-06:
i have a megane 2 Ph2 with ecu Tdb_BC85_EG_MG_84Phase2_Serie_rev6 Cluster/TDB is there anybody who managed to enable cruise control?
this is the image in french but i cannot figure out how to enable it
[image]

cedricp, 2019-11-06:
Prestation rv/lv-> avec

sv3pjl, 2019-11-06:
> Prestation rv/lv-> avec
> […](#)

thanks for reply. Finally i work this out. The right settings are :
 Prestation rv/lv-> avec
Loi d' avance rv/lv-> Loi d' avance appliquee

LasseSchonings, 2022-05-17:
Hi guys, does anyone of you know if i can activate cruise control on a Twingo II 1.2 litre petrol?

AndyIreland, 2022-05-17:
not sure whether this setting would or not and you might even have to enable other settings to get CC - but apart from all that you would surely need a CC/SL button in the car somewhere either on the dash or down by the gear shifter? 
[image]

LasseSchonings, 2022-05-18:
I have mounted the CC/SL button in the car, and the buttons on the steering wheel incl. the steering sensor which all was just plug and play.
I can only read the frames in the injection, i cant write anything. i Have the RenoLink 1.99. Which program do you use? DDT4AAL?

AndyIreland, 2022-05-18:
thats good you have all the switches etc in place - yep that picture is of DDT4ALL

luissilva1971, 2022-06-21:
Hum... i am studying the issue regarding my megane 3 SW 2010. From a dump i 've made with Pyren, my ECM is "DCM34_P4C8_X95_L38_B32_V22". On DDT2000/ DDT4all screen for this ECU, and on my "offline" screen, i don't find nothing "regarding RV/LV - Configuration option", just one screen where there is an indication that my vehicle is "without" CC
[image]

Does this mean i do not have any change to activate it, even if i add on/off button and new steering wheel with CC buttons?
TDB has the parameter ... "CLUSTER_X91_L43_L47_Conti_v2.1":
[image]

Please advise. 
thank you

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif