cedricp/ddt4all #123: R-Link2 errors in ddt4all

IfrimP, 2018-12-24:
Hello all!

I got some errors checking with ddt4all to my Talisman R-Link2 after a failed update and now I got a black screen. Attached two pictures. How can I solve?
Any help will be very appreciated. Thank you!
[image]
[image]

shrlnm, 2018-12-25:
Hello. The description is not clear. Could you describe the problem with more details? 
Did you entered DeveloperSide?
Did you pressed "Virgin" button or "HIGH" or "MID"?
Did it happens after official lgu upgrade?

IfrimP, 2018-12-25:
Actually was an update from 2.2. 19. 300 to 3.3. xxxxx version, that are on the air these days around Renault forums. After this update R-Link2 screen remained black. Sometimes R-Link try to boot but something not allow. Now I ask gently for a procedure to boot. Or reset, go back to 2.2.19.300 version, anything.

maydayanthem, 2018-12-25:
can you communicate via radnav with radio after starting the on/off button? ... normally switch on the on/off button on lcd, wait 1 minute and try to make connection via obd.
btw. the update was finished successfully? last restart (black screen) takes almost 2-2,5 minutes.. beware of that!!!

IfrimP, 2018-12-25:
Update  didn't finished successfully I suppose, because didn't became ON. After more than 15 minute I decided to swich OFF pressing touch button. 
Central panel is permanent black, the instrument panel try to change the color for a milisecond, think trieng to boot.
Now have checked with DDT4ALL and in Radnav configuration found central panel in Landscape position instead Portrait(8.7"). I don't know if could be a reason. Change applied but nothing happened, still black.

maydayanthem, 2018-12-25:
ok, execute this 
[image]

shrlnm, 2018-12-25:
> Change applied but nothing happened, still black.

After any changes you have to reboot radnav by long press on power button.
Let's try to check your configuration. 
1) enable logging.
2) connect to radnav by ddt4all
3) open every screen one by one
4) share file logs/elm_ddt

maydayanthem, 2018-12-25:
and did you make hard restart - 30 seconds holding on/off button?

IfrimP, 2018-12-25:
Yes.

maydayanthem, 2018-12-25:
i dont suppose it... engine was started when you were updating? didnt you open driver´s door?

IfrimP, 2018-12-25:
No. Done before, without problem, another two updates. Sistem one and Maps another.
I think the Update now was for another R-Link, with 7" landscape display.

maydayanthem, 2018-12-25:
no, there is only 1 (universal) type of update file for 7 or 8,7" display, the update process alone chooses how to configure system for mid or high...
if i good remember, while updating, there are also logs on usb,dont you have them??

IfrimP, 2018-12-25:
Yes. Two log file

[Upgrade0.log](https://github.com/cedricp/ddt4all/files/2709027/Upgrade0.log)
[Upgrade1.log](https://github.com/cedricp/ddt4all/files/2709029/Upgrade1.log)

shrlnm, 2018-12-25:
@IfrimP 
What com port are you using for elm? 
Give me your e-mail please

IfrimP, 2018-12-25:
serial usb elm

ifrimp@yahoo.com

alf81, 2018-12-27:
go back and install RLINK 2.2.19.300 with USB key.
Download here software : 
http://rvehconnecte.renault.com.frontcdn.lbn.fr/2-2-19-300/R-LINK.zip

Unzip and put folder in USB key (FAT32 format) , plug key on Talisman usb and accept update and WAIT.

alf81, 2018-12-27:
If update is not accepted by car use software UnpackR2 : 
https://www.msieurlolo.fr/forum/index.php?action=dlattach;topic=57569.0;attach=30069

This software work on official Rlink2 update .

IfrimP, 2018-12-27:
Thank you Alf81

But I have no permission to see the link. 

 
 

https://www.msieurlolo.fr/forum/index.php?action=dlattach;topic=57569.0;attach=30069

maydayanthem, 2018-12-27:
firstly, go to the shop and buy one QUALITY !!!! usb key or sd card, better usb with flashing led (when reading/writting)
it is highly probably your update crashed because of a bad file system (fat32) on your usb, probably cheep (and not so quality/demaged??) memory in the usb
then give the usb (with fw) in usb slot, wait and watch if the led start flashing, if it stops, try to find the position on your (black) LCD where could be the virtual button of update (YES) and wait, if the led start flashing again (hopefully reading the fw)
[image]
or on 8,7
[image]
of course, you can also enable logging on usb key

IfrimP, 2018-12-27:
Good to know the touch screen working when is black. Could you help me with the location of the YES button in a 8.7" display? Thanks a lot.

alf81, 2018-12-27:
http://www.reno.ro/R-Link-2-modificari-t316334-s270-p4802429.html#entry4802429

link here for software

maydayanthem, 2018-12-27:
this we dont know for sure - that the lcd is working, better said responding to commands - because we dont know how much the right code was uploaded to the radio... now you are in situation, that the only thing you can do is to try, try, try... and maybe some of try will be working... no one knows, we are now like blind
btw. you have got an email, did you answer it?

IfrimP, 2018-12-27:
Thank you all for help!

ftf32, 2019-01-06:
I have the same problem..
Anyway solution find that?

aqngelrock, 2019-01-07:
Hello everyone, I have the same problem too. After loading the softweare, r-link no longer starts, or better always in bootloop. Continue to restart. According to you it is possible with DDT4all or other, put r-link in recovery mode to reload the firmware ???

Dianter, 2019-01-07:
I enter Developer mode?
Press the "MID" button by mistake
and it went black
the screen that I have are 10
Someone who can help me.
I'm from Spain the music works

aqngelrock, 2019-01-07:
> ciao, accedo alla modalità sviluppatore? 
> Premi per errore il tasto "MID" 
> e diventa nero 
> lo schermo che ho 10 
> Qualcuno che può aiutarmi. 
> Vengo dalla Spagna, la musica funziona

It is not our same problem .. with ddt4all resolve ... send me your email that I send you the procedure. Do you have a 7 "or 8" screen?

Dianter, 2019-01-07:
it's a big screen link2 of a megane

aqngelrock, 2019-01-07:
[image]

[image]

Dianter, 2019-01-07:
Thank you from Spain

Dianter, 2019-01-07:
Hello, if I take it to the house, renaunlt sedarancuenta that is touched there is the car, this haun in warranty the obd2 in a couple of days I have it
written by the google translator

aqngelrock, 2019-01-07:
Oh yeah. every time you connect with obd it remains marked in the logs. but I do not know how much memory holds.

Dianter, 2019-01-07:
never entered with the obd
entered Developer mode and tap MID,
they sedaran account that touched ay

aqngelrock, 2019-01-07:
With only the developer mode, no trace remains. but now the only way you have to solve is to use ddt4all and it will remain in the logs

aqngelrock, 2019-01-07:
@shrlnm
you have no idea how to solve the problem origin of the discussion. I'm going crazy!!! thank you!!

Dianter, 2019-01-07:
better to touch the renault ddt4all

Dianter, 2019-01-07:
thank you very much for your attention

shrlnm, 2019-01-07:
If the problem fired after touching MID or HIGH button then I may try to help. I need the e-mail of "victim".
If it cyclic reboot then I afraid I can not do anything now. For repair from this state we have to understand how MICOM firmware works. MICOM is a module responsible for CAN conversation and it has own firmware. We do not know if it has some hidden functionality which may permit us to went the r-link into recovery mode or not. We also do not know how to interrupt the booting process and went the box into  recovery mode by buttons. It is definitely that the recovery partition exists in the r-link2 but I do not know how to boot into recovery partition.

aqngelrock, 2019-01-07:
I'm really screwed up, because my problem and even others are just the cyclic restart. I do not know where to bump my head anymore ...

Dianter, 2019-01-08:
I do not speak English translate with Google
He said it's better that the car take him to the renault dealer and that they touch there

kholer, 2019-01-08:
@shrlnm Hello. I have the same issue like Dianter. Can you help me?

kholer, 2019-01-08:
kholer1973@gmail.com. My screen is 7" size.

ftf32, 2019-01-08:
fatihdikici32@hotmail.com
Thank you for help

Arma1987, 2019-01-17:
@shrlnm  I have the same problem :( pushed High on Med screen can you help me ? gim503@wp.pl

ferkam, 2019-01-22:
i pushed MID HIGH screen in developer mode... Can you help me please ( fer_kam@hotmail.com )

Arma1987, 2019-01-23:
@shrlnm so is it possibile to get you help :(??

shrlnm, 2019-01-24:
@Arma1987 I have sent you e-mail on 18-jan at 08:09 (Moscow time). Please check your mail box or give another address.

Arma1987, 2019-01-24:
@shrlnm  didint get it can you send it to gim505@vp.pl

shrlnm, 2019-01-24:
@Arma1987 Done. Check it please

tanyanargile, 2019-01-27:
The screen is turned off (black screen) when updating from 2.2 to 3.3 The USB light does not work. I've tried every way I could not open including ddt4all. please help me. thanks. tanyanargile@hotmail.com
Sorry I Write with translate.

tanyanargile, 2019-01-27:
I guess the radio put itself in self-protection mode. Is it possible to open or install any software via DDT4ALL? My guarantee is over last week.

curseofeanor, 2019-01-29:
@shrlnm hello brother, i have similar problem. i had 2.2.19.300 and r links  opened up 40 - 45 seconds after i entered the car and i wanted to downgrade firmware  to old version. usb drive loading all of them and and after r link  wont start. 4 control lamb is blinking  sometimes  i have no 4 control car is magene. 2 weeks ago i pushed virgin mode a accidentally. car service fixed  it that but  current black screen different from virgin mode. can you help me for fix this :)

shrlnm, 2019-01-29:
I may help after virgin mode but I not sure that I may help in your case. Did you try autoconfiguration with Clip?

hpartovian, 2019-02-01:
i pushed mid high screen in developer mode... Can you help me please 
Hpartovian@gmail.com

aeonbg, 2019-02-03:
@shrlnm Hi, also pushed by accident high or mid button on Kadjar with 3.3 update done. Now - black screen with everything working (radio, bluetooth and so on). Can you help me, please? cvetomir_rusenov@yahoo.com

HagashenNaidu, 2019-02-08:
@shrlnm I did the same please send me fix. Hagashen.naidu@gmail.com

Thanks

shrlnm, 2019-02-08:
Sorry for external link. I tried made most simplest solution for such popular case. There are four known variants of R-Link2 screen configuration. Just choose one of them and wait a minute while r-link 2 is rebooting.  

https://gitlab.com/py_ren/pyren/wikis/how%20to%20rlink2%20black%20screen

kholer, 2019-02-17:
I can't run pyren in my computer. Launcher don't work

Jester80, 2019-03-02:
Hallo everybody. I’ve the same problem. 
During an official update (2.2.x.x to 3.3...) something went wrong.

Jester80, 2019-03-02:
[image]

traiannn, 2019-03-04:
Hello i have the MID problem rlink2 8.7",  whit black screen, please help me @shrlnm, my email adress is traian_micle@yahoo.com, thank you very much

[385 more replies]

closed 2023-06-02 by Furtif