cedricp/ddt4all #125: ELM Benchmark [ errors ]

michalzahor, 2019-01-02:
Hi,

I have few errors running ELM benchmark - is the only cause ELM Module or DDT4ALL SW can have also issue with that ? Should I just buy better one ELM ?

[image]
[image]

cedricp, 2019-01-02:
What are the last lines of the benchmark ?

michalzahor, 2019-01-02:
can reproduce test tomorrow , but every other test below the ones on the last image FAILED, and then last line is like 64 of 83 tests are OK.

cedricp, 2019-01-02:
The very last line tells you if it's a clone (_arm core detected_), if so, you are good to buy a compatible one

michalzahor, 2019-01-03:
Result of ELM Test - no ARM Clone detected

[image]