cedricp/ddt4all #126: ABS / ESP  - ACC deactivated by diag

michalzahor, 2019-01-02:
Hi,

I'm unable to change ACC to active in ABS / ESP .
[ using WinXP latest DDT4ALL version ]

Seems like data are writtencorrectly but the value is still "deactivated".
 
[image]

Is the error in the screenshot related to failed write operation ?

cedricp, 2019-01-02:
Look like an ELM clone issue. Could you send your log files and report result of ELM benchmark ?

michalzahor, 2019-01-02:
attaching logs, ELM bench. was described in another post

[ddt4all_logs.zip](https://github.com/cedricp/ddt4all/files/2722017/ddt4all_logs.zip)

michalzahor, 2019-01-02:
using Carista Bluetooth module

cedricp, 2019-01-02:
The ELM seems to work, are you sure you're using the right XML file for your ECU (did you scan your vehicle ?)

michalzahor, 2019-01-02:
Yes, I did.

It shows few red info that xml is not present for ECU and about 2 are not fully ok. But ABS was not mentioned there.

I'll post result of the scan tomorrow.

cedricp, 2019-01-02:
Try to increase CAN timeout in the bottom of the window, and be careful when working on the ABS ECU of your vehicle, safety before all :)

michalzahor, 2019-01-02:
Actualy I don’t understand why is there (ABS) status of ACC deacrivated by diag.

What I’m tesing to achieve is to enable ACC and DW ( I’ve koleos where this functionality was available for 0€ but it must be checked during order and it wasn’t ).

I have enabled DW and it is working. But ACC not - I have enabled it every where I found it, except ABS ecu.

Is there some special diag ecu that can keep holding this disabled or is it just diag function of ABS ecu ?

Will try to increate timeout tomorrow and see if I’ll be able to change this setting in ABS ecu.

Thank you

Michal

shrlnm, 2019-01-02:
How old is your koleos? 1 or 2 months? Does it have a radar or advanced camera under the windshield? One of them are required for activating ACC. Simple camera is not enough.

michalzahor, 2019-01-02:
Koleos is from end of september 2018.
It has RaCam. I’ve enabled distance warning and is working correctly.

I don’t have steering wheel with a button to change a distance for ACC but I think it should work without it.

After I have setup ACC is most ecus ( expect abs there is still “deactivated by diag” ) I’m not able use tempomant ( even basic CC ).

Michal Zahor

michalzahor, 2019-01-03:
Result of ECU scan

[image]

michalzahor, 2019-01-03:
I've tried to increase timeout ( from Auto, to 600ms ) and I was able to write data.

But then check ESP, check hill assist,  warning in indicated . When I deactivate ACC in ABS/ESP, then the warning is done.

Seems like somewhere is ACC still considered as not active or prohibited.

After setting ACC activated in (ABS/ESP )
[image]

How the parameter looks before activation in ABS/ESP
[image]

michalzahor, 2019-01-03:
I have modified Rlink fw 7.x only using development menu. In DDT4ALL there is only radnav 2.x and 3.x and I was worried about applying changes using DDT4ALL.

In dev mod I have just enabled ACC in one place - don't remenber now where but it was in the same place where DW ( distance warning ) and DW is working.

Can you suggest other changes ? Is it safe to use radnav in DDT4ALL if my ecu database doesn't have radnav 7.x ?

shrlnm, 2019-01-03:
Using ddt4all is preferable to change configuration in radnav then thru DeveloperSide. And do not remember to reset radnav changes

shrlnm, 2019-01-03:
RadNav_2*.xml  is safe for your car

michalzahor, 2019-01-03:
1. Is it safe to use for example RAdNat 3.x v3.3 in DDT4ALL if I have RLINK2 firmware 7.xx ?

2. Rlink reset - long press on RLINK ( will go black for few seconds )  is the same as Hard reset ?

shrlnm, 2019-01-03:
Yes, long press on power button next to screen is equal to hard reset of radav

Don't know anything about resetting racam. Some ECUs require hard reset for changes take effect.

shrlnm, 2019-01-03:
I suggest to save configuration before change something.

michalzahor, 2019-01-03:
Is it possible to save current ECU config or just write down original value ?

cedricp, 2019-01-03:
The last version creates a file named **screen.txt** in the log directory. Here you can find old read values.

cedricp, 2019-01-03:
Here's a sample of a session : 
```
ECU : UCT_E84_SERIE_3.json
	@ 14:05:16.137
	Screen : Misc Values and Timings	Request : ReadDataByLocalIdentifier: general system infos
		{"Status EEPROM": "Writing"}
	@ 14:05:16.144
	Screen : Misc Values and Timings	Request : ReadDataByLocalIdentifier: misc timings and values
		{"Voltage High 1": "193", "Timeout Button Windows": "60", "Voltage High 3": "181", "Voltage Low 1": "92", "Timeout Supply Voltage Range 3": "1000", "Voltage Low 3": "103", "Voltage Low 2": "97", "Timeout Runtime Stop Mode": "60000", "Voltage High 2": "187", "Timeout Window Button Short Push": "220", "Maximum Veh. Speed": "3", "Timeout Supply Voltage Range 5": "1000", "Timeout Supply Voltage Range 4": "1000", "Timeout Supply Voltage Range 1": "6000", "Multiplier Thermo Prot.": "10", "Timeout Supply Voltage Range 2": "1000", "Timeout Button Roof": "60", "Timeout Roof Button Window Close": "500", "Tolerance Window Button Open Timer": "5000", "Timeout Runtime Hydr. Pump": "18000", "Timeout Buzzer Alert": "30"}
	@ 14:05:23.714
	Blocked writing parameter : 
		{"Timeout Roof Button Window Close": "500", "Voltage High 1": "193", "Timeout Button Windows": "60", "Voltage High 3": "181", "Maximum Veh. Speed": "3", "Timeout Supply Voltage Range 3": "1000", "Voltage Low 3": "103", "Voltage Low 2": "97", "Timeout Runtime Stop Mode": "60000", "Voltage High 2": "187", "Timeout Window Button Short Push": "220", "Timeout Supply Voltage Range 5": "1000", "Timeout Supply Voltage Range 4": "1000", "Timeout Supply Voltage Range 1": "6000", "Multiplier Thermo Prot.": "10", "Timeout Supply Voltage Range 2": "1000", "Timeout Button Roof": "60", "Voltage Low 1": "92", "Tolerance Window Button Open Timer": "5000", "Timeout Runtime Hydr. Pump": "18000", "Timeout Buzzer Alert": "80"}
	@ 14:05:26.941
	Screen : Settings	Request : ReadDataByLocalIdentifier: traceability data
		{"Traceability Data": "\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}
	@ 14:05:26.942
	Screen : Settings	Request : ReadDataByLocalIdentifier: general system infos
		{"Status EEPROM": "Writing"}
	@ 14:05:26.943
	Screen : Settings	Request : ReadDataByLocalIdentifier: endurance mode
		{"Endurance Mode": "00"}
	@ 14:05:26.944
	Screen : Settings	Request : ReadDataByLocalIdentifier: rollbar, montage
		{"Rollbar Present": "00", "Montage Mode": "00", "Window Mode": "00"}
	@ 14:05:31.938
	Blocked writing parameter : 
		{"Endurance Mode": "Normal Mode"}
	@ 14:05:34.618
	Blocked writing parameter : 
		{"Rollbar Present": "Rollbar present", "Montage Mode": "Montage Mode", "Window Mode": "after 1s"}
	@ 14:06:02.105
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: times latches
		{"Tolerance Right Latch": "0", "Tolerance Left Latch": "0"}
	@ 14:06:02.106
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: general system infos
		{"Status EEPROM": "Writing"}
	@ 14:06:02.107
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: tolerance values 2
		{"Tolerance Start Hydraulic (close roof)": "0", "Tolerance Close Trunklid (close roof)": "0", "Tolerance Close Roof (close roof)": "0", "Tolerance Open Trunklid (close roof)": "0"}
	@ 14:06:02.108
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: tolerance values 1
		{"Tolerance Open Trunklid (open roof)": "0", "Tolerance Open Roof (open roof)": "0", "Tolerance Start Hydraulic (open roof)": "0", "Tolerance Close Trunklid (open roof)": "0"}
	@ 14:06:02.109
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: timeout values 1
		{"Timeout Open Trunklid 1 (open roof)": "0", "Timeout Close Trunklid 2 (open roof)": "0", "Timeout Unlock Trunklid 1 (open roof)": "0", "Timeout Close Trunklid 1 (open roof)": "0", "Timeout Unlock Trunklid 2 (open roof)": "0", "Timeout Open Roof 1 (open roof)": "0", "Timeout Open Roof 2 (open roof)": "0"}
	@ 14:06:02.111
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: timeout values 2
		{"Timeout Unlock Trunklid 1 (close roof)": "0", "Timeout Close Trunklid 1 (close roof)": "0", "Timeout Unlock Trunklid 2 (close roof)": "0", "Timeout Close Trunklid 2 (close roof)": "0", "Timeout Open Trunklid 1 (close roof)": "0", "Timeout Close Roof 1 (close roof)": "0", "Timeout Close Roof 2 (close roof)": "0"}
	@ 14:06:09.967
	Writing parameter : 
		{"Timeout Open Trunklid 1 (open roof)": "0", "Timeout Close Trunklid 2 (open roof)": "0", "Timeout Unlock Trunklid 1 (open roof)": "0", "Timeout Close Trunklid 1 (open roof)": "56", "Timeout Unlock Trunklid 2 (open roof)": "0", "Timeout Open Roof 1 (open roof)": "0", "Timeout Open Roof 2 (open roof)": "0"}
	@ 14:06:10.092
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: times latches
		{"Tolerance Right Latch": "0", "Tolerance Left Latch": "0"}
	@ 14:06:10.093
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: general system infos
		{"Status EEPROM": "Writing"}
	@ 14:06:10.094
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: tolerance values 2
		{"Tolerance Start Hydraulic (close roof)": "0", "Tolerance Close Trunklid (close roof)": "0", "Tolerance Close Roof (close roof)": "0", "Tolerance Open Trunklid (close roof)": "0"}
	@ 14:06:10.095
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: tolerance values 1
[... cut ...] imeout Open Roof 1 (open roof)": "0", "Timeout Open Roof 2 (open roof)": "0"}
	@ 14:06:10.099
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: timeout values 2
		{"Timeout Unlock Trunklid 1 (close roof)": "0", "Timeout Close Trunklid 1 (close roof)": "0", "Timeout Unlock Trunklid 2 (close roof)": "0", "Timeout Close Trunklid 2 (close roof)": "0", "Timeout Open Trunklid 1 (close roof)": "0", "Timeout Close Roof 1 (close roof)": "0", "Timeout Close Roof 2 (close roof)": "0"}
```

michalzahor, 2019-01-03:
I have do changes but:

Claster - changed 2 values ( don't know name of param, will write down tomorrow )

RadNav - changed 1 value, hard reset

RaCam - using XZG_RaCam_V1_2_20160805T115937.xml
             - unable to find  values with mph, just kmh ( changed kmh values )
             - unable to find there 2 in screens, but they are in XML - how can I change that ?
                   AEBS37_VectorFeedBack.P_tng_m_DistVectorFeedBack_1_2, AEBS37_VectorFeedBack.P_tng_m_DistVectorFeedBack_1_1

ABS/ESP - changing to Activated -> I see on dashboard "Check ESP"

Except 

AEBS37_VectorFeedBack.P_tng_m_DistVectorFeedBack_1_2, AEBS37_VectorFeedBack.P_tng_m_DistVectorFeedBack_1_1
and mph values  

all was changed but doesnt work.

Is it possible that system knows I don't have distance button on the steering wheel and that's why it is blocking activation ?

shrlnm, 2019-01-03:
Try to use RaCam_V4_20170703T152858.xml but I not sure that these parameters are available here. If not then you may try to write own script, make own screen in ddt4all or use a pyren it has synthetic screens for every command. 

About distance button not sure that ECU may know. These buttons usually made as a resistive set and only two wires toward ECU but I may be wrong

michalzahor, 2019-01-04:
I have made custom screens and do the whole process one more time:

1. ddt4all using connected mode
[image]

2. choosing XZG - KOLEOS 2
[image]

3. RaCAM v4
[image]

[image]

[image]

I have created custom screen also for MPH values and changed ( forget to make a photo ). Just in
MPH there is not defined last one variable - DistanceControllerMaxSpeedCommand
[image]

Those values are different from you suggested - I'm little bit worried to mess with that - do you think it is related to ACC not operating ?
[image]

4. RadNav 2.x
[image]

5. Cluster

I have found 2 values in cluster mentioning ACC ( the first one should be not set to ACC ? )
[image]
[image]

6. ABS
[image]
[image]

I'm thinking about the text - Deactivated by diag - like in this ABS ecu it should be auto activated if in other ECUs is all ok or some diagnostic returns that ACC can be activated.  Maybe can I use function in  ddt4all - DTC reset to reset errors if any without any harm ?

shrlnm, 2019-01-04:
Did you reload the TDB after changing 
"1" Text="CC/SL"
to
"2" Text="ACC/SL"
?

Banana009, 2019-01-04:
I tried to activate ACC on my Scenic IV. But without success. It seems that missing some hardware part( different ABS sensors). Look onto previews threads.

Banana009, 2019-01-04:
Here: https://github.com/cedricp/ddt4all/issues/47

michalzahor, 2019-01-05:
Yes I have reloaded TDB after change and there was ACC/SL

I have part number of RaCam and I'll try get info if it is the same as in the version with ACC enabled.

@Banana009 - can be so, courious if there is some possibility to get info from ECUs why ABS ECU is indicating that ACC was deactivated by diag. -  which diag, reason ,etc.

igno77, 2019-01-05:
@michalzahor your ddt4all version seems to be the last one (5.20 with ecu dec2018) and on it I've had a lot of problem to go inside the ABS ecu!! I've disinstalled that version and installad again the "older 5.02 and now is working fine

michalzahor, 2019-01-06:
@shrlnm - you were right with TDB ( cluster ) that only one item should be changed ( I though that I have enabled 2 items but first one was enabled from the factory ). When I now was changing back from ACC to CC mode I have disabled also ACC in the first screen of cluster and I was getting "STOP" warning and message "Program Err". When I enabled back ACC on the cluster frst screen then error is gone and CC is working properly.

shrlnm, 2019-01-06:
We need to explore BVMEXTRACTION database for understanding witch option may be activated on which firmware. You have still very rare option AEBS05. It appeared in second half of 2018 and not too many koleoses have it. The number of koleoses with AEBS05 and RVDIST (it is ACC option) is even more rare.

michalzahor, 2019-01-06:
Will try. This Koleos has AEBS with pedestrian recognition. As I know, koleos before doesn’t have this functionality.

michalzahor, 2019-01-06:
@shrlnm seems like BVMEXTRACTION is not part of ddt2000 database I have downloaded. 

I have database from <Date>2018-12-20T19:10:20</Date>

michalzahor, 2019-01-08:
Waiting for some info and will try end of the week

Banana009, 2019-01-08:
What information are you waiting for?

michalzahor, 2019-01-08:
part number of my racam if is the same as for koleos with activated acc from factory

Michal Zahor
mzahor@dtech.sk

Banana009, 2019-01-08:
Michale, for Scenic IV are 2 types of RaCam. Part numbers 284629358R and 284628548R. 
You can see on allegro:
https://allegro.pl/oferta/renault-grand-scenic-iv-kamera-radar-legalny-z-vat-7395071700 
https://allegro.pl/oferta/renault-scenic-iv-2017-kamera-radar-sensor-szyba-7658649540
Unfortunately, I don't know if it is the reason why ACC can not be activated.
Interesting is  that the software 5.2.2 is the same. 

I suppose that for Koleos it is the same. But the RaCam has different software SW 2.1.0 
I found out only part number 284625670R
https://allegro.pl/renault-koleos-ii-2017-radar-kamera-284625670r-i7662479894.html

michalzahor, 2019-01-09:
I have RaCam SparePart: 284625816R, SW 2.2.1

michalzahor, 2019-01-09:
Seems that I'm not able to get further - guy in the authorised service said that it is not possible to activate because there is missing radar in the front bumper etc. As I know, new version doesn't use radar in bumper as RaCam is able to do the work. Didn't wanted to ask too much and I can not argue with more data from obd as car is under warranty and .... Maybe just it needs time to let somebody discover the way.

cedricp, 2019-01-09:
I added a little tool to send requests that are not on screens : 
[image]

Just next to the hex icon.

michalzahor, 2019-01-10:
@cedricp - is there something related to acc :-) ?
I've tried PyRen to compare screens and parameters. Found that ABS is different and I haven't find ACC related settings for ABS in PyRen. 

Also there was memorised error in ABS related to ACC - I've cleared error but ACC not working.

[image]
[image]
[image]
[image]

michalzahor, 2019-01-10:
Also on one RaCam screen in ddt4all I saw live data while driving about ACC (parameter name contains ACC) - distance, speed, ...   so I believe that RaCam is capable of ACC. The only question is what is blocking the functionality.

shrlnm, 2019-01-10:
@michalzahor you have chosen wrong xml in pyren when run ddt for a first time, used old version (actual is 9n) or were not connected to car. Now you need to remove "cache" folder and scan ECUs again.

When you first run ddt from pyren, the screen should be like this

```
Starting DDT process
Getting ID from 2180
IdRsp: 61 80 38 35 37 30 52 09 30 33 37 38 36 32 32 52 43 66 1E 30 00 13 02 01 00 88 00
Address:"01" DiagVersion:"9" Supplier:"037" Soft:"4366" Version:"1E30"
Loading eculist
01 # ABS
########################################
      0 - ESC___1540_HFE_XZG_Version_2.2_20170620T142139.xml
  15213 - ESC___1540_HFE_XZG_Version_1.9_20170620T141226.xml
  14034 - ESC___1540_HFE_XZG_Version_2.5_20180502T151847.xml
########################################
You may enter the file name by yourself or left empty to exit
File name [ESC___1540_HFE_XZG_Version_2.2_20170620T142139.xml]:
```

shrlnm, 2019-01-10:
@michalzahor I have checked part numbers of RaCam for Koleos with ACC (RVDIST option)
Your koleos VIN:VF1RZG.................82 racam SW part number is 284A61717R. The same PN have all koleos with options AEBS05 and RVDIST 
 
Try to monitor system frames on a CAN, are there something related to racam data?

michalzahor, 2019-01-10:
My VIN was in log file ? [ deleted it just for the case - car is still under warranty ].

Hm will try that tomorrow.

I have noted today that when ACC is activated in RaCam, then I'm not able to use CC ( ACC of course not too )

michalzahor, 2019-01-10:
What is also interesting, that my car have ECUs with functionality that I haven't ordered or is not available for this koleos version at all - HFP ( handsfree parking ) module , ADP ( adjustable seats with position memory ).

michalzahor, 2019-01-11:
@cedricp I've tested new functionality ( tool next to hexa )

I have found some ACC related data in ECU but I'm not able to read them.

I think I'm on the right XML file and tried with ignition on ( motor on ), ignition on ( motor off ) .

[image]

Don't know what is wrong.

cedricp, 2019-01-11:
The problem here, is that you're dealing with a 29bits addressing ECU, and it's not handled. I pushed a fix to manage it, but I cannot test it, I have no 29bits ECU at home.

michalzahor, 2019-01-11:
seems that doesn’t work ( same result, empty response )

Michal

cedricp, 2019-01-11:
Are you sure the XML file is the good one for your ECU ? Can you share your log file ?

shrlnm, 2019-01-11:
As I know there are no  ECUs with 29 bits CAN id on Koleos. Cedric is right, you have chosen wrong XML 

For not to post the VIN with a log, just post responses on command 2180 or on commands 22F1A0, 22F18A, 22F194 and 22F195

michalzahor, 2019-01-11:
What is proper procedure to check ECUs vs XML match  ? When I did run the test ( scan )  few days before few were marked as “not perfect match” and few “like this - not suitable XML found” but the all others are ok or just unknown state ?

Will run the @shrlnm test later today.

Michal

cedricp, 2019-01-11:
If the scan process cannot find the good XML, it means that there's no suitable XML file for that ECU.
A "_not perfect match_" scan means the software version is not the good one. (but can help many times)

michalzahor, 2019-01-11:
Right I forgot to rescan when I have switched from WinXP version to the newest.

I have chosen Koleos II and the result is:

[image]

[image]

Some ECUs were identified, some not perfect match , some no relevant ECU.
I'm suprised that so many were not found ( too new car ? :-) )  - I see that some XLM files are from december 2018 so I believe it is one of the latest ECU XML set.

What about ECUs that seems to be raleted to koleos - they were loaded after choosing koleos, but are not on the test / sscan list. Like ABS, RaCAm, MIU, ... ?

Also TDB said "no relevant ECU found" but is working as I have to use it for ACC activation too.

michalzahor, 2019-01-11:
Result to commands:

[image]

shrlnm, 2019-01-11:
First read current value by 22217C and 22217B commands

michalzahor, 2019-01-11:
22217C: 62 21 7C 00

22217B: 62 21 7B 00

Michal

shrlnm, 2019-01-12:
Ок, now you may try to change values by 2E217Cxx and 2E217Bxx on your own risk.  Further you will be able to try to return configuration back by  2E217C00 and 2E217B00.  It is how to more or less safe change something

I've explored a bit your xml file EDC17C84_170C_* and found one interesting parameter "ACC status from CAN". By my opinion,   it should show us if racam sends ACC frames or not. Also try to carefully analyse what is shown on screens "CRUISE -> RV/LV *"

michalzahor, 2019-01-12:
I’ll check those “read” params.

But I’m warried to change those 2 params especially because of the explicit info that it will be written somewhere else then usual data - eeprom. I don’t know what kind of memory is used to write other type of data in ecus but why there is explicit info about eeprom. What if I’ll be not able to rewrite back to 0x00 and ECU will have problem with inconsistent configuration.

Question is if it is known that eeprom van be rewriten back to 0x00 .

Michal

michalzahor, 2019-01-13:
Done few pictures while driving.

CC switch ON ( no icon of CC on main cluster screen and not able to set CC speed - ACC enabled in RaCam, TDB, RadNav

Michal

michalzahor, 2019-01-13:
> Do you mean that you have activated ACC?

Just enabled in RaCam, RadNav and TDB cluster. But it is not working ( when I switch stndard CC switch on, on main cluster there is not CC or ACC icon and CC / ACC is not operating )

Those screen shots are from EDC17.... ecu.

Interesting is ACC status on CAN bus - saying ACC malfunction.

Question is why is ACC considered as malfunction. 

I haven’t enabled those 2 ACC params inEDC17 ecu as I’m worried that will not be able to set them back to 0x00 

Michal

[102 more replies]

closed 2023-06-02 by Furtif