cedricp/ddt4all #128: multisens picture error in RLink 2

Guile88, 2019-01-08:
Hi, I havebeen using ddt4all for change some things in my Talisman and im very happy with this program, but i dont know what happen now that in my RLink appears a Renault Espace instead of a Talisman.
I find the parameter "Vehicle Type" and when i try to change it (change JFC(Espace) for LFD(Talisman)), dont change. I'm using a chinesse ELM wifi and never before has that given me problems.
Thanks
[image]

cedricp, 2019-01-08:
What is the _ELM response_ in the bottom view when you change this parameter ?
Maybe you need to reset the ECU to validate change ?

cedricp, 2019-01-08:
Did you try to update the application, it seems a bit old from what I see...

Guile88, 2019-01-08:
I just updated but the result is the same
[image]

shrlnm, 2019-01-08:
Your fw version is not compatible with this XML. Your radnav returns 9 bytes response on command 22 21 10. XML contains 10 bytes response definition.

Guile88, 2019-01-08:
so... what do you think I can do?

shrlnm, 2019-01-08:
ELM is good. You need to find appropriate xml for your radnav. It should have 9 bytes response on command 222110
<Received MinBytes="9">
Now you are using file with 10 bytes
<Received MinBytes="10">

Guile88, 2019-01-09:
Thanks for the help but I don't know how do it so maybe I will go to the car dealership. 
Thanks

cedricp, 2019-01-10:
The parameter **SpeedCurve** cannot be interpreted (read as FF)

shrlnm, 2019-01-10:
I suppose that the "Welcome Sequence volume level" was inserted in new new versions.
Try these command definitions

```
<Request Name="DataRead.Car Configuration">
	<Sent>
		<SentBytes>222110</SentBytes>
	</Sent>
	<Received MinBytes="9">
		<ReplyBytes>622110</ReplyBytes>
		<DataItem Name="Car Configuration.GearBox" Ref="1" FirstByte="4"/>
		<DataItem Name="Car Configuration.VehicleType" Ref="1" FirstByte="5"/>
		<DataItem Name="Car Configuration.Engine" Ref="1" FirstByte="6"/>
		<DataItem Name="Car Configuration.Manufacturer" Ref="1" FirstByte="7"/>
		<DataItem Name="Car Configuration.Hand_drive" Ref="1" FirstByte="8"/>
		<DataItem Name="Car Configuration.SDVC_curve" Ref="1" FirstByte="9"/>
	</Received>
</Request>

<Request Name="DataWrite.Car Configuration">
	<ManuelSend/>
	<Sent>
		<SentBytes>2E211008000000000003</SentBytes>
		<DataItem Name="Car Configuration.GearBox" FirstByte="4"/>
		<DataItem Name="Car Configuration.VehicleType" FirstByte="5"/>
		<DataItem Name="Car Configuration.Engine" FirstByte="6"/>
		<DataItem Name="Car Configuration.Manufacturer" FirstByte="7"/>
		<DataItem Name="Car Configuration.Hand_drive" FirstByte="8"/>
		<DataItem Name="Car Configuration.SDVC_curve" FirstByte="9"/>
	</Sent>
	<Received MinBytes="3">
		<ReplyBytes>6E2110</ReplyBytes>
	</Received>
</Request>
```

KarelSvo, 2019-01-13:
> @shrlnm
> 
> from what do you understand that the response to command 222110 is in 9 bytes?
> And above all where can alternate xml be found?
> 
> Thank you!
Talisman audio ecu selection: R0-15 and Radnav
command 222110: R0-15 with 9bytes and Radnav with 10 bytes
identify your ecus with pyren
https://www.drive2.ru/users/zmiterm/

KarelSvo, 2019-01-13:
RadNav_v1.4_20150225T100059.xml
last version with 9 bytes on 222110

Guile88, 2019-01-15:
Where I have to write all of this comands?

Guile88, 2019-01-15:
Ok, thank you for the help! But I need you to forgive me because I'm a litle bit dumb and don't know how I have to do it. How I can use this file (RadNav_v1.4_20150225T100059.xml)? where it is? Do I have to download it? 
Sorry

Guile88, 2019-01-24:
hi again! 
i have tryed something that i saw in a german forum for change the space image for a talisman image pressing "hard reset" buton in ddt4all, but the space still appearing and now the rlink screen is rotated 90º. do you know how to rotate again?
thanks!!

KarelSvo, 2019-01-24:
What you have changed, is show in the log files.

Guile88, 2019-01-24:
I tried to do this:

"audio --- radnav 3.x-v3.0 ---- configuration --- ecu part 1 ---
1.vehicle type made on JFC
2.then the yellow write all over these purple fields
3.then rightmost read all and
4.then left hard reset. Engine was running at this time.

I only took JFC because the type RFD is not available for me. And as soon as you press write all fields must appear white and then if you press read all the values ​​must have been transferred from left to right"

A guy in Talismán forum wrote this and solved the problem

KarelSvo, 2019-01-24:
folder ddt4all/logs

Guile88, 2019-01-25:
Thanks! I fixed screen orientation.
Now  i just have to fix the Talisman picture..
Thank you all

ak1mb0, 2019-01-27:
This is my issue also, I'm unable to write a new vehicle type

ak1mb0, 2019-01-27:
Mine is also reading the same must be a default peramiter, but I need to change it to display a Kadjar but it wont allow me to write the new values from the left.  [image]

ak1mb0, 2019-01-27:
Ah ok, is there a link to download the .XML file?

KarelSvo, 2019-01-27:
check your email

ak1mb0, 2019-01-27:
Thanks KarelSvo

Juicy312, 2019-02-03:
@KarelSvo 
Hello, I have the same issue. Is it possible to share the XML file please?
PS: My oldest version of RadNav in my database is 1.5
[image]

aqngelrock, 2019-02-03:
Good evening... Can you send me the xml, plese?

aqngelrock, 2019-02-03:
Excuse me...
nas.autosender@gmail.com
 Thank you

KarelSvo, 2019-02-03:
check your email

Juicy312, 2019-02-03:
I am sorry, here is my email:
j312.main@gmail.com
Thank you

aqngelrock, 2019-02-04:
Thank a lot KarelSvo.... So How can use it?

wdegiac, 2019-02-04:
Hello I switched to the Rlink 3.x version and now I see the wrong car instead of the megane, how can I solve?

Guile88, 2019-02-04:
Hello, I need this XML. Can you send me?
hachguile@gmail.com
Thank you

wdegiac, 2019-02-04:
you can send me xml and tell me where to copy it in DDT4all to change the vehicle type
williamdegiacomo@libero.it

shrlnm, 2019-02-04:
Do you have logs of your try to change values?

wdegiac, 2019-02-04:
[log ddt.zip](https://github.com/cedricp/ddt4all/files/2827754/log.ddt.zip)
@shrlnm here are the logs

Juicy312, 2019-02-04:
Can someone please share with me the XML file? I did not received it.
My email is: j312.main@gmail.com
Thanks!

wdegiac, 2019-02-04:
@Juicy312 hello I sent you an email with the file, see if it works to you do not change the vehicle

Juicy312, 2019-02-04:
Received the file. Thank you.
Unfortunately, I can't open it via "File" tool menu and then "Open XML". It gives me the following error in DDT console:
[image]
This is the lastest release version.

cedricp, 2019-02-04:
Remove any non-ASCII (or unicode) characters in filename.

cedricp, 2019-02-04:
Did you download latest sources ? (not installer I mean)

Juicy312, 2019-02-04:
I downloaded it from here: https://github.com/cedricp/ddt4all/releases
And yes I used the installer.
I am using Windows version only for viewing and checking the database. I use linux version for the OBD connection with the car on a Raspberry.
I didn't tried yet to open the XML via my Raspberry.

wdegiac, 2019-02-04:
yes after updating to the 3.x made me this change not immediately though 4 febbraio 2019 alle

cedricp, 2019-02-04:
you can replace the code in the installer directory to update.

wdegiac, 2019-02-04:
Forgive me I do not understand what installation are you talking about? I use ddt4all 5.3.x the last available with xml that you indicate in the forum but does not change the figure

shrlnm, 2019-02-04:
> here are the logs

The log is good. There is no problem with parameter changing. 
You always did "key on/off reset" 1102. Had android reloaded after that?
After parameter changing I usually use "hard reset" 1101 or long press on power button

wdegiac, 2019-02-04:
hi, I've done both on / off and both the reset by pressing the power off button for a long time, but it remains the scenic. here are the logsThe log is good. There is no problem with parameter changing. You always did "key on/off reset" 1102. Had android reloaded after that? After parameter changing I usually use "hard reset" 1101 or long press on power button

shrlnm, 2019-02-04:
Then probably your MICOM (the hardware module responsible for CAN) version not compatible with android version you installed. But it is only my guess.

wdegiac, 2019-02-04:
what do you recommend? can we solve? or do I downgrade to version 2.x, modify the machine and upgrade again? Then probably your MICOM (the hardware module responsible for CAN) version not compatible with android version you installed. But it is only my guess.

Juicy312, 2019-02-04:
A little update on my side: I changed this evening the car type of my Mégane and now I switched from Espace V to Talisman picture in MEX (MultiSense). That is weird. My car type has been set to BFB (which is Mégane IV) so why should a Talisman be shown?

shrlnm, 2019-02-04:
> what do you recommend? can we solve? 
You may try to use diagRW but I not sure that it helps

wdegiac, 2019-02-04:
I've already tried diagdw sees it correctly BFB (megame 4) but rlink always remains the scenic.   what do you recommend? can we solve? You may try to use diagRW but I not sure that it helps

wdegiac, 2019-02-05:
@shrlnm can I solve it in some way or are you working on a solution?
since many have this problem from what I have seen
thanks 1000

wdegiac, 2019-02-05:
@shrlnm si je peux vous donner une entrée, si je lis les données avec RadNav 2.x v2.5, je vois correctement la marque (Renault) et le type de voiture (JFC-scenic), mais je ne peux pas les modifier car je ne les lis pas / écrit correctement, alors que le fichier .xml que vous avez mis dans ce groupe, je vois toutes les données correctes, ainsi que le type d’Auto (BFB-Megane).

wdegiac, 2019-02-05:
@cedricp Remove any non-_ASCII_ (or non-_unicode_) characters in filename.?????? What do you mean can I solve?

wdegiac, 2019-02-05:
@shrlnm the log that I sent you with XML RadNav_v1.4_20150225T100059, and in fact the auto parameters are correct, is there a way to change even the parameters that reads android?

shrlnm, 2019-02-05:
@wdegiac for understanding this behaviour and ways to influence on it we need to reverse com.lge.ivi.jar ConfiguratioManagerBase.class from lgu image you installed. This class is responsible for this part of configuration in android. But it is too hard.

You may try to change gear box type and look if it will change car type or not. "Gear box" is because it is situated in 10 bytes commands on a place of car type in a 9 bytes commands.

shrlnm, 2019-02-05:
Also you  may try to change Engine type as a nearest byte from the other side 

<DataItem Name="Car Configuration.GearBox" FirstByte="4"/> 
<DataItem Name="Car Configuration.VehicleType" FirstByte="5"/> 
<DataItem Name="Car Configuration.Engine" FirstByte="6"/> 
<DataItem Name="Car Configuration.Manufacturer" FirstByte="7"/>
 <DataItem Name="Car Configuration.Hand_drive" FirstByte="8"/> 
<DataItem Name="Car Configuration.SDVC_curve" FirstByte="9"/>

wdegiac, 2019-02-06:
@shrlnm hello, in the xml file RadNav_v1.4_20150225T100059 I did various tests modifying only the part "DataRead.Car and DataWrite.car" and always only the "Vehicle Type" and after some attempts I managed to write only that part, for those who wanted to group "MAGANE 4 FRIENDS" we are posting the guide and the file.

Juicy312, 2019-02-10:
@wdegiac
Could you please share the link to the guide?
Thanks!

wdegiac, 2019-02-11:
@Juicy312 Hello, subscribe to the Facebook group "Megane 4 friends" and find the guide in the file section

haki64, 2019-03-27:
> Dosyayı aldı. Teşekkür ederim. 
> Ne yazık ki, "Dosya" araç menüsü ve ardından "XML Aç" komutunu kullanarak açamıyorum. DDT konsolunda bana şu hatayı veriyor: Bu en son sürüm sürümüdür.
> [image]

Hi, I had the same problem with my Talisman. My problem is solved. First of all, I would like to thank the user of "shrlnm" for the solution. If you get the "UNICODE and UTF-8" error when opening the XML file into ddt4all, open the XML file with notepad and save it with UNICODE encoding type. Open the new file again with notepade and save it with the UTF-8 encoding type. Now you can call this file into ddt4all application.

[55 more replies]

closed 2023-06-02 by Furtif