cedricp/ddt4all #130: Delete DTCs problem

pakoto, 2019-01-11:
I´m not sure that delete DTC works, 
After delete error still appears and no response. elm works ok, no arm, change configuration ok, etc.

[image]
[image]

cedricp, 2019-01-11:
Well, it's possible that you're facing a permanent fault (short circuit, bad contact, etc.) so the fault reappears after deleting.
Share your logs so I can check DTC clear command is successfully executed.

cedricp, 2019-01-11:
The DTC message is from engineer database, so difficult to do better. Clip db is more mehanic friendly.

pakoto, 2019-01-14:
> Well, it's possible that you're facing a permanent fault (short circuit, bad contact, etc.) so the fault reappears after deleting.
> Share your logs so I can check DTC clear command is successfully executed.

[Download.zip](https://github.com/cedricp/ddt4all/files/2756745/Download.zip)

cedricp, 2019-01-14:
Strange, indeed, the ClearDiagnosticInformation.All command is 14FFFFFF, but the cluster doesn't respond to it...
Try to increase timeout value and check if you get a response.
Did you try to clear with car off/on, some ECU doesn't want to clear with motor on or off.

KarelSvo, 2019-01-14:
Clio 4 Cluster

ECU : 11493  FG0311121.XML
Screen : DE (STD_B)

#############################  Command parameters  #############################

Prerequisite : Contact mis et moteur arrêté. Pour les véhicules équipés de transmission automatique, positionner le levier sur P ou N. Pour les véhicules électriques, débrancher le cordon de charge du véhicule. Dès que la commande est terminée, retourner sur l’écran des DEFAUTS avant de couper le contact.

name         : R001
codeMR       : RZ001
label        : MEMOIRE DE DEFAUT
type         : RZ
scenario     : 
inputlist    +
serviceID    +
              : (71   ) 14FFFFFF
Are you ready to execute the command? <yes/no>:

pakoto, 2019-01-14:
Thanks. I´ll try next time.

pakoto, 2019-01-15:
It's clio iv cluster installed into dacia dokker

shrlnm, 2019-01-15:
I were wrong they are MEMORISED 
```
1  - DTCC00187 CAN COMMUNICATION                                  1.DEF  MEMORISED 
2  - DTC940811 SEAT BELT WARNING LIGHT CIRCUIT                    CC.0   MEMORISED 
3  - DTC941213 EXTERNAL TEMPERATURE SENSOR CIRCUIT                CO     MEMORISED 
4  - DTC940213 FUEL SENDER CIRCUIT                                CO     MEMORISED 

CC.0:Short circuit to earth
CO:Open circuit
```

KarelSvo, 2019-01-17:
> I were wrong they are MEMORISED

The question remains why the ecu with ddt4all does not respond to 14FFFFFF und with pyren works?
So the reports of some Dacia user,  who have changed at their cluster configurations.
And  by reset the cluster in the error memory have errors the multiplexed networks .

shrlnm, 2019-01-18:
> The question remains why the ecu with ddt4all does not respond to 14FFFFFF and with pyren works?

@cedricp is right, @pakoto have to extend the timeout (AT ST) up to one second, but actually we have no need to wait a response on this command except  to look if there is a negative or positive response. The command consists only from one frame. It was sent and it have been applied if  ignition were on and engine stopped.

We may also check if command (14FFFFFF) have been applied or not by  reading DTC context, it have to show odometer value latched in the moment when the DTC was detected last time. It gives us information if the DTC reappear or just have not been cleared

```
ECU : 11606   (RT:0.0000)   FG0311121.XML
Screen : DE (STD_B) -> DTC940213 FUEL SENDER CIRCUIT                                CO     MEMORISED 

	CO:Open circuit
	----------- Status byte: 28 -----------
	warningIndicatorRequested           : 0
	testNotCompletedThisOperationCycle  : 0
	testFailedSinceLastClear            : 1
	testNotCompletedSinceLastClear      : 0
	confirmedDTC                        : 1
	pendingDTC                          : 0
	testFailedThisOperationCycle        : 0
	testFailed                          : 0

PR030  MILEAGE SINCE LAST APPEARANCE OF FAULT                            0 km                

Press any key to exit

```

KarelSvo, 2019-01-18:
So, the Can Timeout value for the Delete DTC application would need to be raised automatically, to not unsettle the software users?

shrlnm, 2019-01-18:
Yes, it may be done automatically but in this case the ATST timeout  is not a root cause of why DTC reappeared or not cleared.

KarelSvo, 2019-01-18:
Something must to make pyren differently, if there DTC delete immediately works.

shrlnm, 2019-01-18:
This is an example of how pyren do

May be the reason in using odd digit 

```
>[21:46:06.822]0210031
<[0.02]0210031
065003003201F400

>
>[21:46:06.842]0414FFFFFF1
<[0.05]0414FFFFFF1
037F147800000000

>
mon: 21:46:07.624 : 0154000000000000
>[21:46:07.840]AT
<[21:46:07.841]STOPPED

>(shifted)AT
<[0.006]AT
?

>

```

shrlnm, 2019-01-18:
720ms delay and NR:78

cedricp, 2019-01-18:
I pushed some tweaks to extend CAN timeout for _ReadDTC_ command and fixed the translated negative response code (was using bad byte).
_NR 78_ = **Request Correctly Received-Response Pending**

pakoto, 2019-01-19:
i´ll test ASAP. Thanks!
I see you change timeout to 1500, but i can´t see anything about bad byte. Is correct?

cedricp, 2019-01-19:
It's in a previous commit
https://github.com/cedricp/ddt4all/commit/bc0ed388ef56ff2d2f21c803f9c5255f906d2a5c

pakoto, 2019-01-26:
Hi. We tested delete DTC and seems to be solved. Logs attached.

[screens.txt](https://github.com/cedricp/ddt4all/files/2799150/screens.txt)

[image]

P.D. Cedric, do you receive update translation files sent by email?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif