cedricp/ddt4all #131: Is there an option to disable auto fold/unfold of side mirrors ?

bobybc, 2019-01-17:
Hi,
I have Kadjar 2017 with auto folding function for side mirrors
On every lock/unlock of the car it always fold/unfold the side mirrors
I hope there isa na option to disable that?

10x in advance
Boby

gordon6, 2019-01-21:
There's only One option. On or Off Auto folding function for side mirrors in BCM_DDT_11, Mmi, Mmi - Config, Automatic_folding_CF, False, Send. then you can disable auto folding fuction for side mirrors.
When you need that to enable again, change the value to True, Send.  So simple.

bobybc, 2019-01-23:
10x a lot for the answer
Do you know whether this option is available in Developer mode UI?

gordon6, 2019-01-24:
The answer is in my comment.
You have to find 'BCM_DDT_11, Mmi, Mmi - Config, Automatic_folding_CF, False, Send'.
THat's all.

franeti, 2019-02-08:
Hi gordon6! I´m very interested in this. In my case Kadjar 2015. I installed the kit for auto folding. In my case only fold when I start the car. It´s not good because some times the mirror stop during the start engine (like a electricity cut) So I prefer for sure to fold when I unlock the car.

Everything is fine when I lock the car. The mirror unfold good.

So, in the same option that you suggest, is possible to change it like the other user?

Thanks in advance.

Regards.

> There's only One option. On or Off Auto folding function for side mirrors in BCM_DDT_11, Mmi, Mmi - Config, Automatic_folding_CF, False, Send. then you can disable auto folding fuction for side mirrors.
> When you need that to enable again, change the value to True, Send. So simple.

Hi gordon6! I´m very interested in this. In my case Kadjar 2015. I installed the kit for auto folding. In my case only fold when I start the car. It´s not good because some times the mirror stop during the start engine (like a electricity cut) So I prefer for sure to fold when I unlock the car.

Everything is fine when I lock the car. The mirror unfold good.

So, in the same option that you suggest, is possible to change it like the other user?

Thanks in advance.

Regards.

bobybc, 2019-02-17:
Hi Gordon,
i cant't find mentioned configuration. I have Kadjar 2017. The closest values i found is under **UCH** menu and when i select **HFE_BCM_DDT_1_4_1** i can see Mmi . When select Mmi - Configuration from the left i can see the options on the right , but there is no Automatic_folding_CF . There is even nothing related to folding. Please see the screen shot : 
[image]

All other options from UCH menu (CMF1_BCM_DDT_9 or HFE_BCM_DDT_7_2), lead to the same Mmi - Configuration , without Automatic_folding_CF

I tried to list other option from main menu, but didn't manage to find anything related to side mirrors

bobybc, 2019-02-24:
I finally managed to disable automatic folding/unfolding
The issue was due to ecus database
I installed ddt4all using the installer from [Release area](https://github.com/cedricp/ddt4all/releases) 36MB
When i run it for the first time it said that there is no database. i just downloaded some ecus.zip and extracted to the ddt4all and everything was ok, except there is no mentioned BCM_DDT_11
When i downloaded bigger installer (169MB) from some Russian forum - the mentioned item appeared
So it will be good to have instructions how to obtain the latest database

closed 2019-02-24 by bobybc

JeromeTR2909, 2024-07-27:
Hello @bobybc 
Can you share the link for that ecu database?