cedricp/ddt4all #133: ddt4all

andru1973, 2019-01-22:
hi everyone, someone has the definitions of ddt4all example of my car renault laguna 3 2008, in CLUSTER / TdB, here is some that I know, but are some that I do not know what is this "CONFIG GENERALE ALS" OR "INHIBITION SBR " Thank you

cedricp, 2019-01-22:
I think ALS is AntiLag system (for turbo charger) and SBR is Seat Belt Reminder

andru1973, 2019-01-22:
there is a table with each name Thank you ,

ohayyam, 2019-01-22:
Where is that table ? Where can i get that table ?

cedricp, 2019-01-23:
AFAIK, There is no translation table, you need to search on the Internet or interpolate it from different screens/vehicles. We can open a page on the Wiki to put the _already found_ translations, though.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836