cedricp/ddt4all #134: DDT4ALL Won't Read Most... (Says No Data)...

StewartPS, 2019-01-22:
Hi all, I know this is a subject that's been asked before but have searched and haven't been given an answer that solved my issue yet. I've got an ELM327 OBD to USB device. I've assigned it to COM1, laptop loads and runs DDT4ALL fine and the device is connected. It brings a list of different ECU's at the side fine and i'm able to click on each sub-menu... However everything is red pretty much and just displays 'No Data'..

What I am confused about, is when I go into the Navigation menu, and enable Android auto it works fine. I'm able to reset the Rlink system, I can also read and write to different RS Slots. This however is the only page I can read and write data to. All other pages just show no data.

[image]

I've double checked all the pin configuration and it's all correct, I've ran a benchmark test as well and it produced the following result:

_Result: 76 succeeded from 86
ELM Max version: v1.4

Note for STN11xx -interfaces:
Check the STN firmware update on
www.scantool.net/scantool/downloads/
For Media CAN (Can Line 2) on STN1170: Connect OBD pin 3 with13 and 11 with 12.
In the folder ddt4all/tools you will find a terminal and a port finder._

[image]

I'm a little stuck on what to do to be honest, I've also tried increasing the timout duration as well but that didn't do anything. Any help would be hugely appreciated, thanks.

cedricp, 2019-01-23:
You should share your log files so we can figure out what's happening...

StewartPS, 2019-01-28:
Here's a link to the 3 files in the logs folder:

[logs.zip](https://github.com/cedricp/ddt4all/files/2802611/logs.zip)

Mikerhinos, 2019-03-04:
I have the same problem.
Car is a Twingo 3 from 2015 (looks like the same R-Link than ZOE where it's possible to activate Android Auto this way).
I tried to connect with my OBDkey which isn't working, so I bought ELM327 USB from tutorial.
If I scan ECUs, looks like it finds one but it's not exactly the same, is my DB too old ?
Anyone could point me to a recent one ?

mishu28nmv, 2019-03-08:
The most recent DB I could find: https://drive.google.com/open?id=105SA8as4jW9mqkHvIONcNCCDmI7r1lac

Mikerhinos, 2019-03-09:
> The most recent DB I could find: https://drive.google.com/open?id=105SA8as4jW9mqkHvIONcNCCDmI7r1lac

Thanks ! Will test it as soon as I can :)

moutonjr, 2020-11-07:
Hi, any update or troubleshoot guide for this issue ?

mishu28nmv, 2020-11-07:
There are 2 ways to connect to R-Link:
1. When you connect to activate Android Auto
2. When you connect to change other setiings like Folding Mirrors, Auto Dim Beam Lights etc.
When you connect for of those, the options for the other will show no data.
Check this https://www.gps-rlink.com/r-link-1/activate-android-auto-waze-on-r-link-1/

[image]
[image]
[image]
[image]

moutonjr, 2020-11-07:
Hi, @mishu28nmv , thank you very much for your answer.
It's rather a nice-to have feature on ddt4all. Let me explain:

If I use ddt4all with normal ELM ( no cross cable) directly on car OBD, I should be able:
* To test if PC-ELM connection is right ( which is implemented already at ddt4all port selection)
* To test if ELM-car is working as well: cable/OBD/ battery/computer issues. I'm not sure how to test this second part. ( Test if there is data exchanged?)

≠=======
This is general use case.

I'm Indeed trying to connect to CAN2 for Android auto tweak. When setup (elm+wire cross) ,ADAS screen yields "no data", yet this is hard to troubleshoot: bad rewiring, driver issues, bad ECU?

Thank you,

mishu28nmv, 2020-11-08:
I hope you solve this. This is over my head 😂

Andrei-hrz, 2022-09-16:
> I hope you solve this. This is over my head 😂

Can you help me pls. I have same problem :))). I am from Ploiesti if you are close to me I can came to you to activate it :P

mishu28nmv, 2022-09-16:
> > I hope you solve this. This is over my head 😂
> 
> Can you help me pls. I have same problem :))). I am from Ploiesti if you are close to me I can came to you to activate it :P

I no longer live in Romania. Sorry. But in Ploiesti are good guys who can do this. Just search Laguna forum!
Cheers!

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif