cedricp/ddt4all #136: renault Kangoo 2009 Cruze Control

canavarkedi, 2019-01-23:
Hi all 2009 model renault kangoo II from the TDB I opened the cruz control. However, when I scanned my tool, I didn't find a section that would open a cruze control in the injection code DCM12-L4DB5 -v12.0. Will you have any work on the injector computer with code DCM12-L4DB5 -v12.0?

cedricp, 2019-01-25:
Do you have the necessary hardware to make it work ? Is it correctly installed ? I had only to enable it on the cluster ECU on my Megane II 2003

canavarkedi, 2019-01-27:
Yes my device 2009 model renault kangoo 2. When I have done an automatic search I found an area where I can open or close the cruze conrol from the L12B5-v12.0 injection.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif