cedricp/ddt4all #139: Activate Extended Grip in Kadjar

jorgos3, 2019-01-28:
I'b bought Extended Grip selector and there is a connector available under center console. When I connect it the orange light flashes in selector and then nothing. Is it possible to active the Extended Grip function so that the selector would work?

gordon6, 2019-02-07:
Hi, when you find Audio>RadNav x.x and ABS-VDC-ESP>ESC_1540-HFE-XGZ-x.x then you can change 2 values.
1. RadNav > ECU config - Part II > HMI > Setting > extended grip > Present
2. ESC_1540-HFE-XGZ > Config > Function Config > EG(extended grip) > Activated
If you need to reset, just click the Hard Reset...
I drive Koleos2, so, I didn't try this configurations...  ^^;

caccolina, 2019-07-25:
Thanks for the topic. Has anybody tried the solution suggested by Gordon6 on the Kadjar ? thanks

bobcho, 2019-08-16:
> Thanks for the topic. Has anybody tried the solution suggested by Gordon6 on the Kadjar ? thanks

I'll be trying it - looking to purchase the switch

maty307, 2020-08-19:
hello will someone give me a link where to buy the Egrip rotary knob

bobcho, 2020-08-19:
I have one for sale, didn't get the chance to try it out.

maty307, 2020-08-23:
and what would be the price?  you can put some photos here, I'm from the Czech Republic.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif