cedricp/ddt4all #141: Timeout after setting protocol AT SP 6,

jj100, 2019-02-07:
Hi all, I have a problem to connect with ddt4all to any device on CAN2, 

1/ After connect to RENAULT MEGANE III   file elm_ddt.log reports   "INIT CAN"
#[01/27/19 15:59:37.703] Init CAN
2/ Then I try to connect to any device and always I get timeout after setting protocol to 6 
Is it possible to solve this problem without modifying the code?

#connect to: ESP - X95 Version 1.0.4 Addr:01
>[15:59:37.763]AT SH 740         Set Header to 740
<[0.01]AT SH 740
OK
>
>[15:59:37.773]AT CRA 760       Set Receive address to 760
<[0.01]AT CRA 760
OK
>
>[15:59:37.783]AT FC SH 740    Set the header to 740   (Flow Control)
<[0.02]AT FC SH 740
OK
>
>[15:59:37.803]AT FC SD 30 00 00     set data to 30 00 00
<[0.01]AT FC SD 30 00 00
OK
>
>[15:59:37.813]AT FC SM 1       Flow control set mode to 1 
<[0.01]AT FC SM 1
OK
>
>[15:59:37.823]AT SP 6
<[0.017]AT SP 6                   Set protocol to 6 and save it
OK
>
>[15:59:37.843]0210C0
<[5.002]0210C0
CZAS PRZEKROCZONYTIMEOUT
>[15:59:43.295]023E01
<[5.001]023E01

cedricp, 2019-02-07:
Why do you use **CAN2**, ABS-ESP is wired on **CAN1**.
Have you checked your ELM compatibility ?

jj100, 2019-02-07:
Thank you for your response.
Attempting to connect to any device gets timeout
For example
#connect to: R2_08 v3_9 Addr:13
.......................................................
>[16:02:08.367]AT SP 6
<[0.017]AT SP 6
OK
>[16:02:08.384]0210C0
<[5.002]0210C0
CZAS PRZEKROCZONYTIMEOUT
>[16:02:13.428]023E01
<[5.003]023E01
CZAS PRZEKROCZONYTIMEOUT

#connect to: SID305_RM6_9D0_B50_C90_D00_E00_V02 Addr:7A
............................................................................
>[11:44:02.489]AT SP 6
<[0.031]AT SP 6
OK

>
>[11:44:02.520]0210C0
<[5.008]0210C0
CZAS PRZEKROCZONYTIMEOUT
>[11:44:08.058]023E01
<[5.008]023E01
CZAS PRZEKROCZONYTIMEOUT
......................................................
#[11:56:17.114]KeepAlive
>[11:56:17.114]0210C0

Nearest weekend I will carry out further tests.
So  I will be able to more  accurately describe the problem.

cedricp, 2019-02-07:
A _timeout_ means that there's no ECU response, is generally caused by :

- Bad (clone) ELM chip
- Bad connection to the vehicle

shrlnm, 2019-02-07:
You may check ELM wiring and CAN connectivity by monitor mode. Can you catch any system frames or not?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif