cedricp/ddt4all #142: Change Vin Rlink2

franeti, 2019-02-08:
Hi, I would like to change my radio rlink2 2015 for the rlink 2017-18 from Renault. I have located one in a "scrapping", but before I remove the radio I know that I have to change the vin or something like that.

With ddt or even in dev mode, how can I do it? Is there a section in ddt to change for my vin before I remove the radio from the sinister car?

Thanks in advance for your support.

Regards.

shrlnm, 2019-02-08:
You may change VIN after "virgin mode"
https://github.com/cedricp/ddt4all/issues/113

franeti, 2019-02-08:
Ok. I have seen virgin mode in dev mode. Is it the best option before I remove the radio?

And later, when I install the radio in my car, how can I type my vin?

Thanks!

ohayyam, 2019-02-08:
You can try with DDT that need some modififations to able to write VIN. We tried before but could not succes :)) On the other hand if VIRGIN MODE means like low level format of Audio ECU then you have to go to official Renault service to reprog it and reconfigurate under renault net after virginised. 

@shrlnm Did you try reprog with reprog DVD with renault clip probe? If yes as you know latest reprog DVD 174 available on the internet as iso file to download. Unfortunately reprog 174 DVD dont have any Rlink2 Audio data because of its big size. Thats why during audio reprog at renault service they are using R-USB to reprog Audio ECU. Where we can get Audio ECU data ? Is it possible to download it from infotech with service manager token ?

Thanks & Regards

shrlnm, 2019-02-08:
@franeti I suggest not to try to change the VIN for feng shui reaching. I not sure that R-Link2 is locked on it. You may write new VIN  into rlink only after deleting the old one. It is only known way to do it is to press "virgin mode" button. But together with the old VIN you will delete all other configurations. Most difficult to repair from them are acoustic parameters sets which are about 2000 bytes of binary data each. 
I suggest just to save main parameters of car from the old r-link and set them up on the new one with ddt4all. Just in case of trouble with that way we may try others.

shrlnm, 2019-02-08:
@ohayyam what software do you mean? R-Link2 contains at least 3 CPU with own software. Main CPU which is running android and you may upgrade it by yourself without REPROG DVD. https://easyconnect.renault.co.uk/software-upgrade/r-link2
Next CPU is an NXP7755 audio micro-controller responsible for sound processing, ESE,  noise cancelation and so on. It may be reprogrammed from the main CPU. The last one is a MICOM chip responsible for CAN conversation. I suppose that the last one may be reprogrammed thru the CAN bus. It has about 700k bytes own firmware but I do not know if the REPROG DVD has firmware for MICOM or not.

shrlnm, 2019-02-08:
>  if VIRGIN MODE means like low level format of Audio ECU then you have to go to official Renault service to reprog it and reconfigurate under renault net after virginised.
> 
The R-LInk2 may be completely repaired by the ELM327 without the clip but it is quite complex work.

ohayyam, 2019-02-08:
@shrlnm Thanks for Quick reply. i have 2016 December fabricated Talisman and my rlink2 version 7.0.24.136 upgraded already  which is not issued yet officialy ( I have lgu file i can update myself 👍)

I can reprog all ecu with DVD except audio as explained above thats why i asked where we can get reprog data of audio?  in case damage or failure of radio to repair with clip. 

On the other hand rlink2 V7 have RES remote engine start but i could not able to activate it with ddt and clip with modify mtc file. Can you help me for this subject also ?

Regards

franeti, 2019-02-08:
I think if I use virgin mode I will go to renault to program it. 

But if I change the vin maybe I dont need renault. 
Where is the specific site to see and change the vin through ddt?  Thanks!

shrlnm, 2019-02-09:
@ohayyam do you know the vin of any talisman with RES enabled on a plant?
did you try to enable RS-monitor? V7.0.24 should have it inside

ohayyam, 2019-02-09:
@shrlnm I tried RS-monitor from Dev mode but nothing happened. I mean not worked. i can send some photos if you give me your e-mail adress ? 

I never hear any success on RES for Talisman but i know Mgn3 look that video on youtube. He used only DDT4 He said me he did some software modification on DDT4 👍 

https://www.youtube.com/watch?v=LPq0YwMq8B0&t=67s

shrlnm, 2019-02-10:
@ohayyam let's move to another thread for not to be off-topic here. But I tried to enable RES on talisman a month ago without success. I suppose that the problem may be in wiring scheme or in firmware version of ECUs of talisman and it just not support RES functionality. We need at least one working example for find configuration differences.

ohayyam, 2019-02-10:
@shrlnm ok open it please :)

franeti, 2019-02-10:
I have located the section in DDT to change the VIN of the radio before disassembling it from a damaged car.
Ecu Audio. Radnav-Diagnosis-Identification. 

Do I need anything else before I remove the radio?

Thanks in advance

Regards.

ohayyam, 2019-02-11:
[image]

what about serial number of Radio?  I think after replacement of Radio you have to reconfigure your radio under renault.net at official Renault service ?

franeti, 2019-02-13:
To configure the radio, i think you can take screenshots from your original radio (ecu audio) 

About the huid I think yes, its neccesary with renault to change it.

Sig2018, 2019-05-09:
@ohayyam Could you help on buying the canclip I've seen many Chinese versions and price. Which one do you have? Was you able to play with any ECU? I would also like to update the MICOM ECU(Radio can bus interface).Thanks.

aqngelrock, 2020-10-09:
> @shrlnm Thanks for Quick reply. i have 2016 December fabricated Talisman and my rlink2 version 7.0.24.136 upgraded already which is not issued yet officialy ( I have lgu file i can update myself 👍) [...]

https://www.youtube.com/watch?v=8NBT9UKZ248&feature=youtu.be&fbclid=IwAR3dvLajBi5SJhFNUoi-eZaYAAO-wUfJ_2hWB5t-ARk8n7lwxnxgkCd4FB8

MiqdadAli2002, 2022-04-17:
Hello everybody 
My R link 2 has been taken to virgin mode
The Huid number has been changed 
I took my car to Renault agent in my country but unfortunately they couldn’t give me an access 
I need your help If you have any email of Renault Group can help me with this problem 
Thank you

AlMeu, 2022-04-18:
Do you need a contact of Renault dealer or a help to program the unit to install in your car?
Renault dealer in your area you can find even google it

MiqdadAli2002, 2022-04-18:
Not that 
I mean that my R link 2 VAN number has been changed 
Now I can’t use R link 2 toolbox without access while the new r link 2 Van number (huid)
The agent in my country can’t give me this access

AlMeu, 2022-04-18:
VIN number of unit must be the same if you want access your old toolbox and mus tbe the same with VIN of car.

MiqdadAli2002, 2022-04-18:
My problem is that I don’t have the old VIN number of R-Link 
Because I selected the virgin mode by mistake 😩

MiqdadAli2002, 2022-04-18:
Is there anyway to know the old VIN number or to get access with the new number?

AlMeu, 2022-04-18:
The VIN number is on chassis of car and is unique for car or at base on windshield. This VIN number is written on all modules of that car.

MiqdadAli2002, 2022-04-18:
I didn’t mean the VIN number of the car
I mean the VIN number of R-Link2 ( HUID Information)

MiqdadAli2002, 2022-04-18:
Because after Virgin mode, It has been changed 
And I don’t have the previous one

MiqdadAli2002, 2022-04-18:
I mean
 Huid information = VIN number of R-Link2 = serial number of radio

AlMeu, 2022-04-18:
That you information is not writeable as I know.
So you have to leave what unit gave you.

MiqdadAli2002, 2022-04-18:
But with it I can’t update my maps

AlMeu, 2022-04-18:
If you have same VIN in your unit like old one and pay for subscription perhaps you can upgrade the maps.

MiqdadAli2002, 2022-04-21:
Unfortunately I don’t have the old VIN for unit 
Is there anyway to know the old VIN ?

AlMeu, 2022-04-21:
Old VIN on old unit is meaning the VIN of your car. This VIN you can find in all places of your car, even in your car documents.

MiqdadAli2002, 2022-04-21:
How that ?
Huid information (VIN number of unit ) started by CAPSD
AND VIN number of car started by VF1R

AlMeu, 2022-04-21:
Write everything corect what you can write and don't look at what you don't understand, what not even how is build and how is composed. While you can't not even touch that number why you lose so much time insisting?

MiqdadAli2002, 2022-06-21:
> Write everything corect what you can write and don't look at what you don't understand, what not even how is build and how is composed. While you can't not even touch that number why you lose so much time insisting?

Can you email me please 
m-a-m88@msn.com

MiqdadAli2002, 2022-07-21:
> Write everything corect what you can write and don't look at what you don't understand, what not even how is build and how is composed. While you can't not even touch that number why you lose so much time insisting?

Hello my dear 
Finally I found the problem 
My problem is that 
When I connect flash memory to my car Rlink2 send Tow empty files without wls folder
Do you know how to solve this problem?

MiqdadAli2002, 2022-08-05:
> 
..

MiqdadAli2002, 2022-08-05:
> Write everything corect what you can write and don't look at what you don't understand, what not even how is build and how is composed. While you can't not even touch that number why you lose so much time insisting?

.

tolik1978, 2023-02-18:
ск,ажите у меня такая же проблема,не могу записать vin файлы пустые без wls

tolik1978, 2023-02-18:
> > Напишите все, что вы можете написать, и не смотрите на то, чего вы не понимаете, даже на то, как создается и как составляется. Хотя вы не можете даже не коснуться этого номера, почему вы теряете так много времени, настаивая?
> 
> Привет, моя дорогая, наконец, я нашел проблему Моя проблема в том, что когда я подключаю флэш-память к своей машине, Rlink2 отправляет пустые файлы без папки wls Вы знаете, как решить эту проблему?

добрый день вы не решили проблему?

shrlnm, 2023-02-18:
Если vin пустой, то он легко записывается. Дайте больше деталей, как и что вы записывает? Логи  попыток есть?

xpomar, 2023-02-27:
> 

i have black screen rlink 2 after updating from 2.2.19.300 to 3.3 any help i have DD4ALL and ELM327  (even usb don't work and reboot each 15 second (dashboard blink )

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

johnkar87, 2025-03-22:
> [@ohayyam](https://github.com/ohayyam) what software do you mean? R-Link2 contains at least 3 CPU with own software. Main CPU is running android and you may upgrade it by yourself without REPROG DVD. https://easyconnect.renault.co.uk/software-upgrade/r-link2 Next CPU is an NXP7755 audio micro-controller responsible for sound processing, ESE, noise cancelation and so on. It may be reprogrammed from the main CPU. The last one is a MICOM chip responsible for CAN conversation. I suppose that the last one may be reprogrammed thru the CAN bus. It has about 700k bytes own firmware but I do not know if the REPROG DVD has firmware for MICOM or not.

@shrlnm can you tell me tha name of the main chip rlink2 ? so can you send me clear photos if you have from the mainboard of the rlink2?

shrlnm, 2025-03-23:
> [@shrlnm](https://github.com/shrlnm) can you tell me tha name of the main chip rlink2 ? so can you send me clear photos if you have from the mainboard of the rlink2?

@johnkar87 I do not know the exact name of the main chip. But it's Renesas with V850 compatible core.
And I have no clear photos of the main core.  My only rlink2 is in the car.

johnkar87, 2025-03-23:
what do you mean "
Renesas with V850 compatible core."