cedricp/ddt4all #143: Can't write/send values on EMM_EDISON_DDT2000_SW10_V2

sirbig, 2019-02-10:
Hello,

After scanning and searching for ECUs, I got 4 green ones that I can read and write properly.
I got other 4 ECUs that are red and one of them I'm pretty sure is correct for my Clio IV, EMM_EDISON_DDT2000_SW10_V2.

I want to change values on the EMM_Lighting section but I get an error after sending the values and it changes nothing, It can read everything correctly but it can't write anything.

I will provide a printscreen and logs for further help, thanks!
[logs.zip](https://github.com/cedricp/ddt4all/files/2848412/logs.zip)

shrlnm, 2019-02-10:
Try with 
T4_EMM_DDT2000_SW8.4_V1_20121214T153704.xml

sirbig, 2019-02-10:
> Try with
> T4_EMM_DDT2000_SW8.4_V1_20121214T153704.xml

I will try it later tonight and will reply back, thanks!

sirbig, 2019-02-10:
I just tried and no luck, actually even tried with all the T4_EMM_DDT2000 versions listed with my database an all display the same erro:

Everything in the READ column goes red with "NO DATA" after I try to change and send the values. I feel like I'm so close to make it work but there is something stopping me register values on any EMM.

sirbig, 2019-02-10:
The only progress I made using T4_EMM_DDT2000_SW8.4_V1 was the fact that it actually writes the value but then what I described above in my last reply, happens and nothing is changed on my car, per example, follow_me_home doesn't work still even if the value is set to 1 (after sending with error).

sirbig, 2019-02-10:
This is my ELM327:

https://www.aliexpress.com/item/Original-ELM327-Mini-V2-1-ELM-327-OBD2-Bluetooth-Interface-Auto-Scanner-obd-ii-Diagnostic-Tool/32246541705.html?spm=a2g0s.9042311.0.0.27744c4dS9BKgs

shrlnm, 2019-02-11:
Could you please show the logs? They are much more interesting then elm photos)

sirbig, 2019-02-11:
Here you have it... again. 👍 
[logs.zip](https://github.com/cedricp/ddt4all/files/2850490/logs.zip)

shrlnm, 2019-02-11:
@sirbig ok, the log is good and your ELM is not guilty. Try to increase a timeout at least up to 400ms. The xml T4_EMM_DDT2000_SW8.4_V1_20121214T153704.xml is looking suitable for your EEM.

@cedricp Hi, I suppose that this problem may be solved if to add ATST timeout reinitialisation during changing CANid in set_can_addr

```
        if canline == 0:
            self.cmd("AT ST FF")
            self.cmd("AT AT 0")
            if 'brp' in ecu.keys() and ecu['brp'] == "1":  # I suppose that brp=1 means 250kBps CAN
                if CANEXT:
                    self.cmd("AT SP 9")
                else:
                    self.cmd("AT SP 8")
            else:
                if CANEXT:
                    self.cmd("AT SP 7")
                else:
                    self.cmd("AT SP 6")
            self.cmd("AT AT 1")

```

sirbig, 2019-02-11:
Still the same issue with 400ms, I also tried with 1000ms but it does not apply any change.

shrlnm, 2019-02-11:
Could you post New logs with extended timeout?

sirbig, 2019-02-11:
Sure, sorry about that.

[logs.zip](https://github.com/cedricp/ddt4all/files/2853434/logs.zip)

shrlnm, 2019-02-12:
Thank you. Sorry, I was not very attentive before and did not notice some important details.
The problem is more complex then I thought before

Below is my log analysis 
```
>[12:52:34.254]100B3B63181F6064  << start of sending command 3B6318 bla bla 
<[0.438]100B3B63181F6064
3001050000000000    <<-- ECU ask us to send next one frame in 5 ms

>
>[12:52:34.692]212000640F00  <<-- we send second frame
<[0.437]212000640F00
NO DATA                  <<-- no response from the ecu during next 400 ms

>
>[12:52:35.230]022164    <<-- we send next command 2164
<[0.765]022164
037F3B2300000000         <<-- ups, it is NR to first command 3B. 23 means "Routine Not Complete"

```

But a day before you had one successful try. Look at ecu* log

```
21:19:59.318;74D;3B 63 18 1F 60 64 20 00 64 0F 80;7B 63   << success
21:20:10.574;74D;3B 63 08 1F 60 64 20 00 64 0F 80;    << fail

```

My recommendations:
1) try cfc mode
2) use usb-elm with higher com boudrate

If it doesn't help then try pyren as so it has a bit more advanced ELM library.

sirbig, 2019-02-12:
First of all thank you for taking some of your time to look into this issue! 👍 

How could one command be successful when others always fail? That's awkward.

Which one of these ELM would you recommend me to buy?

http://www.tekshop.pt/produto/cabo-diagnostico-elm327-1-5-c-software-can-bus/
http://www.tekshop.pt/produto/cabo-diagnostico-auto-elm327-v1-5-obdii-can-bus-usb/

sirbig, 2019-02-12:
Btw, what is cfc mode?

shrlnm, 2019-02-12:
I avoid to recommend any ELM to buy except original one as so there are a lot of fake devices with constrained command set. But If I were  your  I would prefer aluminium box.  Haven't seen them available for buying for ages and I've never seen fake in such box. 

Regarding the command there is nothing awkward as so it may depends on current load of ECU.

For enabling cfc mode, find the line 
opt_cfc0 = False
in file options.py and change it on True

sirbig, 2019-02-12:
That's what the seller said about the aluminium box one, altough it's a bit old it's still a good one to use since it may get the job well done.
Meanwhile, I will give it a try regarding cfc mode and will update you shortly.

sirbig, 2019-02-12:
I just tried with opt_cfc0 =True, and apart from some different commands shown on console, everything still the same. I also tried with 0ms and 400ms delay.

Should I change this parameter back to False?

[logs.zip](https://github.com/cedricp/ddt4all/files/2855975/logs.zip)

cedricp, 2019-02-12:
It could be a good idea to try with an OBDLink (USB), which has a faster chip (STN1110) and more memory. It's not expensive and more reliable in my sense.

shrlnm, 2019-02-12:
```
>[12:52:34.254]100B3B63181F6064
<[0.438]100B3B63181F6064
3001050000000000

>
>[12:52:34.692]212000640F00
<[0.437]212000640F00
NO DATA                                      << didn't wait 

>
>[12:52:35.230]022164
<[0.765]022164
037F3B2300000000              << NR response to first command

```

try to increase time out step by step 500мс 600 700 800 900

actually we are completely  send whole command but didn't wait the response and interrupted command execution by the next command 2164 

If we would not send 2164 immediate after second frame then first command should take affect

shrlnm, 2019-02-12:
Recent versions of pyren have the next algorithm.

```
        if Fn > 1 or len(raw_command[0])>15: # set elm timeout to 300ms for first response
          self.send_raw('ATST4B')

        while Fc < Fn:

            # enable responses
            frsp = ''
            if not self.ATR1:
                frsp = self.send_raw ('at r1')
                self.ATR1 = True
            
            tb = time.time ()  # time of sending (ff)

            if Fn > 1 and Fc == (Fn-1):  # set elm timeout to maximum for last response on long command
                self.send_raw('ATSTFF')
                self.send_raw('ATAT1')

```

But it would not work in case of NR:78 or when maximum ELM timeout is not enough.

sirbig, 2019-02-12:
I just purchased the ELM327 1.5a aluminium box, which is a real product according to this local shop.
After I do all the tests with that new CAN-BUS I will give you some updates.

Meanwhile, should I change opt_cfc0 back to False?

It would be nice if ddt4all could predict the timeout time response according to the commands automatically.

shrlnm, 2019-02-12:
Yes, you'd better change opt_cfc0 back to False

shrlnm, 2019-02-12:
This option is very helpful in some rare cases and it should automatically switch on in case of internal buffer exhaustion in ELM

sirbig, 2019-02-13:
Well, just got that ELM327 metal box and everything still pretty much the same...
Also ran a bunch of tries with sending and refreshing the values with and without cfc and trying 400ms, 500ms, 600ms and so on until 1000ms.

@shrlnm Would appreciate if you find something different in this logs using this new ELM327 metal box. Also, I have a person in the Reanult Clio IV group on facebook that has the same ELM327 metal box but uses a different version of ddt4all  (an older one) where pretty much all the ECUs detected by ddt4all on his car are the same as my car and it works with him with no issues.

[logs.zip](https://github.com/cedricp/ddt4all/files/2861325/logs.zip)

shrlnm, 2019-02-13:
@sirbig Hi, a few last tries were successful

```
16:31:57.867;74D;3B 63 18 1F 60 64 20 00 64 0F 80;    << fail
16:33:06.448;74D;3B 06 80 00 3C 00 0F 00 DC 05 DC 28 64 2E;7B 06 << success
16:38:34.075;74D;3B 06 80 00 3C 00 0F 00 DC 05 DC 28 64 2E;7B 06 << success
16:42:46.687;74D;3B 63 18 1F 60 64 20 00 64 0F 00;7B 63 << success
16:43:14.402;74D;3B 63 18 1F 60 64 20 00 64 0F 00;7B 63 << success

```
these commands were sent with 900ms timeout.

It is strange but there is one successful try without timeout at all

```
>[13:13:59.821]100B3B63181F6064
<[0.093]100B3B63181F6064
3001050000000000

>
>[13:13:59.914]212000640F001
<[0.079]212000640F001
027B630000000000

```

sirbig, 2019-02-13:
Oh sorry, I should have let you know that. Those successful tries were the ones that I didn't change any value and just clicked on the "Send" button. Everytime I change a value and try to send the problem continues.

Also, I noticed that and the end of the ELM Benchmark, It says to update the firmware and change some pins from one position to another... Is this relevant at all?

shrlnm, 2019-02-13:
@sirbig Which option are you going to activate?

sirbig, 2019-02-13:
Cornering and Lighting sections.

shrlnm, 2019-02-13:
Does your vin ends with 915?

sirbig, 2019-02-13:
Yes it does.

shrlnm, 2019-02-13:
Have not found no one X98 with cornering enabled. Are you sure that every fog lamp has own wiring? Could they be switched on separately?

sirbig, 2019-02-13:
I'm not sure about that. But even though Cornering doesn't work, I'm pretty sure settings on Lighting section should work. Simple changes like turning off/on DLR or Follow_Me_Home should work.

shrlnm, 2019-02-13:
Did you try pyren? It is interesting to compare logs and results of writing such difficult commands. It may give us a hint if there is a way to improve ELM library or it will be just waste of time

sirbig, 2019-02-13:
I'm not familiar with pyren but I will give it a try.

sirbig, 2019-02-13:
Ok so here is what I've done.

1. Uninstalled the ddt4all and installed a new one from drive2.ru which is the version 5.3.5 according to what they say.
2. Ran everything just like the old ddt4all that I had and still exactly the same issues.

[logs_ddt4all.zip](https://github.com/cedricp/ddt4all/files/2862573/logs_ddt4all.zip)

3. Ran PyREN and tested everything on the DDT option just like I did on the ddt4all screen and it sent the values, shows the changed values on the READ column but no changes are actually made on the car. Also, sometimes everything went "NONE" for some reason and I had to restart PyREN.

[logs_pyren.zip](https://github.com/cedricp/ddt4all/files/2862576/logs_pyren.zip)

4. Is [image] thing relevant at all for my case?

I'm about to give up on this ECU, almost seems like it is bricked and not allowing to make changes on the EMM.

shrlnm, 2019-02-14:
@sirbig Thank you very much for such valuable logs. 
With a pyren all writings were successful even with changed values.

```
22:24:24.104;74D;3B63581F60642000640F00;7B 63 
22:24:43.432;74D;3B63181F60642000640F00;7B 63 
22:24:55.354;74D;3B63581F60642000640F00;7B 63 
22:26:04.072;74D;3B0600003C000F00DC05DC28642E;7B 06 
22:26:14.119;74D;3B0680003C000F00DC05DC28642E;7B 06 
22:27:17.994;74D;3B63481F60642000640F00;7B 63 
22:41:46.100;74D;3B63481F60642000640F80;7B 63 
22:42:00.803;74D;3B63581F60642000640F80;7B 63 
22:47:41.703;74D;3B63581F60642000640F00;7B 63 
22:47:55.922;74D;3B63581F60642000640F80;7B 63 
22:48:21.234;74D;3B0600003C000F00DC05DC28642E;7B 06 
22:48:39.297;74D;3B0680003C000F00DC05DC28642E;7B 06 
22:49:34.781;74D;3B63481F60642000640F80;7B 63 
22:49:44.750;74D;3B63581F60642000640F80;7B 63 
22:57:13.376;74D;3B63181F60642000640F00;7B 63 
22:57:52.298;74D;3B63181C60642000640F00;7B 63 
22:58:20.813;74D;3B63581F60642000640F80;7B 63 
22:58:40.204;74D;3B63501F60642000640F80;7B 63 
22:59:04.313;74D;3B63581F60642000640F80;7B 63 
22:59:31.344;74D;3B63581760642000640F80;7B 63 
22:59:43.438;74D;3B63581F60642000640F80;7B 63 

```

So the algorithm with 300ms timeout before first frame and maximum timeout before last frame is looking reasonable for implementing in elm.py

```
 >[22:49:34.422]ATST4B
<[0.0]ATST4B
OK

>
>[22:49:34.422]100B3B63481F6064
<[0.312]100B3B63481F6064
3001050000000000

>
>[22:49:34.734]ATSTFF
<[0.016]ATSTFF
OK

>
>[22:49:34.750]ATAT1
<[0.016]ATAT1
OK

>
>[22:49:34.766]212000640F801
<[0.015]212000640F801
027B630000000000

>

```

You've got "NONE" values as so your computer suddenly lost connection to elm

```
>[22:20:28.420]032220051
<[0.031]032220051
0562200500D10000

>
>[22:20:28.451]032224021
<[0.016]032224021
0462240216000000

>
>[22:20:28.467]032224041
<[1.046]032224041
NO DATA

>
>[22:20:29.513]032224071
<[1.063]032224071
NO DATA

```

The information about firmware on a picture is about  STN, but you are using ELM so it is not relevant to you.
Information about PINs it is how to connect to second (multimedia) CAN bus.

And now about probably most interesting. In the file T4_EMM_DDT2000_SW8.4_V1_20121214T153704.xml I found two commands

"Write_Config_Reset"
"ECUReset.hardReset"

Usually such commands have to be sent after configuration changing for taking effect of them. But not hurry  up to send them. Let's look in clip database how they make configuration changes. Could you please share the screen shot of pyren when it shows ecus list.  For doing that you have no need to connect to car, just run it demo mode and it shows you saved data.

shrlnm, 2019-02-14:
I've filtered the logs more accurate and they shows that the command were accepted only with the second try

```
22:48:20.063;74D;;7F3B23;NR: Routine Not Complete
22:48:21.234;74D;3B0600003C000F00DC05DC28642E;7B 06 
22:48:38.125;74D;;7F3B23;NR: Routine Not Complete
22:48:39.297;74D;3B0680003C000F00DC05DC28642E;7B 06 
22:49:33.922;74D;;7F3B23;NR: Routine Not Complete
22:49:34.781;74D;3B63481F60642000640F80;7B 63 
22:49:43.875;74D;;7F3B23;NR: Routine Not Complete
22:49:44.750;74D;3B63581F60642000640F80;7B 63 

```

The pyren automatically resend the command in 500ms in case of NR:23

sirbig, 2019-02-14:
In which case was data sent successfuly? I tried sending commands in the following way:

1. Normal, just by selecting the COM5 port to connect.
2. with K-line --si activated.
3. with CAN --cfc on
4. with CAN --n1c on

Is there any need to activate does options or it works well sending commands without them?
Also, pyren did not detect **T4_EMM_DDT2000_SW8.4_V1_20121214T153704.xml** as my correct T4_EMM, I actually worked with **T4_EMM_DDT2000_SW8.5_V1.1_20160112T105701.xml** which is the one that pyren detected to be correct for my ECU.

Here it goes the printscreens:

Here I selected 8:
[image]

Here I selected 10:
[image]

Here are the results and where I want to make changes:
[image]
[image]

I didn't find those commands you talked about:

"Write_Config_Reset"
"ECUReset.hardReset"

shrlnm, 2019-02-14:
I checked the clip database and they use only ECUReset.hardReset. You may find this button in menu EMM_EnergyManagement/ECU Reset. Not sure that it helps but it is last resort I may suggest

I suggest to use only cfc option, the others not for your car and not for your ELM.

The T4_EMM_DDT2000_SW8.5_V1.1_20160112T105701.xml not too different and backward compatible, so if it recognized as yours then use it with no doubt.

Not forget to make a dump before any changes.

sirbig, 2019-02-14:
Where do I make a dump and why should I do that before any changes?

shrlnm, 2019-02-14:
The "dump" is a complete set of all parameters values. It may be helpfull for revert configuration back If somting went wrong. It also allows you to study your car offline.

sirbig, 2019-02-14:
Did it both for pyren and ddt4all and still no luck... Everything still the same... The only thing different is when I press the Hard Reset button the power goes slightly low for a split second and then stays normal again.

[logs_ddt4all.zip](https://github.com/cedricp/ddt4all/files/2866862/logs_ddt4all.zip)
[logs_pyren.zip](https://github.com/cedricp/ddt4all/files/2866863/logs_pyren.zip)

sirbig, 2019-02-14:
I can only make real changes when working with the green ECUs shown on the left side of ddt4all after CAN scan.

[image]

shrlnm, 2019-02-15:
Some options have to be enabled in different ecus and off-course the car have to be equipped with appropriate wiring, sensors, fuses and actuators. 
I absolutely sure that your EMM is from T4_EMM_DDT2000_SW8.* series

Did you try to change the parameter LIGHTING_BY_EMM_CF in BCM module?

I suppose that the problem is in level of your car equipment. For example, your car have MTC parameter SALEVE while a lot of "Lighting" options available only on cars with MTC parameter ALEVA. I do not know if this parameter means only configuration option or require some hardware equipment.

sirbig, 2019-02-15:
I did try enabling the LIGHTING_BY_EMM_CF in BCM and did a hard reset but nothing changes, if you take a look at the logs you will find that I tried. If the car comes with fog lights (some clio IV doesn't), the wiring should be connected with the ECU and I should be able to make changes on the Cornering section. There must be something bricked about this ECU.

Honestly, I'm done with this, If nothing else can be made I will just move on and let it be.

sirbig, 2019-02-16:
Btw, I just found this issue https://github.com/cedricp/ddt4all/issues/51:

> ZAFER1646 2013 yılının renault sembolü bir EMM bilgisayarına sahip değildir, bu nedenle aydınlatma eğrilerde etkinleştirilemez.
> alarm hakkında cevap yok
> Hangi bilgisayarları gördüğünüzü görmek için aracı taramanız gerekecek.

According to what @blade909 replied, Clio IV from 2013 wasn't produced with an EMM ECU, is this true?

shrlnm, 2019-02-16:
EMM is just a type of UPC ECU. Every car is equipped with UPC and yours too, as so one of ECU responding on CANid:74D
Also your ECU accept commands 3B06 "Write_Cornering_CF", so in the nature it supports the cornering as a function. But it is not the fact that the wiring scheme of your car supports it too. For example, both lamps may be managed by one line or have single fuse and so can not switched on separately.

I suppose we may check if your fog lamps have separate management lines or not. If they will be switched on separately  then there is a hope to find the way how to enable cornering
Try to run fog lamps test

```
ECU : 11137  FG2011137.XML
Screen : AC : Executing tests -> LIGHTING -> LIGHTING MANAGEMENT -> A004 [Command] 

#############################  Command parameters  #############################

Prerequisite : Ignition on and engine stopped. For vehicles fitted with automatic transmissions, position the gear lever in P or N. For electric vehicles, unplug the vehicle charging cable. When the command is finished, return to the FAULTS screen before switching off the ignition.

name         : A004
codeMR       : AC004
label        : FRONT FOG LIGHTS
type         : AC
scenario     : 
inputlist    +
serviceID    +
              : (40   ) 30142064
              : (41   ) 301411
              : (42   ) 30152064
              : (43   ) 301511
Are you ready to execute the command? <yes/no>:yes 

#############################  Execution  #############################

cmd:30142064                         rsp: 
cmd:301411                           rsp: 
cmd:30152064                         rsp: 
cmd:301511                           rsp: 

###############################  Done  ###############################

Press ENTER to exit 

```

sirbig, 2019-02-16:
If I run this test, Is it supposed to switch on/off the fog lights separately or both at the same time?

shrlnm, 2019-02-16:
30142064 - O_FOG_FRL.PWM 
30152064 - O_FOG_FRR.PWM
it should turn  them on separately for some seconds each

blade909, 2019-02-16:
> EMM es solo un tipo de UPC ECU. Cada automóvil está equipado con UPC y el suyo también, por lo que una de las ECU que responde en CANid: 74D 
> También su ECU acepta los comandos 3B06 "Write_Cornering_CF", por lo que en la naturaleza apoya las curvas como una función. Pero no es el hecho de que el esquema de cableado de su automóvil lo soporte también. Por ejemplo, ambas lámparas pueden administrarse por una línea o tener un fusible único y, por lo tanto, no pueden encenderse por separado.
> 

> ```
what you write is not correct ...
Some renault clio IV models sold in Spain, especially the models with the level of basic equipment do not have an EMM computer associated with the electric system if they do not have the start & stop system and if they use a basic gasoline engine 1.2
in these cases the electrical / electronic system is similar to the dacia sandero that lacks EMM for the management of lights
sirbig you do not cite some data that are essential to help you with this problem such as the year of manufacture of the vehicle, the level of equipment thereof and whether it is a gasoline or diesel car (these data are important to know that it can be activated and It can not be)
Regarding the response you get after the activation: "incorrect response" is normal and it has happened to those who have activated this parameter. After making the changes, have you clicked the icon to set the configuration?
I guess you have unlocked the expert configuration before applying the changes
the procedure after making the changes is to turn the vehicle off and on again
Attention
These changes must be made with the vehicle in "parking mode", ie
introduce a speed in the gear shift, step on the brake, but do not step on the clutch.
at that moment press the power button until the message "remove card" appears in the instrument panel
if it is not done in this way, the changes may not be applied correctly
It is also important to know if your vehicle has the automatic lighting system and the windshield wipers, it is verified that the vehicles that do not have these systems do not work

On the other hand, if you want to activate the "follow me home" function, I strongly recommend that you DO NOT DO IT
the version number of your EMM is less than v.14
is verified in Italian and Turkish forums that the intervention in this function in a vehicle with a smaller EMM version sw14 can lead to its headlights stop working properly and need a full reprogramming of the EMM in the renault dealer.

shrlnm, 2019-02-16:
> what you write is not correct ...

Yes, you are right. I've just checked BVMEXTARCTION database and there is about 15% of Clio IV without UPC ECU.

> sirbig you do not cite some data that are essential to help you with this problem such as the year of manufacture of the vehicle, the level of equipment thereof and whether it is a gasoline or diesel car 

I may help to find these data if it will be useful 
MNEMOCR:CEDC01/608/NOTFRA/AVARC1/TAR30/TAV36;DATE_MADC:16.06.2013;T_BOI:JR5

ECUs
2:478555940R 478557130R
3:285J24092R 285J22547R
4:284B19573R 284B36461R
5:985101389R 285A40830R
8:275319920R 275312573R
18:285041068R 285045045R
20:231A04735R 231A43205R
41:259904115R 259866902R
76:281152928R 283C26321R
99:285J96296R 285J90550R

MTC
X98;PGPRT2;ALCIAV;STDEM1;SFXBAG;SKTPOU;SKTPGR;SSCCPC;SAND41;SSPREM;SAND58;ETRP01;STOSTA;SAND72;FDIU1;AVESM;MAPSTD;SSCALL;ECOMOD;SDBRMO;SDCOSP;SSRCAR;SSTCU;PERA01;SPERTB;SPERTP;SPERTQ;SPERTR;SPERTS;SNVGAM;STANDA;X98;K98;EA2;AL;DIESEL;DG;TEMP;TR4X2;AFURGE;ABS;CA;TN;LAC;VT;EVC;CPE;RET02;PROJAB;RTOL15;CUSFIX;CEAVRH;SANCL;SACCAV;AIRBA2;SERIE;DRA;DRAP09;HARM01;ATAV;3ATRPH;SGAV03;FBANAR;TENNP;BANAL;KM;SDPRPN;AVREPL;SFIRBA;ASRESP;RDAR01;SALEVA;SOP01C;SSADPC;CTHAB1;SKTGRV;VLCUIR;TRNOR;RETC;LVAVEL;LVARMA;SASURV;RSEC01;SGSCHA;APL03;BECQAR;CENDAV;FSTPO;ALOUC;TBOR04;CMAR3P;PROJDO;NAV0G1;RAD45A;BVM5;SAN430;RDIF11;PNRR03;ISOFIX;SAN513;HRGM01;SAN521;SANFLT;TLFRAN;SAN566;SAN613;AVFAP;PLANMO;CHC03;SSAPEX;E05R;CHBASE;SANCML;SSFLEX;PE2012;PHAS1;SAN913;VEC114;X981BK;K9K

> the intervention in this function in a vehicle with a smaller EMM version sw14 can lead to its headlights stop working properly and need a full reprogramming of the EMM in the renault dealer.

It is why I suggested to make dump of configuration before doing any changes

sirbig, 2019-02-16:
I ran some tests regarding lighting and the only tests that actually worked where on BCM, the tests ran on EMM did nothing on the lights, per example, on the BCM I can test the side lights but on the EMM I can't... Pretty much every test I try to do on EMM does nothing, this makes me believe that no EMM module is present on my Renault Clio IV 1.5 dCi 90cv Dynamique S.

Also I don't have automatic lighting system nor wipping system.

[logs.zip](https://github.com/cedricp/ddt4all/files/2872418/logs.zip)

blade909, 2019-02-17:
its strange that clio dynamique dont have EMM...
If you do not have an automatic lighting system or automatic windshield wiper, it is most likely that you do not have an EMM computer to make the changes or your EMM/UPC does not accept these changes....which makes more sense
You can check visually if you have a UPC / EMM unit
It is located in the area of the driver's knees in the lower part of the steering column on the right side, near the medianav, the wiring is visible
perhaps you can try to scan with this older ddt4all version...this one worked for me.
https://cloud.mail.ru/public/4EhF/4GyGkgwR7
good luck

sirbig, 2019-02-17:
I could try to check visually if it exists but I don't know which wires I'm looking for to confirm if it is an UPC-EMM unit.

Since ddt4all couldn't work I now use PyREN to try making this changes. I will give it a try with this version of ddt4all but I'm not hopeful about it.

blade909, 2019-02-17:
[image]
one pic show more than few words this is the EMM/UPC you have to look for
PD
I have done a little research with the data of the log of your vehicle...and yes your car have EMM/UPC...
 reference 231A04735R 

[image]
but.....i have bad news....
seems there are 2 diferent equipement EMM levels L1 and L2 
it seems that this function can only be activated in LM EM2

[image]

sirbig, 2019-02-17:
Really? That would be really unlucky.... I will check it by myself and let you know which one it is. 👍

sirbig, 2019-02-19:
Indeed bad new for me, it's a L1... :(

[image]

I also tried the older ddt4all version you linked me and still the same issue, truly unlucky.

sirbig, 2019-02-20:
Can it be replaced with one L2 or greater bought on ebay?

blade909, 2019-02-20:
> Can it be replaced with one L2 or greater bought on ebay?
"theoretically" could be done but you would have to look for an EMM unit with the same characteristics in terms of engine specifications
the problem is that I would have to reprogram it completely
and I do not know if the existing wiring in your vehicle would be compatible with the new EMM

sirbig, 2019-02-20:
Ah yeah, I get it. Well, bottom line I can't make the desired changes on EMM-T4 L1, so... Case closed with a tear in the corner of my eye and less money in my pocket :'(

[4 more replies]

closed 2019-02-24 by sirbig