cedricp/ddt4all #144: Cornering Clio

PaultjeeeV, 2019-02-12:
Hello this is my last method to find solution: when i activate Nbx_Cornering_CF = 1, my fog lights should work when steering left/right. When i only activate it without further angle parameters, it doesn't work.

Then they say fill in these parameters then it should work:

Cxx_A1_angle_CF => "60"
Cxx_A2_angle_CF => "15"
Cxx_dimming_time_CF => "300"
Cxx_MAX_power_PWM_ratio_CF => "1"
Cxx_min_power_PWM_ratio_CF => "0.46"
Cxx_CorneringSpeed_CF => "40"

When i add these parameters, both fog lights go on turning right. Turning left, nothing happens.

Do someone maybe know a solution? I searched for hours and find nothing. Someone said maybe calibration of steering wheel is bad, is that possible to read out with DDT4ALL? Or even calibrate again? If i could see more where it goes wrong then i maybe can make this work. Thankyou very much.

Regards Paul

sirbig, 2019-02-12:
This is exactly the same issue as me: https://github.com/cedricp/ddt4all/issues/143

I'm able to send values on the cornering section but it doesn't apply on the vehicle as well.
If you try sending values on the Lighting section you will have the same problem as me, it will display "NO DATA" after you try to change/send any value.

PaultjeeeV, 2019-02-12:
Hi,

If i send parameter data I get first write error but then again write and then it does apply.

But when it applys and is fully inserted then my fog lights go both on turning steering wheel right. When i turn it left it nothing happens.

Do you also have only both fog lights turning right? And nothing turning left?

Regards Paul

sirbig, 2019-02-12:
Oh nevermind then, I can't even make it turn on the fog lights eventough everything is reading the values that I sent on Cornering section.

PaultjeeeV, 2019-02-13:
Hello

Try to write two times after each other and make sure you have latest version, run it as administrator, and let the car run when you connect. Press two times write when first fails. That make it writed for me.

Goodluck, hope i can solve my difficult problem to.

Regards
Paul

sirbig, 2019-02-13:
As far as I know, you should never try to write something right after another fails because it could damage the ECU sending crossed values between requests.

My hopes are now with the new ELM237 that I bought and I'm waiting to get it home and test everything out again. 

@shrlnm Should help you out with this one but you need to provide the logs located on the ddt4all folder.

sirbig, 2019-02-16:
Btw, I just found this issue https://github.com/cedricp/ddt4all/issues/51:

> ZAFER1646 2013 yılının renault sembolü bir EMM bilgisayarına sahip değildir, bu nedenle aydınlatma eğrilerde etkinleştirilemez.
> alarm hakkında cevap yok
> Hangi bilgisayarları gördüğünüzü görmek için aracı taramanız gerekecek.

According to what @blade909 replied, Clio IV from 2013 wasn't produced with an EMM ECU, is this true?

PaultjeeeV, 2019-02-16:
Hello,

Your car just needs a EMM for enabling the extra options. When you scan with DDT4ALL, it will show if you have an EMM. In DDT4ALL there is also somewhere to see EMM PRESENT : True or false. Without an EMM there is no computer to edit.

There are a couple older versions without an EMM, but if you have a light sensor, rain sensor, then it's about 80% sure you have an EMM. Cause this emm does activate and control your lights/wipers.

Hope this helps

Regards
Paul

sirbig, 2019-02-16:
Yes EMM is present. Not sure if I need to enable EMM to control external lights on BCM, even if I enable that option the cornering lights won't work lol.

blade909, 2019-02-16:
[image]
After reading your comments and helping some users of the renault clio forum in Spain, I have to make some suggestions....
PaultjeeeV one of the parameters that you have written manually is incorrect, check the picture with the correct values
if after the activation the lights only light to the right, maybe the fog lights that you have in the vehicle do not have independent power for each of the sides, due to this this fault occurs.
Maybe the fog lights have been installed after the manufacturing?

PaultjeeeV, 2019-02-16:
Hello,

Thankyou for reply.
The only other parameter is dimming time? That's what you mean?
Some cars has 300 and some 220, but i think it's just about how long the light burn and not with angles or something like that.

I didn't try anymore yet. I was hoping to find other angle parameters to try. I will try soon again some settings, i wanted it to work and I found also someone with older emm sw, where it does work.

I have fog lights from factory on it.

Thankyou, and yes I try to help others 😊😊 I'm also happy if other people try to help me to.

Regards
Paul

[cf4645ds-960]<https://user-images.githubusercontent.com/38259201/52904092-4983d280-3227-11e9-8d13-569accb6ef1b.jpg>

blade909, 2019-02-16:
> Yes EMM is present. Not sure if I need to enable EMM to control external lights on BCM, even if I enable that option the cornering lights won't work lol.

for sure EMM must be enabled for cornering option.i made some questions to you in your other post.
you have auto light system? auto wipers? if dont usually dont works cornering on clio IV

PaultjeeeV, 2019-02-16:
Yes i have auto lights and auto wipers. I even made the economic driving led working even without having rlink :), it should be smart enough to make my cornering light work to.

Someone else thought maybe my angle indicater is broken or not calibrated. But that will be more difficult. And not to do by DDT4ALL i guess.

Regards
Paul

sirbig, 2019-02-16:
#143

kazaa8, 2019-06-30:
Hello, DDT4ALL shows availalbe cotroller [EMM] T4_EMM_DDT2000_SW8.5_V1.1 and my car doesn't have option auto light system and auto wipers. If I try to send this options by DDT4ALL my controller send feedback that doesn't recognize parameter. Parameters after refresh exist (Nbx_Cornering_CF = 1) but corneting doesn't work. It's impossible to activate in my car this option?

sirbig, 2019-06-30:
@kazaa8 The conclusion for this issue that I had was to do with the fact that EMM T4 L1 does not include the module that controls cornering and some other features that are included on EMM T4 L2 and above.
You could try and buy the L2 but you have to make sure that you have the cornering sensors activated to sucessfuly communicate with the module, it's a lot of work for such a simple feature, worthless IMO.

kazaa8, 2019-06-30:
@sirbig thank you for your feedback!

sirbig, 2019-06-30:
@kazaa8 If you want more details about how I reached this conclusion, you can check [this](https://github.com/cedricp/ddt4all/issues/143) thread that I created in the past discussing with the people envolved in this amazing project.

myshuker, 2023-04-24:
I have EMM_EDISON_DDT2000_SW15 (Green) in my Clio 4 from 2018, but also I cant activate it, nothing works

AndyIreland, 2023-04-24:
so you change this to '1' and press send yep and what does it do when you refresh the screen go back to '0' does it? 

[image]

sirbig, 2023-04-24:
> I have EMM_EDISON_DDT2000_SW15 (Green) in my Clio 4 from 2018, but also I cant activate it, nothing works

Well, the conclusion for this problem was very simple... If you change the parameters and it resets to it's default values or it saves and it doesn't produce any changes, then the problem is that your vehicle is not equipped with the required module to make it work as intended.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif