cedricp/ddt4all #151: Missing ECU information - Megane IV 2019

stich86, 2019-02-23:
Hi all,

i'm using latest version of DDT4ALL, but doing a scan on my Megane IV (from 2019) i got some missing or not matched files. For example i've tried to activate Video\Photo on the run but they got back to "Restricted" after reset of R-Link. Also configuration of L/KM > KM/L doesn't seems to work.

[image]

Any hints?

Thanks!

stich86, 2019-02-24:
Regarding L/KM mod and restricted video I’ve solved. After a complete power off of machine they start to work. I think I need to wait ECU update to match my own (may be car is too new)

KarelSvo, 2019-03-09:
> I think I need to wait ECU update to match my own (may be car is too new)

No, the autoident procedure would have to be adapted to the model of the CLIP software. There, only one to two out of four ident-parameters are compared.
Then there would be better scan results.

Some ecus require a software or hardware reset to save the new configuration.

ohayyam, 2019-03-23:
@KarelSvo could you please inform us with more details.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif