cedricp/ddt4all #156: DDT4all doesn't find ELM (Bluetooth)

nickout10, 2019-03-01:
Hi guys, first of all, great job, I`ve been reading all the topics here and it`s nice to see how people help each other.

So, here it goes my issue:

I recently bought a Y splitter cable (to swap cables easily) + ELM327 1.5V Bluetooth for enabling the Android Auto on my Fluence 2016 (RLink1). I've installed the ddt4all software and inserted the Renault database into the right place, everything seemed to be great. However, ddt4all cannot reach out to ELM Bluetooth, the dongle is paired and connected to my PC, but the software doesn't connect to it. I`ve tried COM3 and COM4 at different port speeds, no success :( 

I connected the dongle to my Samsung S6 and using an app the dongle connected to the ECU correctly I could see everything about the car.

Any clue what could be made? Do I have a bad Chinese ELM clone?

Thanks!

KarelSvo, 2019-03-09:
If the Elm test also does not work, you need to fix your PC problem first.

IZ7QEB, 2019-05-20:
Hi everyone. I resume this post without opening a new one. I have the same problem.
I tried to connect my ELM, but I can't connect via bluetooth.
I tried it with my smartphone and it works. I also tried using the same laptop I'm using, with another software and it works.

What possible solutions? Are there any settings to check?
I attach screenshot.

[image]
[image]
[image]
[image]

THANK YOU.

tk5ep, 2019-07-14:
Hi,
Same for me with a Bluetooth ELM.
Works with other SW . Detected port with this SW is PORT 1, but DDT4ALL only gives me port 3 et 4...
Any guess ?
Thanks

Megane412, 2019-11-08:
Hi,

I also have a Bluetooth ELM. On OBD Auto doctor it reports ELM327 v1.5; Works fine with the software for seeing data. When I try Check ELM on pyren it all the commands return FAIL. In DDT4All I have the same problem. Tried all rate settings for connection, also moved between port 4 and 5 to no success.

I am using this adapter https://www.conrad.at/de/p/exza-obd-ii-interface-hhobd-mini-497288155-uneingeschraenkt-1650373.html

Please let me know

lmikolaj, 2020-04-09:
Hi,

I have same problem when I try connect my ELM via bluetooth.
Connection via bluetooth with OBD Auto doctor android app works.

Below is picture with problems. The dongle is paired and connected to my PC, but the software doesn't connect to it. I`ve tried COM3 and COM4 at different port speeds, no success.
 
[image]

Do you have some solution?

CRACKISH, 2020-06-24:
I have same error.

windows 10

Kingsmart75, 2022-06-14:
I also join, unfortunately after trying with USB and wifi, now even with bluetooth I can't get ddt4all to work, I've tried various versions and with two notebooks (one with w11 and the other with w10) but I always get the same result that's wrong ... with pyren and ecu-tweaker I read all the data ... before putting everything at the stake, could someone help me?
[image]
[image]

AndyIreland, 2022-06-14:
Com7 seems a bit high up - I normally get Com3 or Com4 - what is your device manager looking like in windows 10 or 11?

Kingsmart75, 2022-06-14:
in both systems I have com6 and com7, in the device manager I set the port speed to 38400, I connect perfectly with pyren ...

AndyIreland, 2022-06-14:
if it connects perfectly with PyREN why dont you use the DDT Mode in PyRen to make your changes?

Kingsmart75, 2022-06-15:
unfortunately in emm-lighting configuration I see NO DATA with pyren !!!  in other sections I see the data but in that section nothing!  with ecu-tweaker I see the data in the section but I cannot edit ...

KarelSvo, 2022-06-15:
Which vehicle is it and where are the log files and savedEcus.p file from Pyren?

Kingsmart75, 2022-06-16:
in desperation I changed the disk and put the Windows 7 operating system ... even here nothing to do ... it is not possible to go on like this ... I have bad luck on me?  I would like to know where am I wrong?  I'm banging my head everywhere !!!
[image]
[image]

Kingsmart75, 2022-06-16:
> Which vehicle is it and where are the log files and savedEcus.p file from Pyren?

I have a 2019 smart forfour aka twingo III
Pyren LOGS FOLDER
[ecu_log.txt](https://github.com/cedricp/ddt4all/files/8917359/ecu_log.txt)
[elm_log.txt](https://github.com/cedricp/ddt4all/files/8917360/elm_log.txt)
Pyren savedEcus.p
[savedEcus.p.zip](https://github.com/cedricp/ddt4all/files/8917379/savedEcus.p.zip)

Kingsmart75, 2022-06-16:
I am in absolute desperation, now I took it as a personal challenge, is it possible that I am so stupid that I can't do two encodings?  if i buy a sx obdlink USB cable will i be able to make the DDT4ALL work?  at 100%?  enough with these elm327 chinoiserie, there is no way to do anything, at least unfortunately I can't, I would like to understand what the problem is, where am I wrong?  no one here can give me some tips?

https://amzn.eu/d/7X72MLx[USB obdlink sx Amazon](https://amzn.eu/d/7X72MLx) 
[image]

AndyIreland, 2022-06-16:
its strange, I dunno what is going on with your bluetooth elm327 obd2 adaptor. Could the actual OBD2 socket on the car be faulty or blocked / locked / firewalled by smart fourfour thats what i am wondering? 

If you thought it could work I could remotely log onto your windows 10 (or 7 ) laptop through the internet and control your laptop remotely and see if anything or settings spring  to mind or stand out but doing it blindly like this is pretty difficult without me actually seeing myself what is going on with your laptop /dongle

KarelSvo, 2022-06-17:
> unfortunately in emm-lighting configuration I see NO DATA with pyren !!! in other sections I see the data but in that section nothing! with ecu-tweaker I see the data in the section but I cannot edit ...

No EMM ECU found in clip mode.
You also scanned in ddt mode.
There an EMM Ecu replied, you didn't open the screen of the EMM.

`18:53:45.334;74D;10C0;50 C0 
18:53:45.475;74D;2180;61 80 36 35 37 33 52 08 34 31 34 38 38 32 38 52 C1 52 80 01 00 00 02 01 00 88 00 
18:53:45.600;74D;2181;61 81 57 4D 45 34 35 33 30 34 34 31 59 32 32 38 32 35 34 9C D0 00 00 00 00 00 00 `

Kingsmart75, 2022-06-17:
> > purtroppo in configurazione emm-illuminazione non vedo DATI con pyren !!! in altre sezioni vedo i dati ma in quella sezione niente! con ecu-tweaker vedo i dati nella sezione ma non riesco a modificare...
> 
> Nessuna ECU EMM trovata in modalità clip. Hai anche scansionato in modalità ddt. Lì ha risposto una EMM Ecu, non hai aperto lo schermo della EMM.
> 
> `18:53:45.334;74D;10C0;50 C0
> 
> 18:53:45.475;74D;2180;61 80 36 35 37 33 52 08 34 31 34 38 38 32 38 52 C1 52 80 01 00 00 02 01 00 88 00
> 
> 18:53:45.600;74D;2181;61 81 57 4D 45 34 35 33 30 34 34 31 59 32 32 38 32 35 34 9C D0 00 00 00 00 00 00 `

sorry, so where am I wrong?  can you give me the right steps to take?

Kingsmart75, 2022-06-17:
> è strano, non so cosa sta succedendo con il tuo adattatore bluetooth elm327 obd2. La presa OBD2 effettiva dell'auto potrebbe essere difettosa o bloccata / bloccata / protetta da firewall da smart fourfour, ecco cosa mi chiedo?
> 
> Se pensavi che potesse funzionare, potrei accedere in remoto al tuo laptop Windows 10 (o 7) tramite Internet e controllare il tuo laptop in remoto e vedere se qualcosa o le impostazioni ti vengono in mente o risalta, ma farlo alla cieca in questo modo è piuttosto difficile senza di me in realtà mi vedo cosa sta succedendo con il tuo laptop / dongle

the obd port blocked I don't think, ELM read the data also with other apps, do you think there is a block from the Smart center?  the PC is virgin, I took a new SSD and installed the operating system, I didn't put anything else, I installed DDT4ALL and PYREN.  am I wrong something about the steps of the software?

Kingsmart75, 2022-06-17:
here I am again, I formatted the notebook and installed Windows 10, same thing, DDT4ALL does not work with the ELM Bluetooth, with pyren it goes but I cannot see the data of the ECU emm-lighting configuration.  @KarelSvo I'll put you the pyren logs with Scan clip mode.  at this point i am determined to complete the mission, can someone tell me what it takes to make ddt4all work?  1. notebook: operating system version?  2. cable that 100% works?  3. various and possible?  Thanks everyone for the support.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

IZ7QEB, 2022-06-26:
Ciao @Kingsmart75  .. Assicurati di avere una versione "originale" dell' ELM. A me è successo con una versione fake e non funzionava. Ne ho comprata una adatta su amazon e tutto ha funzionato. 
Successivamente, accertati che il tuo antivirus e il tuo firewall non ti blocchino l'applicazione e di conseguenza il dialogo tra dispositivo e applicazione. Mi è successo e dopo aver escluso dalle protezioni ha funzionato tutto.

Kingsmart75, 2022-07-09:
just for information that with the OBDLINK SX usb cable everything works without problems, I trashed all the fake elms that did not go in any way.  it is better to spend a little more but in the end you save time and health for all the pisses you have had ... damn fake elm !!!

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

afmovement, 2023-06-30:
elm