cedricp/ddt4all #159: help me

mitakbas, 2019-03-06:
I've entered the developer mode but accidentally I've touched the first top right option related to 'Screen display', I think it's switched to 'high'...
Now the screen is off (blank), the functions are working but the screen is blank.
I tried more than once to switch off the engine, restart the display unit, and even i tried to pull out the fuse, but nothing worked...
Please help me
 renault kadjar

gobo-ws, 2019-03-07:
https://gitlab.com/py_ren/pyren/wikis/how%20to%20rlink2%20black%20screen

midoucirine, 2019-05-30:
es que ça marche pour megane 4 aussi ? merci

shrlnm, 2019-05-31:
> es que ça marche pour megane 4 aussi ? merci

yes

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif