cedricp/ddt4all #160: grand kangoo2016 connection problem

Fischmaster3000, 2019-03-07:
I got the ecu database from a russian site, stated as feb2019. I also got a OBDLink SX and a Y Cable where i took out all the cables an rewired 5-5(mass) 13-6(can2) 12-14(can2) and 16-16(12v). got everything up and running, but it in ddt4all it says all the time "no data".
is "Kangoo II" the right model? under ecu navigation ig got may different MFD up to v5.2
i want to unlock android auto

thanks for help :)

KarelSvo, 2019-03-09:
Show a screenshot of the scan result.

Fischmaster3000, 2019-03-14:
sorry had no time to get to the car. i really appreciate your help :)
when i scan it looks like this:
[image]

Fischmaster3000, 2019-03-22:
you already got time to look into it?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif