cedricp/ddt4all #166: need any version with english descriptions

Tahreer1972, 2019-03-18:
Hi 
its amazing program  thanks alot , i dont know french , is there any version with english descriptions for the different parameters. thanks

Tahreer1972, 2019-03-20:
Anyone help a bother who struggles with French ?thanks

shrlnm, 2019-03-20:
The language of parameter descriptions depends on the database content.

The pyren may translate those parameters in ddt database which are in clip database, but usually  clip contains only about 30% of parameters from ddt database.

Tahreer1972, 2019-03-21:
> The language of parameter descriptions depends on the database content.
> 
> The pyren may translate those parameters in ddt database which are in clip database, but usually clip contains only about 30% of parameters from ddt database.

thank you  very helpful . download it and it works perfect..

closed 2019-03-26 by Tahreer1972