cedricp/ddt4all #168: Car Type Table

Sig2018, 2019-03-19:
HI,

Do you know the code for Koleos 2 in the ECU Vehicle type.
There is a table below but Koleos is not there.
BFB – Megane 
BFB-GT – Megane GT 
HZG – Kadjar
JFA – Scenic
JFC – Espace
KFB – Megane SW
KFB-GT – Megane SW GT
KFD – Talisman SW
LFD – Talisman Sedan
LFF - Megane Sedan
RFA – Grand Scenic

Thank you for your help.

shrlnm, 2019-03-19:
XZG

pcboys, 2019-03-19:
XZG  koleos II
XZJ   Koleos II CN

Sig2018, 2019-03-19:
@pcboys @shrlnm 
I'm using the ECU from Koleos XZG but there is no option XZG on vehicle as you can see on Screen capture below.
[image]
[image]

shrlnm, 2019-03-19:
HZG

Sig2018, 2019-03-19:
Thank you! Any reference for that?

shrlnm, 2019-03-19:
I have this value in my koleos. 
And your list above have a mistake. HZG is not Kadjar
[image]

shrlnm, 2019-03-19:
I may be wrong but first sign means the type of body
H - SUV
x - any type of body. (the koleos has no others)

so for Koleos XZG==HZG

Sig2018, 2019-03-19:
Thanks a Lot. Is it your Koleos a New Koleos? Have you played with RLINK 2?
I did a customized upgrade and  I need to set some values to Camera Guidelines, maybe we can share some info.
And yes there are some forums saying Kadjar is  HFE.

shrlnm, 2019-03-19:
First Koleos was H45
What version of firmware did you have before customized upgrade and what version do you have now?

Sig2018, 2019-03-19:
I moved from 2.2.19 to 3.3.

Sig2018, 2019-03-19:
Additionally I found few ECU for new Koleos.

Sig2018, 2019-03-19:
I'm talking I find just few ECUs compatible with New Koleos. I will scan and send you the list. sermass16-f@yahoo.com

Sig2018, 2019-03-19:
Sorry I found few ECUs in ddt4 database. I have the 2019 database.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif