cedricp/ddt4all #172: Does ddt4all supports KWP2000

Tahreer1972, 2019-03-26:
Hi Elm327 supports  both KWP2000 and CAN bus . My older car clio II x65 2001 has 4 ecus 3 on can bus and one on KWP2000. Would ddt4all see all of them in one scan or need to  select which bus ? Thanks

cedricp, 2019-03-26:
It works with KWP2000, but I tested it a little.

Tahreer1972, 2019-04-04:
Hi  thank you for your response . yes it works  with kwp2000 and actually my car has 4 ecu's on that bus . ddt4all managed to link 3 ecu with xml file but not the 4th one . is there anyway to get it communicate with the UCH ecu . Any help will be much appreciated . thanks 

Identified ECU [ECM]@7A : SIRIUS32___EMS3132___Soft_RE1_maj.json DIAGVERSION [16]SUPPLIER [001] SOFT [00E1] VERSION [A300] {540}
Identified ECU [Airbag-SRS]@2C : AB80_X65_X90_V1.json DIAGVERSION [4]SUPPLIER [037] SOFT [0900] VERSION [0200] {169}
Identified ECU [HVAC]@29 : BAC_X65.json DIAGVERSION [0]SUPPLIER [A50] SOFT [0001] VERSION [0000] {854}
Found ECU [UCH] (no relevant ECU file found) :DIAGVERSION [0] SUPPLIER [213] SOFT [0036] VERSION [0001]

Tahreer1972, 2019-04-05:
hi again how to make ddt connect to unidentified ecu manually?? thanks

closed 2019-04-05 by Tahreer1972