cedricp/ddt4all #173: Torque PID

Tahreer1972, 2019-03-26:
Hi  Is there a posibility to produce torque pid in ddt4all ?thanks

cedricp, 2019-03-26:
You can calculate them from the _request_ page in _edit mode_

Tahreer1972, 2019-04-04:
Hi , iam sorry i could not find the edit mode . and i have another question if you dont mind .
i need the first steps to how create my own ECU parameters screen. which files i need to start with.  i hope you have a short tutorial somewhere . thanks

Tahreer1972, 2019-04-05:
hi  it seems to me that my ddt4all windows version has no request page nor data , buttons  or uiedit

closed 2019-04-05 by Tahreer1972

KarelSvo, 2019-04-30:
https://www.club-renault.ru/forum/topic/5528-diagnostika-svoimi-rukami/page/146/?tab=comments#comment-2145011

Tahreer1972, 2019-04-30:
thanks   done !!!