cedricp/ddt4all #175: Bricked R-link 1 EOL unconfigured device

mnigten, 2019-04-01:
Based on same issue as the closed issue #92 i have a question.
My R-link 1 is bricked. i know it is a software issue. It is or the radio or the MIU. 
Can someone provide me a dump of the following ECU's?

[audio] Radio 2012 v2.4
[audio] Radio 2012 NoDab v2.0
[MIU] BIC/A2/A3 v24

I want to try to set the dump back. Maybe that will work for my bricked system.

Thanks in advance

mnigten, 2019-05-01:
Nobody able to help? Just a dump of these ecu’s.

nazmi02551, 2019-07-01:
Hello there.  I'm having the same problem.  Did you solve the problem?

mnigten, 2019-07-01:
Hi mazmi02551, 

No still have the issue. I have to go to garage in August for annual check. For me Renault Netherlands proposed me a free diagnose for this particular problem. On a side tour I ordered a 2nd hand bic miu to see if the issue is this piece of hardware.

nazmi02551, 2019-07-01:
Sorry. What is  "bic miu" and  how much money?

mnigten, 2019-07-01:
The bic or other name = miu , multimedia interface unit , is interfacing multimedia with car through canbus,  there is a possibility that settings are corrupted after a software reset. Please read full threat above and issue 92. It will come more clear. Replacing the miu is 1 of the cheapest steps . Costs are around €70

nazmi02551, 2019-07-01:
[image]
First of all, thank you very much for the information you provided.  That's the error on my screen.  Is it not possible to solve this error with software or "ddt4aal"?

JPCM3D, 2019-12-09:
I also have this problem and even Renault said that they don´t know how to fix it... 
[image]

Theu said that we have to put the pendrive with the last upodate... doesn´t work because the OS of the radio don´t start... and never get to the point of taking the update file...

should have to be a way of fixing it...

or the last option is to find a used radio or to remove it and put a Android radio....

https://s.click.aliexpress.com/e/pBdUD9vA

any help?

nazmi02551, 2019-12-09:
> I also have this problem and even Renault said that they don´t know how to fix it...
> [image]
>
> Theu said that we have to put the pendrive with the last upodate... doesn´t work because the OS of the [...]

my problem is solved.  renault technical service, said it was a problem that has not encountered before.  They reinstalled the navigation device with can clip.  there was something to be considered here.  After the operation with can clip, the connection cable needs to be disconnected to continue the process.

JPCM3D, 2019-12-09:
Thanks a lot Nazmi

I will try at Renault dealer with this information... 

I think they don´t know how to do it... but let´s try

Best regards
JP

mnigten, 2019-12-10:
In my case they changed the BIC. because the changed the BIC they needed to configure it with CLIP. My conclusion is that maybe it was not needed to change the BIC but they where not able to tell me because the changed it without notice. The reason was because Renault was convinced it was related to a software issue after last update and we made a deal to charge me 25% of the costs.

mnigten, 2019-12-10:
DDT4ALL is not (Yet?) able to reconfigure the BIC and navigation. Or i just do not know how.

seb0031, 2020-07-05:
Bonjour, 

je suis dans la même galère, avec le message d'erreur "eol settings". pouvez-vous m'aider ? j'ai une zoe de 2013. j'ai voulu mettre a jour android auto avec ddt4all. depuis je peu plus rien faire. help me help me

cordialement

seb0031, 2020-07-05:
> [image]
> Tout d'abord, merci beaucoup pour les informations que vous nous avez fournies. Voilà l'erreur sur mon écran. N'est-il pas possible de résoudre cette erreur avec un logiciel ou "ddt4aal"?

Bonjour, as tu trouver une solution ?

seb0031, 2020-07-05:
> J'ai aussi ce problème et même Renault a dit qu'ils ne savent pas comment le résoudre ...
> [image]
> 
> Theu a dit que nous devons mettre la clé USB avec la dernière mise à jour ... ne fonctionne pas car le système d'exploitation de la radio ne démarre pas ... et n'arrive jamais au point de prendre le fichier de mise à jour ...
> 
> devrait être un moyen de le réparer ...
> 
> ou la dernière option est de trouver une radio utilisée ou de la supprimer et de mettre une radio Android ....
> 
> https://s.click.aliexpress.com/e/pBdUD9vA
> 
> de l'aide?

Bonjour, as tu trouvé la solution ?

seb0031, 2020-07-05:
> J'ai aussi ce problème et même Renault a dit qu'ils ne savent pas comment le résoudre ...
> [image]
>
> Theu a dit que nous devons mettre la clé USB avec la dernière mise à jour ... ne fonctionne pas car [...]

bonjour, as tu trouvé la solution ?

JPCM3D, 2020-07-05:
Ola

Nunca arranjei solução, o carro nao era meu.

O dono do carro teve de retirar de mandar flashar o radio e meter o firmware novamente

Mas nao assisti e nao vi o processo

Cumprimentos

Enviado do meu iPhone

seb0031, 2020-07-06:
Il a fait réparer chez Renault ?

JPCM3D, 2020-07-06:
Non

Enviado do meu iPhone

gaouar, 2020-08-07:
bonjour,
je suis sur le même; configuration EOL
vous avez trouvé la solution?

je vous remercie

seb0031, 2020-08-10:
je l'ai amené dans un garage renault qui a remis a zero le gps

chuppito, 2020-10-29:
Hello, all you have to do is fill in the ecu data correctly in navigation then mfd (12, 13 and 14) with a write all at each ecu for the station to restart

Franel44, 2021-01-13:
Bonjour,

Je suis dans la même situation, ddt4all pour Android auto et après redémarrage, triangle jaune.

J'ai essayé de réécrire les ecu dans navigation (mfd 12, 13, 14) enlevé le fusible... Rien n'y fait 🥴.

Avez-vous une solution ?

chuppito, 2021-01-13:
Bonjour, regarder les autres mfd et trouver lequel est a réécrire

Franel44, 2021-01-13:
Bonjour Chuppito et merci pour ta réponse rapide,

Je viens de vérifier avec les 4 mfd, celui nommé v1.8 est parfois accessible et parfois plante de ddt4all.
Je vais dans reset ecu configuration puis je vérifie dans Configuration les ecu 12,13 et 14. Du coup, ce qui est en rose est inscrit en vert côté droit, est-ce bien le signe que la réécriture est faite ?

Le R-link est toujour bloqué malgré ça.
[image]

Franel44, 2021-01-23:
Chuppito, as-tu une nouvelle aide pour débloquer le R-link ? 
Je n'y arrive vraiment pas ...

chuppito, 2021-01-23:
Non il faut remettre tous les ecu en place.

Franel44, 2021-01-23:
Comment savoir quand ils sont en place ?

AndyIreland, 2021-02-27:
> Hello, all you have to do is fill in the ecu data correctly in navigation then mfd (12, 13 and 14) with a write all at each ecu for the station to restart

hello @chuppito - happened to me today i pressed the dreaded 'reset configuration' icon and then only went searching for a solution when i got the dreaded yellow triangle EOL warning. 

i set up all the parameters now (but in radio 2012) and now I have a beep out of each speaker in rotation. do you think I am any nearer to fixing my rlink with ddt4all at home ? 

many thanks.
Andy

Franel44, 2021-03-07:
Bonjour,

Je suis passé au garage Renault et ils ont réinitialisé le R-Link, tout fonctionne parfaitement ! j'ai réussi aussi à débloquer Android auto.
Renault à pris en charge le coût de la réparation !

vraug666, 2022-06-21:
Hello, I have a r link after the reset does not want to run, black screen. Via the modified cable both DDT4all and Reno link cannot find computers from navigation or radio. Can somebody help me?

vraug666, 2022-06-21:
>Not everything is dead. I used to activate android car via modified cable. After that I found the upgrade, installed and the r link no longer wants to run. The car is a megane3 2015.
 Installed new radio, display, BIC, UCH and again does not respond. I don't know why I can't find computers from navi, radio and BIC.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

BarektE, 2023-10-29:
It's very simple to do

https://youtu.be/tKro_OGZtRM

https://www.youtube.com/@AtuMOtOElektroniK/search?query=r-link

matzel82, 2023-10-29:
You need a original activated clip.... of course. Only a Renault Garage have this one and a granted access to Renault config files with a token

BarektE, 2023-10-29:
> Orijinal, aktif bir klibe ihtiyacınız var... elbette. Yalnızca bir Renault Garajı buna sahiptir ve bir jetonla Renault yapılandırma dosyalarına erişim iznine sahiptir

It can also be done with DDT4AL and PyRen

BarektE, 2023-10-29:
Daha dün aynı sorunu hallettim. Yetkili servis tarafından robroc yapıldıysa 1 harita yok olur
2 dış sıcaklık yok olur
3 geri görüş kamerası yok olur
4 park yardım asistanı yok olur 
Servisin güncellemesinden sonra internetten başka bir yazılım bulun yükleyin düzelecektir. Aslında servis yükleme yaparken bu cihazların alt yapısının olduğu USBS takmamış onu takmadığınızda hatalar oluşuyor.

Proton9999, 2024-02-08:
Good evening. I downgraded the rlink firmware and then installed the newest one. The menu button is now inactive on the screen. Resets when pressed. Who can help?

bigmac5753, 2024-07-01:
how do I get the CRC?