cedricp/ddt4all #176: Changing cluster K/mh Speedo to mph

AndyIreland, 2019-04-01:
my car: Renault Clio Mk4 1149cc 16v 2016 medianav version
Hello with my cheap ELM327 (wi-fi) and your DDT4ALL I have managed to get outside temp and clock showing in the info panel where it wasnt showing before. But I am trying to see if I can change the trip meter to miles instead of kms and the odometer to miles instead of km and the large speedo digit to MPH from KM/h - I should be able to do that with your DDT4ALL programme shouldnt I ? 

The cluster I have chose is : Cluster X98_X87_X82_xHA_X98ph2_X87ph2_xHAph2_v7.8 Cluster/TDB 

but I get message of: Bad ELM response : WRONG RESPONSE : NR: SubFunction Not Supported(7F2212) 
does this mean the function cannot be changed or I just need to get a better ELM327? 

its the same with the '0 seatbelt disconnected' - I want to get rid of that message on the dashboard, again I should be able to do that with DDT4ALL shouldnt I please? - but when i write it I get same message and on restarting the car I still get the rear seatbelt message and the odometer is still in km - it seems its not writing it into the ECU
Many thanks for any help - I think this will be a great program if I could get it to work how I want it. 
here is pic of my elm327 - I can buy a iLC OBD2 OBD Wireless Car Code Reader Scan Tool  from AMAZON if I need to .  - I am using DDT4ALL version 5.4.0 all cars edition.

[image]

cedricp, 2019-04-11:
If you get such a response, it means your ELM is working. You must check that you use the correct XML file. Did you scan your ECUs with auto identification (select vehicle type then click the magnifier icon) ?

cedricp, 2019-04-11:
After analyzing your logs, you get an error for all request, and that's really weird... What about reading parameters from other calculators ?

AndyIreland, 2019-04-11:
many thanks for your reply cedric. - since i wrote the post I have managed to change from Km/h to mph on the dash using PyREN184 , and ticked the boxes about 'cheap china ELM' parts and it changes. How would I be emulate the cheap Chinese ELM settings in DDT4ALL program please? 

here is what I have enabled in PyREN184:

shrlnm, 2019-04-12:
I suppose that the problem was  not in these options. You  need to reload the dash every time you change something in it. The pyren do it automatically and ddt4all do not.

```
ECU : 11493  FG0311121.XML
Screen : ECM -> C140 [Command] 

#############################  Command parameters  #############################

Prerequisite : Ignition on and engine stopped. For vehicles fitted with 
automatic transmissions, position the gear lever in P or N. For electric 
vehicles, unplug the vehicle charging cable. When the command 
is finished, return to the FAULTS screen before switching off the ignition.

name         : C140
codeMR       : CF140
label        : DISTANCE UNITS
type         : CF
scenario     : 
inputlist    +
              : (00   ) km
              : (01   ) miles
serviceID    +
              : (59   ) 2E0104 <Params>
              : (60   ) 1103
              : (61   ) 10C0

##########################  Current values  ##########################

LC051  DISTANCE UNITS                                                    km   

######################################################################

Are you ready to execute the command? <yes/no>:

```
The command 1103 reloads the dash

AndyIreland, 2019-04-12:
> I suppose that the problem was not in these options. You need to reload the dash every time you change something in it. The pyren do it automatically and ddt4all do not.
>
> ```
> ECU : 11493 [...]

ah right , thank you - yes that makes sense now

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

stocky777, 2024-08-03:
Hello, I don't intend much about ddt4all, and I was wondering if you guys could help. I am trying to set the dashboard to mph as it was originally but I am having trouble as it goes back to kmh. Like instantly I am in cluster_15-40_3.30.02/comf prestations VEH from there I change it but it goes back to that, and I have trouble understanding the speed because of it. Can I have a help please

AndyIreland, 2024-08-04:
> Hello, I don't intend much about ddt4all, and I was wondering if you guys could help. I am trying to set the dashboard to mph as it was originally but I am having trouble as it goes back to kmh. [...]

What Renault do you have? - you might not even have to touch DDT4all or Pyren to change you might be able to do it with the end buttons on the stalk and pressing the start buttons like here as this grand scenic car for an example:
https://youtu.be/Qd1pEwr4RkI?si=YR1eqBlDrSBcSFYy

stocky777, 2024-08-04:
Hello, I have a Renault kadjar 2017 eco2, I was hoping there was a solution
that didn’t require any of those software, the problem is the dashboard is
digital and I’m scared that I may need the multimedia screen in order to
change it, but I accidentally put it on virgin mode

matzel82, 2024-08-04:
I can originally change with clip the units… so you can change the same mit ddt. Take a look to „configuration“

AndyIreland, 2024-08-04:
> Renault kadjar 2017

Try and see if this works then for models without navigation - it might:
[image]

[image]

stocky777, 2024-08-04:
Oh my god, I love you so much, it worked. YESSSSS