cedricp/ddt4all #179: how to make ddt connect to unidentified ecu manually?? thanks

Tahreer1972, 2019-04-05:
Hi  My older car clio II x65 2000 has 4 ecus  on KWP2000 . ddt4all managed to link 3 ecu with xml file but not the 4th one . is there anyway to get it communicate with the UCH ecu . Any help will be much appreciated . thanks

Identified ECU [ECM]@7a : SIRIUS32___EMS3132___Soft_RE1_maj.json DIAGVERSION [16]SUPPLIER [001] SOFT [00E1] VERSION [A300] {540}
Identified ECU [Airbag-SRS]@2c : AB80_X65_X90_V1.json DIAGVERSION [4]SUPPLIER [037] SOFT [0900] VERSION [0200] {169}
Identified ECU [HVAC]@29 : BAC_X65.json DIAGVERSION [0]SUPPLIER [A50] SOFT [0001] VERSION [0000] {854}
Found ECU [UCH] (no relevant ECU file found) :DIAGVERSION [0] SUPPLIER [213] SOFT [0036] VERSION [0001].

 how to make ddt connect to unidentified ecu manually?? thanks

cedricp, 2019-04-08:
Unfortunately, if there's no definition file for that ECU, there's nothing more you can do... I looked into the database, and found nothing that matches (even at leat partially) these identifiers.
Maybe you need an older database.

Tahreer1972, 2019-04-08:
Hi    thank you for your response . I am happy with older database . But
where can i find one? thanks

cedricp, 2019-04-08:
Don't know... You can try PyRen (Clip clone) software too, maybe there's BCM definition file.
https://gitlab.com/py_ren/pyren

Tahreer1972, 2019-04-08:
hi again 
yes  i am in contact with Mr shrlnm and he is helping me with Pyren , but its the same issue as Clip doesnt have older database in the sessions format but rather ddl files. my last question  is it possible through part numbers to my ecu's to find a closer ecu where i can make minor modifications as alternative. thanks  
[IDENTIFICATION_2019-04-06_21-32-02.zip](https://github.com/cedricp/ddt4all/files/3053579/IDENTIFICATION_2019-04-06_21-32-02.zip)

Tahreer1972, 2019-04-12:
@cedricp  hi 
i have tried 2004 / 2008 /2017/2018 database  same results . i am 95% sure my UCH ecu xml is BMT_X65_ph1_p.xml  even though this one by sagem 1999 and mine is sirius 2000 . i think the fact mine was made after thats why BMT_X65_ph1_p.xml has no <AutoIdents> with my specs .. i am thinking to add 
<AutoIdents>
<AutoIdent DiagVersion="0" Supplier="213" Soft="0036" Version="0001"/>
</AutoIdents

into both BMT_X65_ph1_p.xml  and eculist.xml ..  is this the way ddt4all auto detect the ecus ? or do i need to make other modifications . thanks

Tahreer1972, 2019-04-12:
sorry to add  "<AutoIdent DiagVersion="0" Supplier="213" Soft="0036" Version="0001"/>"  and hope yr ddt4all identifiy it

Tahreer1972, 2019-04-12:
sorry to add

Tahreer1972, 2019-04-12:
sorry  to add <     AutoIdent DiagVersion="0" Supplier="213" Soft="0036" Version="0001"     /    >

closed 2019-04-17 by Tahreer1972