cedricp/ddt4all #180: fonts problem 2

Andrengg, 2019-04-09:
this default font Android
[image]

pakoto, 2019-04-09:
Same here

[image]
[image]
[image]

cedricp, 2019-04-09:
Did you try with V1.4 ? I fixed some problems like this one recently

cedricp, 2019-04-09:
Except the font issue, what are your thoughts about the application ?

pakoto, 2019-04-09:
App works ok! Thanks again.
Full uninstall / install 1.4 again. dokker - edc17c42 - fap
big zoom

[image]

Andrengg, 2019-04-10:
this v1.4 ...

cedricp, 2019-04-10:
I added a font scale override in screen settings (gear icon) in **V1.5**. You can scale up/down.

pakoto, 2019-04-10:
Works ok. Thanks!

pakoto, 2019-04-10:
[image]

Andrengg, 2019-04-11:
v.1.5
problems. ..
https://ibb.co/d50CC2P
https://ibb.co/cCkPJ9H

igogo2, 2019-04-11:
[My](http://mytab.com.ua/ontario_2.html) device.
System - Windows 10 version 1809
My fonts.
[image]

cedricp, 2019-04-11:
It's quite difficult to manage fonts issues for all platforms, and as long as there's so few donation, I can't afford hardware to test it...

cedricp, 2019-04-11:
@igogo2 : I just commited a fix for font size (remove DPI denpendency)

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif