cedricp/ddt4all #181: Erratic DTC reading

Sig2018, 2019-04-10:
Guys,
As life sometimes is too easy I complicated my self. 
After an upgrade from 2.2 to 3.3 and the activation of AAuto relatively successful I read the DTC with DDT4 and to my surprise it shows errors. 
There are always two, but when you delete those two new ones appear and they are different.
Here I leave a partial screen capture but some of these are:
Internal system error
Component faults
Bus signal / message faults
General electrical faults
System programming errors
Error-based algorithm
Etc
Has it ever happened to you?
It is a New Koleos With RLINK2 upgraded from 2.2 to 3.3
[image]

cedricp, 2019-04-11:
Some DTC are showed even when there's no real fault. As an example, my Sagem S3000 ECM calculator shows an alternator fault when engine is not running (obvious), and it disappears when I start it.

Sig2018, 2019-04-11:
Thank you Cedricp,
It is very weird because after deleting the errors another different show up.
I've tried with Pyren and there are not errors for that ECU (audio).
Is it possible a fake ELM is causing this? ELM is working on ECU reading and Programing.

cedricp, 2019-04-11:
Are you up to date with source code ? I recently fixed a bug with DTC reading (some extra (undesired) bytes were taken into account)

shrlnm, 2019-04-11:
> If I remember correctly, PyRen removes some DTC that are not relevant.

Yes, you are right. Clip database contains filters for irrelevant DTC. Highly likely the dtc mentioned above are not relevant and so not shown by pyren

Sig2018, 2019-04-11:
I have version 5.4.0 of ddt. Is there any new version?

shrlnm, 2019-04-11:
For example the filter for DTC 37713 (0x9351)

```
(((_STATUSOFDTC#0==0x01)&&(_DTCFAILURETYPE==0x4B))||
 ((_STATUSOFDTC#0==0x00)&&((_STATUSOFDTC#3==0x01)||
  (_STATUSOFDTC#5==0x01))&&(_DTCFAILURETYPE==0x4B)))?1:0

```

cedricp, 2019-04-11:
@shrlnm Thanks for the confirmation :)
5.4.0 is not an "official" version, probably packaged by someone else. But you can update the program by replacing the source code with the latest code : https://github.com/cedricp/ddt4all/archive/master.zip
Just overwrite the files in the installation directory.

Sig2018, 2019-04-11:
Ok thank you. I will download, try and let you know.

Sig2018, 2019-04-13:
@cedricp  unfortunately there was no luck.
The erratic errors are still there.
I saw the command after sent the clear dtc and it says DTC clear OK with cote "54"

Is DTC reading 7F1012? because it is next to the DTC clear and there is a NR subfunction not supported.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif