cedricp/ddt4all #182: Bluetooth Disconect on ECU Tweak android version

Sig2018, 2019-04-11:
Hi,
I've just tried the ddt4 android version (Without unlock yet) and every time I select the ECU and the program start to show the ECU Screen the Bluetooth get disconnected. Then I have to go to refresh button (Rounded arrows) and after a few seconds it connects again.
Is it normal?
Also is there a way to go from one Screen to other in the same ECU without going back and had to select again the ECU?
I'm waiting unlock code to try more.

cedricp, 2019-04-11:
Yes, I didn't find a way to keep connection between activities, so it disconnects and reconnects.Maybe I'll find a way to avoid that in a near future.

Sig2018, 2019-04-11:
Ok , No problem. Just wanted to know if it was not a problem with my OBD.

pakoto, 2019-04-11:
Also disconnect bt from car audio... ;)

Sig2018, 2019-04-11:
Thats funny because in my case it works even the Bluetooth is connected to audio. It looks like ECU tweak just use the Bluetooth a short time and then release it.

senoller, 2019-10-09:
no

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif