cedricp/ddt4all #184: what is the minimum i need in ecu.zip to run ddt for one car

Tahreer1972, 2019-04-12:
@cedricp   hi 
i have  made  ecu.zip file just with my 3 ecus file/ json and json.layout  but the ddt4all  flashes and closes . how to make it to work just for one car / model   please see  this zip file . thanks 
[ecu.zip](https://github.com/cedricp/ddt4all/files/3075380/ecu.zip)

KarelSvo, 2019-04-28:
The ecu list db.json was missing.
[ecu.zip](https://github.com/cedricp/ddt4all/files/3125463/ecu.zip)

Tahreer1972, 2019-04-29:
thank you for your response  but i guess i have also to edit db.json and reduce it just to my ecus list?

Tahreer1972, 2019-04-29:
thanks  much appreciated

Tahreer1972, 2019-04-30:
hi @KarelSvo   sorry   had to reopen this issue . your  ecu.zip file worked perfect  but when i needed to add the 4th and last ecu  and i manually edited db.json   . it did not work for me  i must be missing something in here . could please have a look at 
[Uploading ecu.zip…]()

Tahreer1972, 2019-04-30:
[Uploading ecu.zip…]()

Tahreer1972, 2019-04-30:
[ecu.zip](https://github.com/cedricp/ddt4all/files/3129677/ecu.zip)

KarelSvo, 2019-04-30:
I edit the ecu files in XML format with the DDT2000 BUILDER program.
Then with the function File / ZIP database create a new ecu.zip, automatically including db.json.

Tahreer1972, 2019-04-30:
@KarelSvo  hi thanks for you response . i used ddt2000 builder to create local database with eculist.xml . with ddt4  converted all to json format  but failed to create the db.json file   as  file/zip database  did not create anything  even the  it tells me job finished .  i upload both  the xml files and the json ones
[mycar_xml.zip](https://github.com/cedricp/ddt4all/files/3129954/mycar_xml.zip)
[ecu.zip](https://github.com/cedricp/ddt4all/files/3129955/ecu.zip)

Tahreer1972, 2019-04-30:
hi again @KarelSvo    all done  now   thank  you .. was my mistake   missing ","  in db.json
thanks again

closed 2019-04-30 by Tahreer1972