cedricp/ddt4all #187: Activate Android Auto/Apple Carplay on New Koleos

ratheeshalon, 2019-04-18:
Hi,
Does anyone activated Android Auto/Apple carplay in new Koleos?!!
Which is the suitable ecu for this?

[image]

Any hints?
Thanks!

franeti, 2019-05-01:
What rlink software version you have?

ratheeshalon, 2019-05-02:
Hi, Rlink2- 3.3.16.966
I tried with different Adapters all went wrong. I couldn’t scan koleos with those cheap Chinese clone obd adapters.
So i am waiting for OBDlink SX to delivery. 

Koleos 2017
Australian 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

franeti, 2019-05-05:
Carplay is compatible Koleos 2018. Maybe you havent one component for Carplay like usb ports. Try anyway. Android auto no problem with that software. 

I think ecu radnav 2 is yours. 
Radnav 3 for V7 software for sure.

ratheeshalon, 2019-05-06:
I read in some other forum like GPS-Rlink, If we replace that usb module; we could use Apple Car play(Original USB Module only support Android Auto).

Can you suggest proper supportive USB Module and its part number? Where i can buy it? 

To accessing Radnav ,is that really required to modify ODB pins for Koleos?

Any Hints
Thank you!

shrlnm, 2019-05-06:
> If we replace that usb module; we could use Apple Car play

It not help. Car Play is supported on R-Link2 GEN2.1 and higher. You may activate only AA

> To accessing Radnav ,is that really required to modify ODB pins for Koleos?

It doesn't required. R-Link2 is on the primary CAN bus

ratheeshalon, 2019-05-06:
> > It not help. Car Play is supported on R-Link2 GEN2.1 and higher. You may activate only AA
>
> Some Australian owners posted on that forum as they  got Apple car play after replacing USB Module. However, I'm not sure which Rlink Generation is with them.

(how can i check my current Version of Rlink GEN?)

Thank you!

shrlnm, 2019-05-06:
> (how can i check my current Version of Rlink GEN?)
I know only the way thru DeveloperSide

[image]

ratheeshalon, 2019-05-06:
> I know only the way thru DeveloperSide

Is that safe to enter DeveloperSide? Exit from that could be trouble ?
(I read Entering DeveoperSide is extremely dangerous)
Thanks

shrlnm, 2019-05-06:
It is dangerous only pressing "High" "Mid" and "Virgin mode" buttons. But if you have working ELM then you may completely repair device after pressing any of this button.

ratheeshalon, 2019-05-06:
> It is dangerous only pressing "High" "Mid" and "Virgin mode" buttons. But if you have working ELM then you may completely repair device after pressing any of this button.

After some search in internet,I found  some people actually accessed Car play through DeveloperSide by went to DiagRw Button

Thanks!

ohayyam, 2019-05-07:
Can Virgin mode erase written vin number on radio ecu ?

shrlnm, 2019-05-07:
Yes, virgin mode erasing the vin. After erasing you may write a new one

shrlnm, 2019-05-07:
Virgin mode alse erase all configs, so you Will be need to do autoconfiguration

ratheeshalon, 2019-05-07:
@shrlnm If i enter DeveloperSide and never updated/explored anything; So how to return from that? Just "BACK" Button or need to reset Rlink2 by pressing power button?

shrlnm, 2019-05-07:
@ratheeshalon just "BACK" button for exit from DeveloperSide. No need to reboot the devise if you did not change anything thru DiagRW

ratheeshalon, 2019-05-07:
@shrlnm ,Thanks mate! ,**DiagRw** is located in between **Mid** and **VirginMode** ,So there is any confirmation warning before entering those functions? if i accidentally pressed it. (Haven't tried Developer mode,Sorry for being silly)
Thank you!

shrlnm, 2019-05-07:
@ratheeshalon The virginmode require confirmation for sure. The Mid button I do not know.
Any way, you may use ordinary usb mouse connected to rlink2 for precise clicking

ratheeshalon, 2019-05-07:
@shrlnm Definitely try soon , will let u know, Thanks you!

franeti, 2019-05-09:
If you have gen 2.1 replace the usb port to get carplay. And is better with ddt to activate it, with debug mode it fail something about microphone. 

Renault can sell you the piece, or locate a crashed car in scrap?. I dont know the name in english where the olds car go. sorry. 
280239665R Look this in google. 

It works I did it after renault change my rlink.

kadjarred, 2019-05-27:
Hi guys
Having deep fun with the RLink2 system and trying to activate CarPlay. 
Used a dd4... to active AA & CP, got AA going but have IOS device so no good. 

Swapped out the usb to a 2017 port which hoped would do CarPlay but no. 
Is the issue with main LG unit?

midoucirine, 2019-05-29:
salut les gars
j'ai un grand problème avec mon r-link2 de ma MEGANE 4, je suis entré dans le menu "DeveloperSide" et j'ai Accidentellement appuyé sur "Mid" .ecran noir. comment fair. please......;

shrlnm, 2019-05-31:
> salut les gars
> j'ai un grand problème avec mon r-link2 de ma MEGANE 4, je suis entré dans le menu "DeveloperSide" et j'ai Accidentellement appuyé sur "Mid" .ecran noir. comment fair. please......;

https://gitlab.com/py_ren/pyren/wikis/how%20to%20rlink2%20black%20screen

franchicob, 2019-06-18:
Hi ratheeshalon

I also have a Koleos and I'm also located in Australia and thinking on getting the car updated. Did you have any luck with what you've done?

thanks mate

midoucirine, 2019-06-18:
salut pote . oui j'ai eu beaucoup de chance et j'ai réactiver mon r-link2
aprés avoir toucher (mid) en mode devlopeur.si  t'as le méme problème
contacte moi 👍

franchicob, 2019-06-27:
> If you have gen 2.1 replace the usb port to get carplay. And is better with ddt to activate it, with debug mode it fail something about microphone.
> 
> Renault can sell you the piece (170€), or locate a crashed car in scrap?. I dont know the name in english where the olds car go. sorry.
> 280239665R Look this in google.
> 
> It works I did it after renault change my rlink.

Hi franeti, I checked my car last night via developer mode and it is gen 2.0. If I replace the USB module will carplay work?
Thank you in advance mate.

franchicob, 2019-08-05:
Ok so I managed to get both working. Android Auto and Apple Carplay.
Very unfortunately only on v8 Carplay takes advantage of full screen. Does anybody know a way of upgrading from v3 to v8?

pinto1230766, 2019-08-22:
salut
message pour midoucirine.
j'ai le même problème quf vous, comment je fais pour résoudre.
merci dème contacté.
pinto123@free.fr

franchicob, 2019-08-23:
> salut
> message pour midoucirine.
> j'ai le même problème quf vous, comment je fais pour résoudre.
> merci dème contacté.
> [pinto123@free.fr](mailto:pinto123@free.fr)

English please

xmenmaxx, 2020-08-25:
Hi,
Can anyone help me with this error? R-Link software release 3.3. 16.961, boot release 4992.
[image]

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

Proton9999, 2024-01-31:
I have a 2018 talisman. I updated the firmware and activated everything, but after a reboot the Android car is not saved to “on”. Has anyone encountered this? Thank you in advance.
[image]
[image]
[image]
[image]

matzel82, 2024-01-31:
> I have a 2018 talisman. I updated the firmware and activated everything, but after a reboot the Android car is not saved to “on”. Has anyone encountered this? Thank you in advance.
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image]
> 
> 

3.3.16.xxx is to old.... the latest u it s 7.x.x.xxx

At a Garage you get the official latest SW.

shrlnm, 2024-02-01:
> Thanks / ср, 31 янв. 2024 г. в 23:24, matzel82 ***@***.***>:
3.3.16 is supporting AA/CP. Try to activate it thru pyren macro or any other way thru the CAN bus. Do not use DeveloperSide for it.

**!!!!! BE CAREFUL. You HAVE to upgrade MICOM/BIOS in official garage BEFORE upgrading LGU from 3.x.x. to 7.x.x !!!
If you do not do it and just install the flash stick with LGU 7.x 8.x. 9.x the you will broke the R-Link2** 

**!!!!!  Обновлять 3.3.16 на 7.x.x можно ТОЛЬКО после замены  MICOM/BIOS у официалов !!!!!
Если просто вставите флешку с 7 прошивкой, то окирпичите свой R-Link**

Proton9999, 2024-02-01:
Good evening. We do not have an official dealer. If there is a bios file,
please send it. I can do it myself. Thank you in advance

shrlnm, 2024-02-02:
@Proton9999 Unfortunately I don't have this file. 
But it is really interesting for me how you are going to reflash the MICOM. There are two ways. 
1) Directly to the flash chip by the reprogrammer. Then you will have to have a damp file 
2) Unlock the security mode and reflash it thru the CAN. but in this case you have to know the algorithm  of calculating the key from the seed.

Proton9999, 2024-02-08:
Good evening. I downgraded the rlink firmware and then installed the newest one. The menu button is now inactive on the screen. Resets when pressed. Who can help?