cedricp/ddt4all #189: Koleos key lock and horn activation (help needed)

Dazzle1983, 2019-04-23:
Issue #1
I own koleos 2018 with remote key engine start, however whenever I open the door the engine shuts off and I need to start again.
Issue #2
I can't lock the car from outside while the engine is running any idea how to force lock with the remote? 

Issue #3
Panic button only flashes after activation without honk, is there a way to make the honk active or siren to be active?

gordon6, 2019-05-10:
Hello,
This is what you need...
#1. If you have Diesel, when ISG activated, engine stopped and door opened, the engine would shuts off due to safety issue.
#2. No, you can't.  Due to safety issue.
#3. UCH - CMF1_BCM_DDT_11 - AlarmPerimatic - PANIC_BUTTON_CF - True is the answer.

Dazzle1983, 2019-05-10:
1#i got a petrol version, is there a way to override this safety issue?
2#same as above? Anyway way to override it? 
3#i did it previously, although it works but without siren or sound.... Just flashes.... 
I saw a video of a talisman with siren on

gordon6, 2019-05-10:
#1. I think your car had a problem, have to contact to dealer or manufacturer...
#2. If you really really want to.  You have to use real KEY inside of Smart key, open the cover of Smart key.
#3. My 2017 Koleos works good with Flash and Horn sound...

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif