cedricp/ddt4all #192: Weird Behavior on ddt4 after wrinting an ECU and try to read.

Sig2018, 2019-04-25:
Hi Cedric all,

When writing on USM_CMF1_Sailing_B4.2_20150401T141744 after I writhe value DRLInhibitByFlasher next refresh shows a lot of "No Data" in several fields.

Looking at log it looks like the ECU may be a response time issue but since I don't know the program I'm not sure.

For background I'm using an USB ELM327 and data is wiritten correctly on ECU.
Doing refresh of values bring everithing again but I'm afraid of keep using becasue next writing something could be wrong.

Looking at the log there are some messages like this:
20:08:42.102;74D;2E 00 54 80; 
20:08:42.414;74D;220431; 
20:08:42.633;74D;220437; 
20:08:42.711;74D;220756;7F2E23;NR: Routine Not Complete
20:08:42.773;74D;220440;7F2221;NR: Busy Repeat Request
20:08:42.836;74D;220445;7F2221;NR: Busy Repeat Request

Any idea?
Thank you.

shrlnm, 2019-04-27:
Could you please show these commands in elm* log?
You may try to solve this issue by extending timeout but probably there is more clever way

Sig2018, 2019-04-27:
Of course.
Please see files attached. Since I'm not sure the time sync between two files matches I've uploaded the complete session of that day in order you can find it.
It looks like that ECU is very slow, using ddt in Pyrem some times reading leaves several "No data" fields" on screen.
[ECU.txt](https://github.com/cedricp/ddt4all/files/3124052/ECU.txt)
[ELM.txt](https://github.com/cedricp/ddt4all/files/3124053/ELM.txt)

shrlnm, 2019-04-27:
try to use software flow control

Sig2018, 2019-04-27:
If you say SW control on port config. I already tried with no luck. If there any config in ddt4 I didn't.

shrlnm, 2019-04-28:
Sorry, I mentioned --cfc option. I also can not find how to enable it thru UI. You may enable it by editing options.py file. Change the value opt_cfc0  in line 14
from
`opt_cfc0 = False`
to
`opt_cfc0 = True`

cedricp, 2019-04-29:
Last version has flow control button in the interface.

Sig2018, 2019-04-30:
Thank you @cedricp , I found it.
I can confirm nothing changes with Software control.
My finding is ddt4 for reading without SW control read almost all values but had a problem when writing and refreshing. 
Pyren has more troubles reading values (more fields with "NO data").

KarelSvo, 2019-05-03:
I would recommend a obdlink sx interface to exclude hardware and firmware problems ;-)

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif