cedricp/ddt4all #193: No tuning guidelines rear camara in a new rlink 2 unit v7

franeti, 2019-05-01:
Hi!

Renault changed me the old rlink2 with 2.2 software for a new unit with software 7. 

Now I havent guidelines in the rear camara. The camera is working fine but without the lines. 
They are investigating. They have activated it with canclip but the problem persists. 

With ddt I have desactivated and activated rear camera and guidelines. Reboot... Nothing.
In I/O control, rear camera, I tried to set the guideline, I used display button, the rlink show me the rear camera and ussing the 6 yellow button to set the guideline, nothing was showed in the rlink display. 

Any ideas? I have updated to the latest rlink2 version. 
Thanks in advance. Regards. 

[image]

franeti, 2019-05-17:
Nobody? 😞

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif