cedricp/ddt4all #194: nbx_swiss_cf and nbx_scandinav_running_lights_CF is in emm lighting?

AndyIreland, 2019-05-05:
any idea nbx_swiss_cf and nbx_scandinav_running_lights_CF is in emm lighting settings please?

[image]
[image]

AndyIreland, 2019-07-10:
I have since found out that if you disable the DRL lights CF (0 instead of 1) and enable nbx_scandinav_running_lights_CF then the sidelights as well as the dipped headlights light up when engine is running...

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

marcogeraci991-spec, 2026-10-01:
Hi Andy, I want to keep main lights on when engine is running, so I tried setting up nbx_scandinav_running_lights_CF to 1 but nothing changed. You said that you have to set DRL lights CF to 0, did you mean nbx_DRL_ACTIVE_CF? Thank you