cedricp/ddt4all #198: Change ECO Button Function

aqngelrock, 2019-05-11:
Hi for all,

Is it possible to change the function of the echo button on Megane IV? I would like to activate the SPORT function instead like on the GT version.
Many thanks for all.

m0xz0r, 2019-05-14:
This is something I would really like to know too. For starters, does anyone know to which computer the button is connected?

tomidix, 2020-02-08:
You can change it by changing HFE_BCM_DDT_8_0 -> Mmi -> Config -> Sport_mode_cf to true.
Then you open the Radnav 2.X v2.5 -> Radnav configuration 1.1 -> CF_Car_type from standard to GT or RS.
Then click On/Off Reset or Hard Reset (I do it with Hard reset always) and wait for Rlink to boot. After booting your "Eco" button should now trigger Sport Mode. 

**The problem with this modification is that you will lose your Eco mode and will have only Sport, Comfort, Normal and Perso.** However you can modify Perso to simulate Eco if you want. If you revert your changes, eco mode will come back. 
**Be wise and backup your settings before modifications.**

m0xz0r, 2020-03-23:
Thanks for the tip! Do you happen to know if these parameters also apply to a 2017 Clio 4 TCe 120? The Clio 4 GT (with exactly the same engine) has an RS Drive button, no ECO.

tomidix, 2020-03-23:
Don't know. You can try to change the values on your ecu if you have both of them..
Backup first!

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif