cedricp/ddt4all #199: Remote Engine Start (RES) for Scenic IV

sinwe, 2019-05-12:
Is anyone able to activate RES? I always fail to write on CMF1_BCM_DDT_11->WITH_REMOTE_START_CD to true. Does anyone have any clue?

Rcarnero73, 2020-01-17:
Yes, I activated it by mistake and now I am trying to remove it.
In spanish is”**programacion del arrranque motor**”
[image]
[image]

sinwe, 2020-07-24:
Does it really start the engine?

051X, 2020-10-05:
I have the same icon actually I was trying but it didn't work

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif