cedricp/ddt4all #200: Unicode?

vavincavent, 2019-05-12:
I have Linux installed (Debian) and for the first step, i have a bug with ddt4all :
To begin, some informations :
vincent@ASUS-R558UV:~$ uname -a
Linux ASUS-R558UV 4.9.0-9-amd64 #1 SMP Debian 4.9.168-1 (2019-04-12) x86_64 GNU/Linux

vincent@ASUS-R558UV:~$ python -V
Python 2.7.13

and the launching command :
vincent@ASUS-R558UV:~$ python '/home/vincent/Documents/FIAT TALENTO/DDT4ALL/ddt4all-master/ddt4all.py' 

after i click on "ELM Benchmark"  i get :
Trying to open port192.168.0.10:35000 @ 38400
Traceback (most recent call last):
  File "/home/vincent/Documents/FIAT TALENTO/DDT4ALL/ddt4all-master/ddt4all.py", line 1212, in check_elm
    res = elm.elm_checker(port, speed, self.logview, core.QCoreApplication)
  File "/home/vincent/Documents/FIAT TALENTO/DDT4ALL/ddt4all-master/elm.py", line 1389, in elm_checker
    print cm[2] + " " + res.strip()
  File "/usr/lib/python2.7/codecs.py", line 369, in write
    data, consumed = self.encode(object, self.errors)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xfc in position 47: ordinal not in range(128)
Abandon

Do i have missed something?

Vavincavent

cedricp, 2019-05-18:
It's strange you got unicode issue in this part of the code. Maybe an ELM issue... Have you checked at port speed ? Sometimes weird characters may output if badly set.
I remember a problem like this one and a terminal configuration issue, but can't recall what the problem was.
I can see a "FIAT" reference in your paths, hope you're not trying to connect this app to a FIAT vehicle, that won't work.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif