cedricp/ddt4all #201: Rlink 2 starts late

noville89, 2019-05-13:
Hi all, it is clear on title
I ve megand 4 bfb 2016 and 8.7 inch rlink 2 media nav
Version 2.2.19.300 (updated 3.3 and downgraded 2.2 again) 
I only had false multisense pictures ( espace pictures but i solved this) and once i ve touched mid button mistakenly while tryig to touch mid button ( solved with pyren scripts)

Now the problem is welcome screens, when i come closer the car, mirrors opens, i open the front gate and enter the car, main screen comes in 5 secs and rlink doesnt open. After 35-40 sec it opens and seems some laggy for 10-15 seconds. It responses like you hard reset it. Then everything turns normal.
This happens when i come to car after long time. For example at evening i come home and at morning when i go the car for going work it happens and so. If i close car and open it in short period this doesnt happen.
[image]

System info by developer mode is shown above.

Whats the problem? Does anyone know what causes this

shrlnm, 2019-05-13:
Did you pressed Log or Debug buttons?
[image]

Did you try to disconnect all USB flash drive?

noville89, 2019-05-13:
No i dont think so. I didnt press any of them. And it doesnt matter whether
usb plugged in or not. For example usb is empty there is only developer
files inside it

13 May 2019 Pzt 10:02 tarihinde shrlnm <notifications@github.com> şunu
yazdı:

franeti, 2019-05-13:
Always after a downgrade from 3 to 2, this happen. 
Try to upgrade to 3.X again, i think fix your issue.

noville89, 2019-05-13:
Would it be problem for guarantee? I want to make an appointment with service for v7.0 update. Does it matter going to service with v3.3? Maybe it seems stupid but could they say "you should be on 2.2 but you are on 3.3. you are out of guarantee"?

noville89, 2019-05-14:
Updating v3.3 does not solve the problem. I ve reverted back to v2.2.17. I will take the car to the service with native firmware for updating v7

noville89, 2019-05-16:
I take the car to the service. They confirm 7.0.24 update. I gave the car for upgrading. While i was working they called and said "your r-link was broken down it is not working anymore we will replace it under warranty" and gave me 20 working days to wait for☺️

closed 2019-05-16 by noville89