cedricp/ddt4all #204: turning off ASR/Traction Control with DDT4all ?

AndyIreland, 2019-05-24:
hello - i dont know if this is an issue or if its my fault or if the car cannot do this but i have been trying to disable the ASR/Traction control - I have sort of been succesfull but there is only 2 settings I can do for it. I can choose 'deactivated by diag' and 'activated by diag' and that seems to work it but when I disable traction by 'deactivated by diag' I of course have the words in the CDB 'traction off' and 'check ESC'  and the car light with the skid marks underneath it and the spanner light on the dashboard/cluster

So , i really want to achieve , if I can, to turn off the Traction Control with all the lights and messages off , (well I dont mind if the car light with the skid marks underneath) is still lit, thats OK .. so right I thought I will select 'Not Implemented' instead in the DDT4all ASR setting to see if that disables traction control but without warnings on the dash and i select 'not implemented' and then press 'write ASR config' then 'soft reset' but its not staying like that - when I refresh it goes back to previous and the spanner light and other messages are on the dash. 

I have got CAN timeout on 600ms and set it using extended (10C0) 

Is it something I am doing wrong or is it maybe a bug in DDT4all can someone tell me please? 

Car: Clio IV 2016 16v 1149cc  ABSESC- X10527998 version 3.8 ECU

[image]

AndyIreland, 2019-09-28:
Hello, I have since found out if you want to disable ASR but not have the spanner light and warnings to show you have to turn off ESC in cluster or put it to 'without ESC' - hope this helps others.

cvjensen, 2021-12-07:
Hi @AndyIreland perhaps you can help me disable ASR on my megane III? I would also like to change the display of Liter/100 KM to KM/L like we do it here in Denmark :)

AndyIreland, 2021-12-07:
> Hi @AndyIreland perhaps you can help me disable ASR on my megane III? I would also like to change the display of Liter/100 KM to KM/L like we do it here in Denmark :)

Hi yes I can do that for you remotely from here if you have a bluetooth or USB ELM327 and windows 10 laptop in the car , and Internet connection on the laptop in the car

AndyIreland, 2021-12-07:
It would be easier for me to log onto your computer, have a look at what ECU's show up in a search on your particular car and test the ELM if I did it remotely - rather than go through blindly explaining it

cvjensen, 2021-12-12:
> It would be easier for me to log onto your computer, have a look at what ECU's show up in a search on your particular car and test the ELM if I did it remotely - rather than go through blindly explaining it

I understand - perhaps we can find a day in the near future where we both have time to take a look at it :)

AndyIreland, 2021-12-12:
> I understand - perhaps we can find a day in the near future where we both have time to take a look at it :)

indeed - I am normally free to look in between hours of 2pm - 9pm British UK time (unless I am tied up with sorting out someone else) :)

duster91, 2023-04-08:
Hi Andy, really cool to see that it worked for you! I've been doing some research, and I was wondering if it's possible to do this by creating a physical switch to activate and deactivate the ASR/ESP. The Dacia Duster, which as I understand uses the same computer module as many other Renaults (Capture + Clio), has this switch in its 4x4 Dusters. The 4x2 Duster does not have this option. What do you think? Would this be possible, or the only way to do it via laptop. If it is possible, what steps would you recommend?
Cheers, Killian

AndyIreland, 2023-04-08:
> Hi Andy, really cool to see that it worked for you! I've been doing some research, and I was wondering if it's possible to do this by creating a physical switch to activate and deactivate the [...]

hiya Killian . I am not sure about the Dacia but with the clio 4 if you pull the fuse or put a switch in I think the ABS is on the same fuse/system so if you do pull the fuse you disable ABS as well. coding with DDT4ALL you can disable ASR/ESC separately and stll have the ABS working

duster91, 2023-04-09:
Thanks for your reply! Do you think it would be possible to Code a switch, just for the asr? If so, how would you approach it?

AndyIreland, 2023-04-09:
I wouldn't know for sure but I would say no, you would have to make up the wiring loom then there is not an existing parameter in the ECU to enable or disable a switch for ASR. - if there was then possibly doable

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

monalos, 2025-02-02:
Sometimes ddt4all cannot turn off the ASR from the ABS, it seems to have closed it, but when I start it again, it turns on, when I do it with Renolink, there is no problem. My question is, I want to write a code with Arduino and send an asr shutdown command to the required canbus addresses, can you comment on how I can do it? Which canbus address should I send it to? I looked at the ecu tweaker logs and it is sending data, but I couldn't find which address it sent it to.

monalos, 2025-02-02:
@shrlnm

monalos, 2025-02-03:
@AndyIreland