cedricp/ddt4all #207: Talisman Rlink 2 Black Screen After wrong version Update

danieltroni, 2019-05-29:
Hello.This is a half size rlink Talisman, has version 2.x and i was not well informed and found another talisman ver 3.3xx from internet and tried to uptade via usb. First it is really updated but when i controlled version number I saw the old 2.x numbers and i tried to update again it took same duration but never ended. That i stopped the car and system was black but it is try every 25 second to restart ( i know this because it in my dashport it shows for 1 second every 25 second the signal 4control label appears in my talisman dont have it. So i understood that the version belongs to a talisman witg 4control. I have usb 2 upload lohs if you want. And one more info every day only one time the radio and screen comes for 2 seconds and they gone. Button reset, fuse reset dont work. I have dd4all and epl32 usb but i never used them. Can anybody help me? Is there any possiblity that the brain is ok ?

shrlnm, 2019-05-31:
Hello. Do you suppose that the rlink2 is in cyclic reload? If it so I afraid I can't help, sorry. I may only try to write a script for reset rlink2 configuration to the factory defaults.

danieltroni, 2019-06-03:
Hello @shrlnm, yes it seems but i am not sure cause i am a newbee for
Rlink. The only think that I am observing, sometimes the system comes for a
3-4 seconds, for example when parking using back sensors, the screen comes,
radio comes, sensor sound comes for a 3-4 seconds that it goes. Renault
service cant do and reported to me that the brain is gone. Yes there is a
bad situation but I dont think that brain is dead, otherwise nothing can
observed. It seems that a cyclic position stops the system to be in action.

Can you send your script to my email ? onderimiz@gmail.com   I ordered an
ELM327 kit, next week I will try. It is possible that as my car dont have
4control and the bad version of Rlink try to activate 4control for a cyclic
reload that effects Rlink to run properly? Any help I can try

Thanks

shrlnm <notifications@github.com>, 31 May 2019 Cum, 09:04 tarihinde şunu
yazdı:

shrlnm, 2019-06-03:
What version of LGU was installed last time?

danieltroni, 2019-06-04:
Hello again

All my purpose was to have android enable rlink
 And i found another VIN number and tried to download a newer version. I
think it was a 3.3xx or 3.2xx if it is important after 3.x i can check it.

If you can able to resolve my situation i would like to donate to you
honestly..

I can even find if u like the wrong VIN number that i used to download the
file.

By the way my usb ports do not work dont have electricity

Thanks

danieltroni, 2019-06-04:
My original version was 2.xx i think 2.3 something..

4 Haz 2019 Sal 16:29 tarihinde onder yilmaz <onderimiz@gmail.com> şunu
yazdı:

tom3212, 2019-11-26:
Hello everyone
You can help me in this situation

R-link2 replaced radio * original display *
there is nothing on the display - black, the radio turns on can be seen on the meter,
how to run it

shrlnm, 2019-11-27:
> how to run it

Was it new one or from other car? Any way, first try https://gitlab.com/py_ren/pyren/wikis/how%20to%20rlink2%20black%20screen

tom3212, 2019-11-28:
Thanks for the help
the display turned on,
but it&#39;s twisted. how to reverse it

shrlnm, 2019-11-28:
@tom3212 give your e-mail please

tom3212, 2019-11-28:
Hello
The display was able to start, it works on MID 1i2 version,
but it is twisted, how to rotate it macro "rotata" does not work

this is a display in megane 4 and shows how it would be the big, vertical

tom3212, 2019-11-28:
tom3212@wp.pl 

       
    
       
         github.com @tom3212  give your e-mail please

engindalmis, 2019-12-21:
engindalmis26@hotmail.com

morfej13, 2020-08-05:
@shrlnm could you help me with Rlink2 on megane4? .... my email ciril13@yahoo.com

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif