cedricp/ddt4all #208: renault I reset my car's radio with dd4all. The radio does not work.  on the radio screen. Install the digital device. warning

atumelektronik, 2019-05-30:
renault I reset my car's radio with dd4all. The radio does not work.

on the radio screen. Install the digital device. warning

shrlnm, 2019-05-31:
What is the car?
What is the radio?
Which xml file did you use for reset?
Have you saved logs?

JAzevedo351, 2019-05-31:
Hello, the car is SCENIC IV BOSE EDITION, the radio is R-link2 software version (3.3.15.416), unfortunately I did not save the logs before the modification, nor later, I used the .xml available in DDT4ALL version 5.4.0, XFA - SCENIC IV / RadNav 3.x v3.3), I hope these data will help, if it does not help I will give the necessary information, thank you very much for the readiness you undertook to help.
[image]

JAzevedo351, 2019-05-31:
Sorry, that questions not for me...

atumelektronik, 2019-06-02:
> Araba nedir? 2011 Fluence
> Radyo nedir Oem no 281155040R
> Sıfırlama için hangi xml dosyasını kullandınız? ddt4all audio rest
> Günlükleri kaydettiniz mi?
problem. (connect diag unit)
[https://i.hizliresim.com/mXqyB4.png](url)
[https://www.drive2.ru/b/532332953708003369/](url)

BarektE, 2019-09-23:
bu bilgiyi yazanlar bilgi sahibi değiller. sayfa yararlı değil

closed 2019-09-23 by BarektE