cedricp/ddt4all #209: Set the clock

VD934, 2019-05-31:
How do I set the clock on the display? I managed to display the temp, but the clock only shows 0:00, dont know how to set it right.

AndyIreland, 2019-05-31:
> How do I set the clock on the display? I managed to display the temp, but the clock only shows 0:00, dont know how to set it right.

Hello - long press the top computer button on steering wheel stalk and 0 (hours digit) will start flashing, then press button again to advance hour then leave it for a while and the 00 (minutes) will flash and again press the button to advance the minutes

http://www.b-auto.net/renault_clio-4/display_a-2015.html

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif